gnuradio/gnuradio #3648: Gnuradio outputs error ModuleNotFoundError: No module named 'osmosdr'

tinyxuyan, 2020-07-16:
I have installed gr-osmosdr through `pybombs install gr-osmosdr`, but it output error `ModuleNotFoundError: No module named 'osmosdr'` when executing the flow graph.
I am sure I have installed the gr-osmosdr. When I type gqrx in the terminal, it outputs 
`gr-osmosdr v0.1.4-134-gf3905d35 (0.1.5git) gnuradio 3.8.1.0`. 
And I can find the OsmoSDR modules in gnuradio-companion.
I have tried to remove gr-osmosdr and reboot several times but got the same errors.
Does it mean that I don't build gr-osmosdr correctly or anything else?

jdemel, 2020-07-24:
So GQRX runs properly? Also with hardware that is connected through osmosdr? Does GNU Radio itself run properly? Are you able to execute example flowgraphs? 

Besides, you'd need to provide more info about your system. Which OS do you use? How did you install GNU Radio? Did you try different routes already? This might be a hint to a version conflict. How did you install GQRX? This might have caused a version conflict as well.

tinyxuyan, 2020-07-25:
I guess it is not the problem of GQRX, because when GQRX runs properly or outputs error like #https://github.com/csete/gqrx/issues/815, the issue of gnuradio still exists. And hardware is properly connected through osmosdr, I can find the information of hackRFONE through `hacker_info`. I have tested some flowgraphs without the hardware connection and all of them are run normally.

OS: Ubuntu 18.04LTS 
And I installed gnuradio through pybombs installation. I have tried several other installation methods, such as from source files, directly through `sudo apt-get install gnuradio`. The only available method is through pybombs installation. And I installed GQRX through `pybombs install gqrx`.

I am not sure the cause of version conflict, because I have tried many methods to install gnuradio and might not thoroughly remove the files installed by different methods.

jdemel, 2020-07-25:
So you figured out a PyBOMBS install is your favorite method to install everything. Mixing different installation methods is probably a recipe for desaster. I'd recommend to uninstall gnuradio, gqrx, osmosdr. Search your system for anything that still has "gnuradio" in it. e.g. shared libraries. Remove them too. Then reinstall everything with pybombs. Since you're using pybombs, it's advisable to install everything in a prefix. e.g. `~/prefix/gnuradio`.
Chances are some method you tried to install GNU Radio has an incomplete uninstall for some reason.
Besides, you can start to track down which libraries (python and C++ wise) are loaded for GQRX and GNU Radio, their locations, if their are any differences, try to load osmosdr in an interactive python shell (I prefer bpython3 for that). Though, that's probably the more difficult route to go. Especially if you haven't done this before.

tinyxuyan, 2020-07-28:
After removing all the files related to GNURadio, I reinstalled the GNURadio using PyBOMBS install. The gr-osmosdr is switched to version 0.2.0.0. I have tested gqrx with hackrfOne, and can demodulate FM signal normally which means that my hardware is correctly configured.
However, when I generate the flow graph with osmocom sink in it using GNURadio, the terminal outputs error listing as below:
```
Generating: '/home/xuyan/Github_repo/GNUradio_grc/hackrfONE/top_block.py'
Generate Error: (NameError("'dc_offset_mode1' is not defined",), 'osmosdr.sink(\n    args="numchan=" + str(${nchan}) + " " + ${args}\n)\n% if context.get(\'num_mboards\')() > 0:\n% if context.get(\'clock_source0\')():\nself.${id}.set_clock_source(${clock_source0}, 0)\n% endif\n% if context.get(\'time_source0\')():\nself.${id}.set_time_source(${time_source0}, 0)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 1:\n% if context.get(\'clock_source1\')():\nself.${id}.set_clock_source(${clock_source1}, 1)\n% endif\n% if context.get(\'tim`
```
It is still the error caused by osmosdr. Meanwhile, I have checked all the blocks' basic parameters which should be set and there is no parameter related to dc_offset_mode1. 
Is it still an error caused by wrong installation or version conflicts?

jdemel, 2020-07-28:
This looks like an osmosdr error. I recommend to ask directly there.

tinyxuyan, 2020-07-29:
Thank you for your advice.

closed 2020-07-29 by tinyxuyan

maximus-sallam, 2022-05-24:
Hi there. I just stumbled upon this thread while troubleshooting the same compilation error, "ModuleNotFoundError: No module named 'osmosdr'" I too am using Ubuntu 18.04 and I also attempted to install Gnuradio via the package manager initially. The version of the Gnuradio package available in the Ubuntu 18.04 repo (Gnuradio 3.7) is not compatible with gr-gsm. The minimum requirement is version 3.8 so I uninstalled 3.7 and installed Gnuradio 3.8 via the Gnuradio 3.8 PPA (https://wiki.gnuradio.org/index.php/InstallingGR). That resolved the Gnuradio error. As for the 'osmosdr' error, gr-osmosdr is installed as a Linux package, however, the error doesn't appear to be referring to the Linux package, but instead the Python module  'osmosdr' without the "gr-" prefixed to the beginning of the name. I was able to confirm this by attempting to import the module in both Python3 and Python2. Python3 failed to import the module but Python2 did not return an error when I imported the module. Upon further digging, I see that the env set in the Python files that contain the "import osmosdr" line are set as "#!/usr/bin/env python." This is the env for Python2, not Python3. In addition, to this, I found the following lines in the Python files: "#PYTHONPATH is needed as gr-osmosdr gets installed in /usr/local/lib/python3/...
ENV PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH." It looks like the Python2 path needs to be added to the env.

jdemel, 2022-05-24:
@maximus-sallam Ubuntu 18.04 ships with GR 3.7. Thus, all OOTs you install via Ubuntu repositories expect GR 3.7. After you installed GR 3.8 via the GR PPA, the OOTs need to be installed correctly.
You may compile them from source. Or search for the correct PPAs. Essentially, your issue seems to be that you try to mix an OOT and GR that belong to different versions.

The transition from GR 3.7 to 3.8 implies that we moved from Python2 to Python3 which is both absolutely required and a breaking change. Don't mix Python2 and 3 paths. That'll lead to more problems.

jdemel, 2023-04-21:
@yasmine-stu `gr-osmosdr` is a GR out of tree module. Depending on how you installed GR and gr-osmosdr, your approach to fix your issue might differ.

I suggest to ask for help on the GR mailing list, or in the GR matrix chat. You may find more info on that here:
https://www.gnuradio.org/community/

yasmine-stu, 2023-04-21:
i just extracted the zip file from git to install the GNU radio companion . it already contains the osmolib.
is there is smthg to do other than that ?
(i'm new in this field)

Yasmine Hentati
Final  year telecommunication engineering student at SUP'COM
phone: +216 54030234

https://www.gnuradio.org/community/