gnuradio/gnuradio #3659: What is this data type for the dark blue port on the FFT block?

alexxmarsh44, 2020-07-19:
On the FFT block I was trying to copy a flowgraph that was made in gnuradio 3.7 and on the video he connects his FFT directitly to a vector to streams block.  Mine however has a dark blue output port and I can't figure out how to connect it to a complex signal port.   What is the dark blue data type exactly and how can you use the FFT in the same way as it was used in the older GNUradio?  I have looked all over for an answer and have found none.  Thanks

mormj, 2020-07-20:
Without seeing a picture, I'm guessing you are referring to a vector of complex, which shows up as darker blue.  So you need to connect a complex signal port to a "stream to vector" port with `num items` set to the `fft size`

dkozel, 2020-08-11:
Hi @alexxmarsh44, did Josh's advice help? If not, feel free to reopen this or, better, join us on the main chat channel.
https://chat.gnuradio.org

closed 2020-08-11 by dkozel