gnuradio/gnuradio #3661: Frequency modulator is slowly drifting when normalizing to [-pi;+pi]

dernasherbrezon, 2020-07-20:
Hi all,

I'm seeing very weird behavior here:
https://github.com/gnuradio/gnuradio/blob/master/gr-analog/lib/frequency_modulator_fc_impl.cc#L52

I extracted this portion of code into separate C++ project and executed:
```c++
	float d_phase = -0.000246628;
	#define GR_M_PI 3.14159265358979323846
	#define F_PI ((float)(GR_M_PI))
	d_phase = std::fmod(d_phase + F_PI, 2.0f * F_PI) - F_PI;
	std::cout << d_phase << std::endl;
```

The output was ```-0.000246525```. Which is not the same as input: ```-0.000246628```. This angle is exactly between -pi and +pi, but the result value is a bit different. If I run it for a while, the phase rounding error will accumulate.

Is it the correct behavior and I am missing something?

Yamakaja, 2020-07-21:
> Is it the correct behavior and I am missing something?

While this behavior isn't optimal, it is to be expected - you're running against the limits of 32-bit floating point here:

IEEE-754 32 bit floats can be separated into three parts: The sign bit, the exponent and the fraction:
[image]

The value that is being represented by those three parts can then be calculated as follows:
[image]

Now the important part: Significant digits (I.e. those which actually carry meaningful information):
If look at our value in a fixed point binary representation, you'll see that the exponent merely shifts around our fraction bits, but it doesn't actually add anything to our number - which means that our significant digits are completely determined by the size of our fraction. In decimal, we would have log10(2) * 24 bits ~= 7 significant digits.

Now lets take a look at the situation above: You've started out with `d_phase = -0.000246628;`. Here we're specifying 6 significant digits, which can be stored in a float without significant error. But `d_phase + F_PI` might not behave as you expected, so i'll visualize the addition here:

```
#define F_PI 3.1415927
float d_phase = -0.000246628

        F_PI
+    d_phase
========== ===
  3.141592 7
- 0.000246 628
-       1  11 
========== ===
  3.141346 072

```
Notice that i've split off the last three digits; to add or subtract two floating point numbers, their exponent needs to be equal, which means that if our exponent is increased, our fraction needs to be shifted right. But since there isn't really more space available, we've just lost three decimal digits! => Our result after the addition will be ~ `3.141346`.

Assuming that fmod simply subtracts F_PI again we receive: ~`-0.000246`. Following a similar argument as for the addition, the only our first three digits here came out of that subtraction, while the rest of the mantissa is - well - "meaningless" (For the lack of a better word).

Looking at the two numbers you observed (`-0.000246525` and `-0.000246628`) you can tell that your first three significant digits are indeed the same.

Now a caveat: When i'm talking but integer counts of significant decimal digits, that isn't correct - technically `log10(2) * 24 = 7.224719` which is rather ugly, so you either look at this problem in base 2, or you round a little here and there ...
Also, while i've described an approach to floating point arithmetic, there are tricks that avoid some pitfalls, and i'm not super familiar with IEEE-754 and modern floating point hardware, so don't quote me here ^^

dernasherbrezon, 2020-07-21:
Yes, my major concern was here not with float pointing arithmetic. It is not precise and arithmetic operations can introduce errors. 

I was mostly puzzled by the algorithm itself. Here, we're trying to fit the number within the range. If the number already within the range, then I would expect to return that number, rather than doing floating point arithmetic and introducing errors.

For example I would expect something like this:

```
if( phase + PI > 2*PI) {
    number = phase + PI
    while number > 2*PI {
         number = number - 2*PI
    }
    phase = number
}
```

I think I saw something similar to this here: https://github.com/gnuradio/gnuradio/blob/a56c3a7f4908133dc6ceb93ffd3b2f9e1dda8e0e/gr-blocks/include/gnuradio/blocks/control_loop.h#L95

Yamakaja, 2020-07-21:
I'm curious, what is your application that you can't tolerate these absolutely minute amounts of drift?

dernasherbrezon, 2020-07-21:
After couple of samples phase error will accumulate and sin/cos will give error in 3rd number after the comma ( 0.001f ). 

I'm writing binary compatible GNURadio blocks, but in Java: https://github.com/dernasherbrezon/jradio

There, I will simulate BER curve for FSK demodulator. 

And in Java, there is no "fmod" function, because "%" works on float-pointing numbers very well.  I.e. if a < b, then for a % b = exactly a.

Yamakaja, 2020-07-21:
> After couple of samples phase error will accumulate and sin/cos will give error in 3rd number after the comma ( 0.001f ).
Yes, it does appear to do so. But my question was: _Why_ does it matter? The slightest LO mismatch between your transmitter and receiver will have exactly the same effect, if not much worse

dernasherbrezon, 2020-07-21:
I'm simulating BER curve. I.e. there is no physical transmitter and receiver. And everything is running in the same processes. When running simulation, I configure AWGN with 0 phase drift and expect my curve to be close to theory. Which might be or might not. 

Understanding impact of this issue might take more time than writing BER simulation itself.

dernasherbrezon, 2020-07-21:
Just realized another thing:

https://github.com/gnuradio/gnuradio/blob/master/gr-analog/lib/frequency_modulator_fc_impl.cc#L50

The comment says phase would be  [-pi ; +pi]. But according to the code, it will be in the interval (-3pi ; +pi).

This looks very weird to me.

Yamakaja, 2020-07-21:
It does indeed. And while our invariant has been violated, the code will still do what we expect, because d_phase is only used in the [conversion to an angle](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/include/gnuradio/fxpt.h#L40) that performs another (correct) -PI/PI fold.

ValZapod, 2020-08-17:
And that was like that before this commit (you need to git blame before the clang-format commit). https://github.com/gnuradio/gnuradio/commit/9ae9a10cadec5c1ea7c03677cc685b3a92591967

This was the commit that implemented it. https://github.com/gnuradio/gnuradio/commit/8c9a5fdb19046c3cb8d25d6b77dbac87d49d8391

marcusmueller, 2024-01-22:
Still present, as seen in #6800