gnuradio/gnuradio #3662: Correlation Estimator for float streams

mildsunrise, 2020-07-21:
Hi! The Correlation Estimator block is currently designed with PSK systems in mind (it seems) and only offers a complex->complex version. However GFSK packetized transmissions are common as well, so maybe it would be good to have a float version too.

Shunmathi10, 2026-10-21:
Hello! I'm interested in wireless communication and software-defined radio. I'm learning to contribute to open-source projects and would like to explore this issue.
Is this feature still needed? As a first-time contributor to GNU Radio, I'd appreciate guidance on a small, well-defined starting task.

Thank you!

marcusmueller, 2026-10-21:
After having re-read this issue: removing the good first issue tag. Yes, of course this feature is still desired (we haven't closed it!), but two things:

- what the correlation estimator currently does is a three-point complex interpolation estimate around the correlation peak. One can do that, but it doesn't make as much sense on bandpass-subsampled band pass signals, so you'd need to assume things are already in baseband.
- I don't think we should be maintaining a copy of this block just modified to use real-valued signals, this should all be templated, just as our filters are, for type of input, sequence, and output. Conversion of an existing block to templates without breakage is not a beginner-friendly issue.