gnuradio/gnuradio #3663: QT Sinks: Configuration for More Than 10 Inputs Generates Broken Code

ekigwana, 2020-07-22:
Version: 3.9.0.0-git master

To reproduce:

1. Create flowgraph as shown below or use attached GRC script.
[image]

2. Execute flowgraph.

3. Expected behavior:
11 signals overlayed

4. Console output:
```
UOTraceback (most recent call last):
  File "/home/ekigwana/gnuradio/qt_freq_sink_input_test.py", line 191, in <module>
    main()
  File "/home/ekigwana/gnuradio/qt_freq_sink_input_test.py", line 167, in main
    tb = top_block_cls()
  File "/home/ekigwana/gnuradio/qt_freq_sink_input_test.py", line 110, in __init__
    if len(labels[i]) == 0:
IndexError: list index out of range

>>> Done (return code -11)
```
5. Notes:
- Looks like the generated code for the trace data is limited to 10. If this is a hard limit, the block should have an assert. By the looks of it the frequency sink configuration page does not show options for the eleventh item either.
- From this [code](https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/grc/qtgui_freq_sink_x.block.yml#L160) (one of several affected blocks) it appears the max number of inputs is limited due to the options in the YAML files
- Possible fixes: Assert if [nconnections](https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/grc/qtgui_freq_sink_x.block.yml#L91) > 10 and do something similar for grc/qtgui_time_sink_x.block.yml, grc/qtgui_eye_sink_x.block.yml, grc/qtgui_histogram_sink_x.block.yml, grc/qtgui_vector_sink_f.block.yml. The alternative would be auto generate hard coded items such as Line labels etc and that would be way more effort.

[qt_freq_sink_input_test.zip](https://github.com/gnuradio/gnuradio/files/4961805/qt_freq_sink_input_test.zip)

dkozel, 2020-07-30:
Hi @ekigwana Thanks for reporting this.

Adding an assert would be a good quick fix to make it clear what the current limitation of the block in GRC is. It would be good to find a generic way of solving this issue of multiple identical elements in YAML. I've forgotten what the exact term is, but there's at least partial support for declaring that some element is to be repeated.

marcusmueller, 2024-01-22:
still reproducible