gnuradio/gnuradio #3664: QT RangeWidge Error

ChengDaHaI, 2020-07-22:
Hi,

My GNU Radio version is 3.8. My python version is 3.8.2. My OS is Ubuntu 20.04.

I installed GNU Radio by:

```
sudo add-apt-repository ppa:gnuradio/gnuradio-releases
sudo apt-get update
sudo apt install gnuradio
```

When i first run a sample rate demo, i got the following error:

```
self._snr_win = RangeWidget(self._snr_range, self.set_snr, 'snr', "counter_slider", float, QtCore.Qt.Horizontal)
TypeError: __init__() takes from 5 to 6 positional arguments but 7 were given
```
I solve the error by manually removing the latest parameter QtCore.Qt.Horizontal. Some guys said that results from my previous installed GNU Radio (master branch, version 3.9 ). But i have removed the old gnu radio by 

```
sudo apt-get remove --auto-remove gnuradio
sudo apt-get purge --auto-remove gnuradio  
```
It doesn't help.

Any idea about how to solve the problem?

Paul-G8DVO, 2020-07-28:
Did same update to 3.8 using PPA and got same problem
Seems to be a missing line to import QtCore in file:
/usr/share/gnuradio/grc/blocks/qtgui_range.block.yml

open file (eg sudo gedit <location/filename>) and edit imports to:
    imports: |- 
        from gnuradio.qtgui import Range, RangeWidget
        from PyQt5 import QtCore

ChengDaHaI, 2020-07-28:
> Did same update to 3.8 using PPA and got same problem
> Seems to be a missing line to import QtCore in file:
> /usr/share/gnuradio/grc/blocks/qtgui_range.block.yml
>
> open file (eg sudo gedit [...]

It works for me! Thank you very much!

closed 2020-07-28 by ChengDaHaI