gnuradio/gnuradio #3666: How can I downgrade from 3.8 to 3.7?

ccskying, 2020-07-23:


jdemel, 2020-07-24:
You use `apt` thus you probably use a debianoid distribution. The exact version would be required to properly answer your question properly. I suspect your distro repositories provide a GR 3.8 package. `apt` installs the package with the highest version string, in this case GR 3.8 apparently.
Since python2 is effectively dead since 2020-01-01, it would probably be favorable to move your python code to python3 instead of a GR downgrade. In case this is supposed to be a quick test you may be better off with e.g. a Docker container.

dkozel, 2020-07-30:
@ccskying I would guess that you're on Ubuntu 20.04 which has GNU Radio 3.8 in it's repository and that Apt is using the newer release by default. You can look into manually selecting in apt which package to install.

If you try this and have more questions please join us in the chat as that's a much better place to answer questions.
https://chat.gnuradio.org

closed 2020-07-30 by dkozel