gnuradio/gnuradio #3669: No video output with module "video SDL sink"

MKesenheimer, 2020-07-24:
I can't get the video output of the module "video SDL sink" on macOS Catalina (10.15.6) to work.
I generated a flow graph which passes random values to the video SDL sink. However, no video is shown.

This is the flow graph I tested the issue with:
[video_sdl_test.zip](https://github.com/gnuradio/gnuradio/files/4974348/video_sdl_test.zip)

And this is the result (the colors of the screenshot are a bit off. But no worries, in reality the windows look a lot better).
[image]

gnuradio was installed with macports and has version number 3.8.1.0 (python 3.7.8). I tested the exact same flow graph on a linux machine and the video is displayed correctly.

MKesenheimer, 2020-09-04:
With the update to  gnuradio 3.8.2.0 and Python 3.8.5, now a more meaningful error message is displayed:

```
Unable to obtain graphics context for NSWindow (Mojave behavior)
SDL screen_mode 32 bits-per-pixel
SDL overlay_mode 842094169 
```

Maybe someone can deal with this error message.

MKesenheimer, 2020-09-09:
I may have the answer to this problem: The issue lies not in gnuradio per se but in the incompatibility of SDL and macOS Mojave and Catalina.
The solution would be to upgrade to SDL2.

MKesenheimer, 2020-09-11:
I created a simple SDL2 program which reads data from a named pipe and displays the data as a video. Exactly how it is done with the SDL video sink of gnuradio, however this program relies on SDL2. This program is not part of gnuradio and can be used as a standalone program.

https://github.com/MKesenheimer/SDLDisplay

You have to create a named pipe (mkfifo </path/to/named/pipe>) and write the video data to the named pipe with the file sink block. More information can be found on my github page.

michaelld, 2021-09-25:
@MKesenheimer thanks for the sleuthing work. I'm going to close this issue since it doesn't seem to be with GR per se. If anyone disagrees, please reopen and post info about what's going on.

closed 2021-09-25 by michaelld