gnuradio/gnuradio #3670: gqrx fails linking with undefined reference to `gr::digital::map_bb_impl::s_map_size'

skinkie, 2020-07-25:
I am trying to compile gqrx on gentoo against gnuradio 3.8.1 (to resolve boost issues, a patch was applied). The linking of gqrx fails eventually with the output below. The closest issue here was #3014, but that seems to be related to compiling gnuradio itself. It seems the symbol itself is part of the .so; _ZN2gr7digital11map_bb_impl10s_map_size

Could anyone give a hint where I should look for a solution?

```
FAILED: src/gqrx 
: && /usr/bin/x86_64-pc-linux-gnu-g++  -O2 -march=znver1 -O2 -pipe -pipe -Wl,-O1 -Wl,--as-needed  -Wl,-O1 -Wl,--as-needed    -rdynamic src/CMakeFiles/gqrx.dir/gqrx_autogen/mocs_compilation.cpp.o src/CMakeFiles/gqrx.dir/applications/gqrx/main.cpp.o src/CMakeFiles/gqrx.dir/applications/gqrx/mainwindow.cpp.o src/CMakeFiles/gqrx.dir/applications/gqrx/receiver.cpp.o src/CMakeFiles/gqrx.dir/applications/gqrx/remote_control_settings.cpp.o src/CMakeFiles/gqrx.dir/applications/gqrx/remote_control.cpp.o src/CMakeFiles/gqrx.dir/applications/gqrx/file_resources.cpp.o src/CMakeFiles/gqrx.dir/dsp/afsk1200/cafsk12.cpp.o src/CMakeFiles/gqrx.dir/dsp/afsk1200/costabf.c.o src/CMakeFiles/gqrx.dir/dsp/filter/fir_decim.cpp.o src/CMakeFiles/gqrx.dir/dsp/rds/decoder_impl.cc.o src/CMakeFiles/gqrx.dir/dsp/rds/parser_impl.cc.o src/CMakeFiles/gqrx.dir/dsp/agc_impl.cpp.o src/CMakeFiles/gqrx.dir/dsp/correct_iq_cc.cpp.o src/CMakeFiles/gqrx.dir/dsp/downconverter.cpp.o src/CMakeFiles/gqrx.dir/dsp/fm_deemph.cpp.o src/CMakeFiles/gqrx.dir/dsp/lpf.cpp.o src/CMakeFiles/gqrx.dir/dsp/resampler_xx.cpp.o src/CMakeFiles/gqrx.dir/dsp/rx_agc_xx.cpp.o src/CMakeFiles/gqrx.dir/dsp/rx_demod_am.cpp.o src/CMakeFiles/gqrx.dir/dsp/rx_demod_fm.cpp.o src/CMakeFiles/gqrx.dir/dsp/rx_fft.cpp.o src/CMakeFiles/gqrx.dir/dsp/rx_filter.cpp.o src/CMakeFiles/gqrx.dir/dsp/rx_meter.cpp.o src/CMakeFiles/gqrx.dir/dsp/rx_noise_blanker_cc.cpp.o src/CMakeFiles/gqrx.dir/dsp/rx_rds.cpp.o src/CMakeFiles/gqrx.dir/dsp/sniffer_f.cpp.o src/CMakeFiles/gqrx.dir/dsp/stereo_demod.cpp.o src/CMakeFiles/gqrx.dir/interfaces/udp_sink_f.cpp.o src/CMakeFiles/gqrx.dir/qtgui/afsk1200win.cpp.o src/CMakeFiles/gqrx.dir/qtgui/agc_options.cpp.o src/CMakeFiles/gqrx.dir/qtgui/audio_options.cpp.o src/CMakeFiles/gqrx.dir/qtgui/bookmarks.cpp.o src/CMakeFiles/gqrx.dir/qtgui/bookmarkstablemodel.cpp.o src/CMakeFiles/gqrx.dir/qtgui/bookmarkstaglist.cpp.o src/CMakeFiles/gqrx.dir/qtgui/ctk/ctkRangeSlider.cpp.o src/CMakeFiles/gqrx.dir/qtgui/demod_options.cpp.o src/CMakeFiles/gqrx.dir/qtgui/dockaudio.cpp.o src/CMakeFiles/gqrx.dir/qtgui/dockbookmarks.cpp.o src/CMakeFiles/gqrx.dir/qtgui/dockfft.cpp.o src/CMakeFiles/gqrx.dir/qtgui/dockinputctl.cpp.o src/CMakeFiles/gqrx.dir/qtgui/dockrds.cpp.o src/CMakeFiles/gqrx.dir/qtgui/dockrxopt.cpp.o src/CMakeFiles/gqrx.dir/qtgui/freqctrl.cpp.o src/CMakeFiles/gqrx.dir/qtgui/ioconfig.cpp.o src/CMakeFiles/gqrx.dir/qtgui/iq_tool.cpp.o src/CMakeFiles/gqrx.dir/qtgui/meter.cpp.o src/CMakeFiles/gqrx.dir/qtgui/nb_options.cpp.o src/CMakeFiles/gqrx.dir/qtgui/plotter.cpp.o src/CMakeFiles/gqrx.dir/qtgui/qtcolorpicker.cpp.o src/CMakeFiles/gqrx.dir/receivers/nbrx.cpp.o src/CMakeFiles/gqrx.dir/receivers/receiver_base.cpp.o src/CMakeFiles/gqrx.dir/receivers/wfmrx.cpp.o src/CMakeFiles/gqrx.dir/portaudio/device_list.cpp.o src/CMakeFiles/gqrx.dir/portaudio/portaudio_sink.cpp.o src/CMakeFiles/gqrx.dir/qrc_icons.cpp.o src/CMakeFiles/gqrx.dir/qrc_textfiles.cpp.o  -o src/gqrx  /usr/lib64/libQt5Network.so.5.15.0  /usr/lib64/libQt5Svg.so.5.15.0  -lboost_date_time-mt  -lboost_program_options-mt  -lboost_filesystem-mt  -lboost_system-mt  -lboost_regex-mt  -lboost_thread-mt  -lpthread  -lboost_chrono-mt  -lboost_atomic-mt  -lgnuradio-osmosdr  -lportaudio  -lvolk  /usr/lib64/libgnuradio-digital.so.3.8.1.0  /usr/lib64/libgnuradio-audio.so.3.8.1.0  -lvolk  /usr/lib64/libQt5Widgets.so.5.15.0  /usr/lib64/libQt5Gui.so.5.15.0  /usr/lib64/libQt5Core.so.5.15.0  /usr/lib64/libgnuradio-analog.so.3.8.1.0  /usr/lib64/libgnuradio-filter.so.3.8.1.0  /usr/lib64/libgnuradio-blocks.so.3.8.1.0  /usr/lib64/libgnuradio-fft.so.3.8.1.0  -lfftw3f  -lfftw3f_threads  /usr/lib64/libgnuradio-runtime.so.3.8.1.0  /usr/lib64/libvolk.so.2.3  -ldl  -lm  /usr/lib64/libgnuradio-pmt.so.3.8.1.0  /usr/lib64/libboost_program_options-mt.so  /usr/lib64/libboost_filesystem-mt.so  /usr/lib64/libboost_system-mt.so  /usr/lib64/libboost_regex-mt.so  /usr/lib64/libboost_thread-mt.so  /usr/lib64/libboost_chrono-mt.so  /usr/lib64/libboost_date_time-mt.so  /usr/lib64/libboost_atomic-mt.so  -lpthread  -llog4cpp  -lgmpxx  -lgmp  -lrt  -lasound && :
/usr/lib/gcc/x86_64-pc-linux-gnu/10.1.0/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib64/libgnuradio-digital.so.3.8.1.0: undefined reference to `gr::digital::map_bb_impl::s_map_size'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

skinkie, 2020-07-25:
I was able to compile gqrx using this patch to gnuradio.

```diff
--- a/gr-digital/lib/map_bb_impl.cc	2020-07-25 23:45:27.775120413 +0200
+++ b/gr-digital/lib/map_bb_impl.cc	2020-07-25 23:47:39.145062437 +0200
@@ -49,20 +49,20 @@
 {
     gr::thread::scoped_lock guard(d_mutex);
 
-    for (int i = 0; i < s_map_size; i++) {
+    for (int i = 0; i < gr::digital::map_bb_impl::s_map_size; i++) {
         d_map[i] = i;
     }
 
-    const unsigned int size = std::min(s_map_size, map.size());
+    const unsigned int size = std::min(gr::digital::map_bb_impl::s_map_size, map.size());
     for (unsigned int i = 0; i < size; i++)
         d_map[i] = map[i];
 }
 
 std::vector<int> map_bb_impl::map() const
 {
-    std::vector<int> m(s_map_size);
+    std::vector<int> m(gr::digital::map_bb_impl::s_map_size);
     gr::thread::scoped_lock guard(d_mutex);
-    for (unsigned i = 0; i < s_map_size; i++) {
+    for (unsigned i = 0; i < gr::digital::map_bb_impl::s_map_size; i++) {
         m[i] = d_map[i];
     }
     return m;
```

rear1019, 2020-07-26:
Oh, well. `s_map_size` is declared but not defined. Could you please check whether adding the line `constexpr size_t map_bb_impl::s_map_size;` to `map_bb_impl.cc` fixes the issue? This is the “proper” fix (which should not be required with C++17 as `constexpr` implies `inline`).

skinkie, 2020-07-26:
Do you have a suggestion on what line to add it?

rear1019, 2020-07-26:
Below
```
namespace gr {
namespace digital {
```

skinkie, 2020-07-26:
This works:

```C
--- a/gr-digital/lib/map_bb_impl.cc	2020-04-08 19:21:10.000000000 +0200
+++ b/gr-digital/lib/map_bb_impl.cc	2020-07-26 20:46:30.469120343 +0200
@@ -29,6 +29,7 @@
 
 namespace gr {
 namespace digital {
+constexpr size_t map_bb_impl::s_map_size;
 
 map_bb::sptr map_bb::make(const std::vector<int>& map)
 {
```

willcode, 2021-03-22:
Fix, verified by reporter, was added to 3.8 by ref'd PR. Closing.

PR #3677 (2020-08-03): digital/map_bb: Add missing definition for a static class member
This fixes #3670 and possibly #3014.

Which C++ revision is used on master? As far as I know, this patch shouldn’t be required with C++17 (and shouldn’t hurt either).
commit 3282fa75 (2020-08-03): digital/map_bb: Add missing definition for a static class member

This fixes #3670.
closed 2021-03-22 by willcode