gnuradio/gnuradio #3673: GNU Radio ControlPort Error (Possiblely related with Thrift)

ChengDaHaI, 2020-07-28:
Hi,

My GNU Radio version is 3.8. My python version is 3.8.2. My OS is Ubuntu 20.04.

I installed GNU Radio by:

```
sudo add-apt-repository ppa:gnuradio/gnuradio-releases
sudo apt-get update
sudo apt install gnuradio
```
When I am running the wifi loopback example from [gr-ieee802-11](https://github.com/bastibl/gr-ieee802-11), i got the following error

`python3: /build/gnuradio-HcCmR9/gnuradio-3.8.1.0-0~gnuradio~focal/gnuradio-runtime/lib/controlport/rpcmanager.cc:38: static rpcserver_booter_base* rpcmanager::get(): Assertion `booter_registered || aggregator_registered' failed`

This seems to be related with ControlPort module as well as Thrift lib. I have install Thrift by 
`sudo apt install thrift-compiler python3-thrift 
`
following the [suggestion](https://lists.gnu.org/archive/html/discuss-gnuradio/2018-06/msg00150.html). However, it doesn't help. 
Have anyone faced the issue?

Thanks.

zombielinux, 2020-10-22:
Has there been any progress made on this issue? I am suffering from the same bug.

ChengDaHaI, 2020-10-23:
> Has there been any progress made on this issue? I am suffering from the same bug.

I solve this issue by reinstalling GNU Radio and gr-ieee802-11, both in version 3.7. (Note: strictly follow the installation steps of gr-ieee802-11).

zombielinux, 2020-10-23:
> > Has there been any progress made on this issue? I am suffering from the same bug.
> 
> I solve this issue by reinstalling GNU Radio and gr-ieee802-11, both in version 3.7. (Note: strictly follow the installation steps of gr-ieee802-11).

That doesn't solve the problem with GnuRadio 3.8. I'm using the same ieee802-11 library. As long as the controlport block is set to false, the simulation runs perfectly.

bastibl, 2020-10-26:
Maybe it's an issue with Ubuntu's Thrift (which ships with 0.13) and GNU Radio's Thrift (which ships Thrift 0.11 in its PPA, since it is not yet compatible with 0.13). Can you check your installed versions (apt search thrift) and, in case, try uninstalling any 0.13 packages.

zombielinux, 2020-10-27:
@bastibl The PPA for the latest LTS release (20.04 Focal) is built around 0.13.0, and does not have the 0.11.0 packages in the repository. I'd say that its a bug

bastibl, 2020-10-27:
Not sure I understand. Ubuntu 20.04 comes w/ 0.13. The PPA you added includes 0.11 https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-releases

zombielinux, 2020-10-27:
@bastibl The PPA only includes 0.11 for ubuntu bionic. For Focal (the latest LTS) it doesn't include any libthrift at all. 

FWIW, the bug exists with CtrlPort Monitor as well as CtrlPort Performance Monitor.

bastibl, 2020-10-27:
Oh, I see... Sorry. I just tested the PPA version in a VM. Neither `gnuradio` nor `libgnuradio-runtime3.8.2` depend on Thrift (`apt-cache depends gnuradio` etc). If you run `gnuradio-config-info --enabled-components` it shows that `gr-ctrlport` is enabled but not `thirft`.
The initial idea was that there could be several backends that implement the CtrlPort interface. In practice, there is only Thrift. AFAIS, the current config enables CtrlPort but no backend that would implement it. Seems like the result of this is that the blocks show up in GRC, but if you use them you'll get an error.
I think there are two ways to solve this. Either compile the package without `gr-ctrlport` or manually apply the two patches that are required to support Thrift 0.13 in the build process of the package (with these patches you can use Thrift from the main repository). But that's something for @mormj to decide

zombielinux, 2020-10-27:
@bastibl Where are those two patches? Could you link them here?

bastibl, 2020-10-27:
First one is https://github.com/gnuradio/gnuradio/pull/2739
Second one is: https://github.com/gnuradio/gnuradio/pull/2739#issuecomment-593910061 (i.e. `s/PlatformThreadFactory/PlatformThreadFactory/` in `thrift_server_template.h`).

rafaelcalcada, 2021-02-11:
I'm using GNU Radio 3.8 and Ubuntu 20.04 and found this bug as well. I solved it running:
```
$ sudo apt install thrift-compiler python3-thrift
```
Then I reinstalled GNU Radio 3.8 and everything worked fine.

mbr0wn, 2021-05-14:
Looks like this is fixed. Please reopen if that's not actually the case.

closed 2021-05-14 by mbr0wn