gnuradio/gnuradio #3675: Adaptive resampling for audio sources

NerdyProjects, 2020-07-29:
Dear all,

I would like to use gnuradio for an application where I'd want to have good quality low latency audio passing through a flowgraph. As far as I can see, there is no adaptive audio resampling build into any of the available gr-audio sources/sinks which would be necessary to solve the two-clock problem without regular over- or underruns.

Have there been any conceptual thoughts already? I would be mostly interested in having a jack source with these capabilities and would implement it there - Would it make sense to try doing it in a state that it can be upstreamed?

I would basically follow the approach described in that paper: https://kokkinizita.linuxaudio.org/papers/adapt-resamp.pdf using the zita-resampler (https://kokkinizita.linuxaudio.org/linuxaudio/zita-resampler/resampler.html)

bistromath, 2020-07-29:
This has been discussed a number of times in the past, usually accompanied
by wistful sighing that it sure would be great if someone were to take this
on. It would be a perfect GSoC project, for instance.

Given that Gnuradio has a number of state-of-the-art resamplers already,
I'd recommend starting with one of them (likely the polyphase arbitrary
resampler) rather than zita-resampler, unless zita-resampler does something
that is intrinsically impossible using one of the existing resamplers.

If you were to take this on and get it upstreamed, I'd be happy to fund you
with beer, USRPs, or a donation to your charity of choice.

Nick

NerdyProjects, 2020-07-30:
Thanks for the writeup, Nick.

My reason to use zita-resampler would be that `VResampler` provides everything needed here, e.g. the ability to fine-tune the resampling rate (without recalculating filter taps). I am not at all in the maths behind but from a brief read I don't see similar functionality in e.g. the polyphase resampler.

dkozel, 2020-07-30:
Hi @NerdyProjects,

This is a really desirable feature and it'd be great to have it upstreamed! Adding an additional dependency is always a big deal and GNU Radio has (at least) three backends for audio on Linux including Jack.

I don't know enough about the variable resampling demands of this application and the abilities of the existing filters to know if one of them could be appropriate. Do you have an idea of where in the GNU Radio blocks this functionality would be implemented? Jack (in theory) could also work on Windows so it'd ideally be something optional at compile time.

NerdyProjects, 2020-07-31:
So a system as described by fons adriaensen needs two main components:
a) a resampler which resample ratio can be fine tuned at runtime (e.g. applying a factor 0.98..1.02) without resetting the filter itself
b) a control loop, that fine tunes the resampling filter.

Both are independent problems. The paper uses an approach where they have two threads per "signal barrier" (e.g. the audio thread writing and the signal processing thread reading) working on a lock-free audio buffer. The model as described in the paper would match exactly what can be implemented in the audio source plugin itself, e.g. in the jack module.

I'd like to play with an approach that I see a much broader usecase in gnuradio: Trying to use the control loop inside a regular gnuradio block that can be attached to the flowgraph at any point (e.g. after the audio source), so it does not have any dependency to the specific audio source used (and there might be other use cases of synchronizing two clocks anyway). Conceptually, it feels to me this approach can work by using the number of elements in the input/output buffer as input to the control loop.

I'll make time for this in the next days as I am a bit excited now :-) Both your comments sounded very motivating to me, thanks for that!

NerdyProjects, 2020-08-01:
If the described approach work, this doesn't need upstreaming soon, as it can just be and stay a 3rd party module until people consider it worthy enough. So I started it as an oot module here: https://github.com/NerdyProjects/gr-adaptive_resampler

(No code yet; I'll get back here as soon as it's done)

ast, 2021-11-27:
I've done this. I wrote an Alsa sink that used a PID controller to adjust the resampling rate. It worked well. It was easy with Alsa since there's an API to get ring buffer fill level. The ring buffer can't be too small though to account for non deterministic behaviour of a none real time operating system.

NerdyProjects, 2021-11-27:
Oh nice, @ast is your code available somewhere? I unfortunately didn't get around to doing anything yet.

ast, 2021-11-27:
https://github.com/ast/gr-sm6wjm

If there's any interest I can put some more work into it.