gnuradio/gnuradio #3678: MSK symbol synchronizers are broken

bistromath, 2020-08-01:
Created re: the [discussion](https://lists.gnu.org/archive/html/discuss-gnuradio/2020-06/msg00173.html) on the list.

The older MSK timing recovery block works while the newer symbol synchronizer doesn't because the MSK timing recovery block deviates from the paper to conjugate the squared symbol **half** a sample delayed, rather than a full sample. This is equivalent to changing the nonlinearity in `timing_estimator_impl.cc` to:

`u = (d_input[0] * d_input[0] * conj(d_input[1] * d_input[1])) - (d_input[1] * d_input[1] * conj(d_input[2] * d_input[2]))`

However, this results in sampling at exactly the wrong time -- the above needs to be modified to delay by an additional half symbol period to lock onto the correct symbol phase:

`u = (d_input[1] * d_input[1] * conj(d_input[2] * d_input[2])) - (d_input[2] * d_input[2] * conj(d_input[3] * d_input[3]))`

The older timing recovery block's behavior appears to be a happy accident due to an order of operations bug: `d_dly_conj_2` is set after `d_dly_conj_1`, and thus isn't actually delayed, resulting in the nonlinearity using a delay of half a symbol period rather than a full symbol period.

I can't account for two things: why a working nonlinearity deviates from the one in the D'Andrea and Mengali paper, and why the timing estimator needs to be delayed an additional half a symbol period in order to arrive at the correct sampling instant.

I believe the latter can be answered by looking at the original paper -- there's no resampler in the output, and so the paper doesn't have to deal with answering the question of what phase to output. Also regarding the latter, I **think** that eq. 9.4.4 (from Synch. Techn. for Dig. Recv. p. 498) implies a one-half-symbol offset in the error detector since it asks for the previous and next half-symbol periods, not the current and 2x periods.

As for the former, I just don't know. I'm trying to work that out now.

bistromath, 2020-08-01:
Eq. 9.4.4, for reference:
[image]

awalls-cx18, 2020-08-16:
I just checked with an Octave script simulation:

The Symbol Sync Block's Gen MSK TED S-curve is weird, in a way that one would think it is fixable.  Maybe.

Your proposed TED's S-Curve looks right except that it is inverted and has negative gain (so it corrects you in the wrong direction) and it's response isn't as strong.  The direction problem can be fixed by multiplication with -1.

I tested a variant of your proposed TED, but *not* delaying that half of a sample.  The response was the same as yours, but the sign problem was fixed. :)

I'm attaching my Octave script for your enjoyment.  Changing the N_bits variable will greatly affect the time the script takes to run, and moderately affects the S-Curve estimate.  I'm also adding a few pictures.

S-Curves:
[image]

TED Gains (Slope of S-Curves)
[image]

awalls-cx18, 2020-08-16:
I think I see what's going on with the original TED.  It's just shifted 1/4 of a symbol timing offset.  Probably because of the points of the discrete derivative.  I'll try and experiment to find a fix.

awalls-cx18, 2020-08-16:
Shifting the original TED computations by 1/4 of a symbol and negating the output appears to work.  The TED response is stronger with a higher gain, that what you proposed for the TED.  I had to negate the sign probably because I chose to look back 1/4 symbol vs. try to peek ahead a 1/4 symbol (since peeking ahead is a hassle even though the Symbol Sync block does already do it for some TEDs).

S-Curves
[image]

TED Gains (S-Curve slopes)
[image]

[msk_ted_investigation.m.txt](https://github.com/gnuradio/gnuradio/files/5081466/msk_ted_investigation.m.txt)

closed 2021-03-10 by mbr0wn

daniestevez, 2021-03-18:
Hi! I've been taking a look at the [D'Andrea and Mengali](https://www.researchgate.net/publication/3153478_A_digital_approach_to_clock_recovery_in_generalized_MSK) paper to try to understand all this. I'm not 100% sure about everything yet, but I wanted to share my findings/ideas here in case someone finds them useful. My main goal is to try to explain why the TED needs to have that specific form. In particular, why Andy needed to shift the TED by 1/4 symbol.

In the paper, the function

<img src="https://render.githubusercontent.com/render/math?math=c(t)=z^2(t)\overline{z}^2(t-T)">

is used as the basis for the TED. It is observed that for MSK

<img src="https://render.githubusercontent.com/render/math?math=c(t)=-\frac{1}{2}\left(1%2B\cos\frac{2\pi t}{T}\right)">.

Thus, an appropriate TED can be built out of

<img src="https://render.githubusercontent.com/render/math?math=c'(t)=\frac{\pi}{T}\sin\frac{2\pi t}{T}">.

The S-curve of this TED has a stable zero at t = kT.

However, something important is to note the definition of the function q(t) for MSK in page 233 of the paper. This implies that the symbol centres happen at t = T/2 + kT (the symbols last from kT to (k+1)T).

Now, the idea to approximate the derivative c'(t) as a difference is to use c(t) - c(t-T/2). However, this should be thought as an approximation to c'(t - T/4), not to c'(t). More formally, we can check that the S-curve of c(t) - c(t-T/2) has stable zeros at t = T/4 + kT (it's very easy to plot this or even work the math using the sum to product trig formulas). Therefore, instead of working with c(t) - c(t-T/2), we decide to work with c(t - T/4) - c(t - 3T/4). This is an approximation for c'(t - T/2), and has stable zeros at t = T/2  + kT, which as remarked above, coincide with the symbol centres.

I think this is basically what is happening in Andy's MATLAB simulation for the new TED. His proposed TED is basically c(t - T/4) - c(t - 3T/4):
```
% Sample indices of the optimal case of being at the symbol center
opt2_idx = (i + [0, -1/4, -1/2, -3/4, -1, -5/4, -3/2, -7/4, -2]) * tea_sps + pf_delay + 1;
d2 = modulated_2(opt2_idx - tau(j));
% Andy Walls test                             
gmsk3_ted_output(i,j) = -real(  d2(2)*d2(2) * conj(d2(6)*d2(6)) ...
                                      -(d2(4)*d2(4) * conj(d2(8)*d2(8))));
```

Now, something that I haven't understood is why Andy needed to add the minus sign to his proposed TED, in comparison to the original TED in the symbol sync block (which is basically c(t) - c(t-T/2) and so has the stable zero in the wrong place):

```
opt_idx = (i + [0, -1/2, -1, -3/2, -2]) * tea_sps + pf_delay + 1;
d = modulated_2(opt_idx - tau(j));
gmsk_ted_output(i,j) = real(  d(1)*d(1) * conj(d(3)*d(3)) ...
                                    -(d(2)*d(2) * conj(d(4)*d(4))));
```

It seems to me that Nick didn't include the minus sign in [his MR](https://github.com/gnuradio/gnuradio/pull/4335/files#diff-dcb7ce218cf462bc8752505ed35918c3df856cc48d39f9ec8af829b507bd4a4dL314). Maybe there is some additional change of sign or sign convention somewhere that I'm missing.

awalls-cx18, 2021-03-18:
On March 18, 2021 1:12:07 PM EDT, "Daniel Estévez" ***@***.***> wrote:
> Hi! I've been taking a look at the [D'Andrea and
> Mengali](https://www.researchgate.net/publication/3153478_A_digital_approach_to_clock_recovery_in_generalized_MSK)
> paper to try to understand all [...]

Daniel,

Nice. :)

The minus sign is a mess from an error I made early on.

I made my TED simulations match textbook curves by reversing the time delay axis, instead of just plotting what was generated by the simulation.  That is a mistake I regret.

My process should have been:
1. Define a TED based of of academic papers or textbooks.
2. Simulate it properly to find the S-curve
3. Revise the TED definition (e.g. tweak the TED input sample offsets by T/4) so the the S curve passes through (0,0) in the simulation output.
4. Use the slope of the Scurve at (0,0) as the TED gain, even if it is negative. (E.g. don't flip the time axis to make it look like the textbook or paper.)

Sorry for the messy email.  I'm on my phone.

PR #4335 (2021-03-10): Fix for #3678
This fixes #3678 (MSK timing error detector) for the "MSK Gen TED". The fix can be cherry-picked onto maint-3.8 and maint-3.9 as it is not an API change.

It does not change the "GMSK TED", which should be deprecated as it is redundant and does not work for its intended application, confusing users.