gnuradio/gnuradio #3679: Stereo audio source not working

doomscar, 2020-08-02:
Audio Source block does not work with more than one output.

> runtime_swig.py", line 3508, in primitive_connect
return _runtime_swig.top_block_sptr_primitive_connect(self, *args)
ValueError: port number 1 exceeds max of 0

A similar problem was described on [reddit](https://www.reddit.com/r/GNURadio/comments/fbejws/stereo_audio_source_not_working/). Is there a solution to the problem?

drmpeg, 2020-08-03:
That rather cryptic error message is telling you that your audio source and mono and you can't create two channels from it.

doomscar, 2020-08-03:
I tried to use a virtual audio cable and a vb-cable as an audio source. The result is the same. Are there examples of using a dual channel audio source? I could not find it.

drmpeg, 2020-08-03:
The Device Name "pulse_monitor" works here for capturing audio that's being played by VLC.

marcusmueller, 2020-08-04:
what's your GNU Radio version, and OS? If it's windows and 3.7, fixed in newer GNU Radio by preferring portaudio.

doomscar, 2020-08-04:
> 
> 
> what's your GNU Radio version, and OS? If it's windows and 3.7, fixed in newer GNU Radio by preferring portaudio.

GNU Radio Companion 3.7.13.5
OS: Windows 7

dkozel, 2020-08-06:
@doomscar Could you please try the newer 3.8 release?
http://www.gcndevelopment.com/gnuradio/downloads/installers/v1.7.0/gnuradio_3.8.1.0_win64.msi

doomscar, 2020-08-12:
installed the latest gnuradio (3.8.1.0_win64.msi) while executing graph python crashes (Windows 7)

> Problem Event Name:   APPCRASH
Application Name: python.exe
Application Version:  0.0.0.0
Application Timestamp:    5d038d9f
Fault Module Name:    gnuradio-runtime.dll
Fault Module Version: 3.8.1.0
Fault Module Timestamp:   5ef363d1
Exception Code:   c000001d

SumiSpam, 2020-09-18:
Hi there!

I had the same problem using GNU Radio Companion 3.7.13.5 on Windows 10.

In my case, the problem was solved creating a user configuration file for changing the default audio to 'portaudio', and putting this file in the GNU Radio user preferences directory.

I followed these steps:

1.- Check the audio used by GNU Radio using the command:

C:\GNURadio-3.7\bin> gnuradio-config-info --prefs

I saw (among other preferences):

[audio]
audio_module = auto
...
...

2.- Obtain the location of the GNU Radio user preferences directory using the command:

C:\GNURadio-3.7\bin> gnuradio-config-info --userprefs

in my case it is:

C:\Users\Usuario\AppData\Roaming\.gnuradio

3.- Go there and create a file named 'config.conf' containing:

[audio]
audio_module = portaudio

4.- Recheck the audio used by GNU Radio (as step 1).

And that's all: after restarting GNU Radio Companion, Audio Source can now use 2 output channels.

Many thanks to the people involved in the discussion "Audio source cannot use 2 outputs (Stereo) in windows":
https://lists.gnu.org/archive/html/discuss-gnuradio/2019-05/msg00022.html

marcusmueller, 2021-03-22:
Solved in later versions of GNU Radio.

closed 2021-03-22 by marcusmueller