gnuradio/gnuradio #3680: Error while installing gnuradio

ik8ozv, 2020-08-02:
Hello all,
i am trying to install gnuradio but when i give the command:

make test

i get a long list of "passed"but at the end i get these errors:

99% tests passed, 1 tests failed out of 365

Total Test time (real) = 176.15 sec

The following tests FAILED:
	 25 - qa_volk_32f_8u_polarbutterflypuppet_32f (Failed)
Errors while running CTest
make: *** [Makefile:130: test] Errore 8

Please how to solve?
Thanks
73's de Enzo IK8OZV

drmpeg, 2020-08-03:
It can be safely ignored. It's just used for Polar error correcting codes, which hardly anyone is using.

marcusmueller, 2020-08-04:
Also, that's a VOLK, not a GNU Radio issue (and I think it's also fixed on current VOLK).

closed 2020-08-04 by marcusmueller