gnuradio/gnuradio #3689: Correlation estimator treats symbols provided in constructor differently from when `set_symbols` is called.

ThomasHabets, 2020-08-12:
In the constructor the passed symbols are [saved as time-reversed conjugate](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/corr_est_cc_impl.cc#L60).

But in `set_symbols()` [it's used as-is](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/corr_est_cc_impl.cc#L116).

I'm not an expert on DSP, but this can't be right, can it?

mormj, 2020-08-12:
At a minimum it is sloppy and confusing.  The underlying filter object stores the conjugate time reversed symbols, because it does straight dot products.  There is a difference there between instantiating the filter and then calling set_taps() could  be the difference.  But I haven't worked through this code yet.  d_symbols appears to only be used to set the filter taps, and is not used in work().

ThomasHabets, 2020-08-12:
But the [filter constructor ](https://github.com/gnuradio/gnuradio/blob/master/gr-filter/lib/fft_filter.cc#L34) just calls `set_taps`. The constructing path reverses, while the `set_symbols()` path, from what I can see, doesn't.

willcode, 2020-10-28:
Agree with @ThomasHabets. This looks wrong, probably isn't used, and there's no QA for it.

Also, symbols() returns d_symbols, which would not match what the caller originally set (or would in the second case).