gnuradio/gnuradio #3698: Multiply Const by 0 has No Effect

Semnodime, 2020-08-13:
# Reproduce
 1. Install GnuRadioCompanion 3.7.9 via official apt repository on Linux Mint 18.3 64bit
 1. Take a signal from file source and multiply it by 0.
 1. Then display the signal.

# Expected
Flatline (Constant signal value 0, both for real part and imaginary part)

# Instead
[image]
[image]
[image]

lanceman2, 2020-08-15:
FYI:  I tried running something like the first model with the source being a Signal Source Cosine on gnuradio-v3.8.1.0 (on unbuntu 20.20) and it made it all 0 plotted out, as is expected with Multiply Const 0.   Without your file for input.
[image]
In the name of debugging I ask: Will this model work for you?  If it does I'd look at your input file more closely.  Are there large numbers in it?  Large float * 0  can maybe be finite????   or NAN somewhere.

I'd recommend stripping down the model until it stops acting strange, and show the minimum "strange model".  Less (not more) is better for debugging.

mormj, 2020-08-17:
3.7.9 is very ancient GR.  Perhaps try installing from the GNU Radio PPA repository to at least get the latest 3.7 (and preferably move up to 3.8 if you can)

https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation

csylvain, 2020-09-27:
the attempt to reproduce this issue (lanceman2) used "Multiply Const" with Real values in/out.
the issue author used "Multiply Const" with Complex values in/out.
does it not weaken the attempt to duplicate/verify the issue by using a different datatype?

willcode, 2020-10-28:
Verified that multiply_const/complex by 0 works correctly with a sig source and a file source. I like the theory that the OP's source file has something in it that makes volk_32fc_s32fc_multiply_32fc() produce non-zero values. The plot scales above (that stop at 10) could be misleading.

Recommend OP provide a small test file that shows this effect, otherwise close.

mormj, 2021-03-24:
Closing this due to lack of response.

closed 2021-03-24 by mormj

csylvain, 2021-03-25:
even without a pathological dataset, should not multiply by 0 always produce 0?

willcode, 2021-03-25:
NaN * 0 = NaN
```
    #include <math.h>
    #include <stdio.h>
     
    int main() {
        float x = 0.0f;
        float y = x * -1;
        float z = x * NAN;
        float a = x * INFINITY;
        float b = y * INFINITY;
        printf("%.08f\n", x);
        printf("%.08f\n", y);
        printf("%.08f\n", z);
        printf("%.08f\n", a);
        printf("%.08f\n", b);
        return 0;
    }
```

csylvain, 2021-03-25:
yes. computer arithmetic != arithmetic

which suggests: fix it in the documentation