gnuradio/gnuradio #3702: Writing a variable in a thread while reading the same variable in another thread without a thread sync

lanceman2, 2020-08-15:
I found two function in GNUradio that run in two different threads, one writing while another is reading the same C++ variable (complex) without any thread synchronization calls like for example (on GNU/Linux Ubuntu 20.20) pthread_mutex_lock() and pthread_mutex_unlock() (or r/w lock).   I confirmed that the variable was indeed accessed by the two threads at the same time by adding pthread_mutex_trylock() and pthread_mutex_unlock() around the two codes that accessed the variable, also printed the address of the variable in both threads with the return status of the pthread_mutex_trylock().   I'm fairly sure that neither of the variable accesses (r/w) were atomic.  I ran a test that proves that the multi-thread r/w access is occurring.  I was looking for an example of the GNUradio "message passing interface", but found this instead.

Q: Is this kind of code a concern?  I know some people that write code, where reading and writing one variable in different threads is not coordinated, and they know it.  They argue that the corruption, if it occurs, is only temporary.   But since the use cast of GNUradio is to be a general toolkit, I'd think that the possibility of memory corruption, or undetermined behavior, was not tolerable.  I think I may find many more occurrences of this in the GNUradio code.

Someone please answer this question.  I do not think this is just a single bug, but is a general design problem and may be the source of many bugs.  I'll elaborate (expand/code/patch/whatever) if there is interest.  I'm not expecting a positive response.

gnuradio-v3.8.1.0

Yamakaja, 2020-08-16:
While i cannot comment on the issue itself, i'd be interested in some examples. (Where exactly did you observe this behavior?)

lanceman2, 2020-08-16:
[image]
You may want to add a "displaying block" in place of the Null Sink, but this is what I thought was the smallest model that showed this "problem".

I added edits to the source in files:
   * gr-blocks/lib/multiply_const_impl.h in
      - function:void set_k(T k), 
   * gr-blocks/lib/multiply_const_impl.cc in
      - function: template <> int multiply_const_impl<gr_complex>::work().

The edits just enclose the body of the two functions with pthread mutex lock and unlock (in addition to printing to stderr), as if we where trying to make "thread-safe" code.  By replacing one of the lock calls (pthread_mutex_lock()) with pthread_mutex_trylock() I can catch and print when the two functions are accessing the variable d_k at the same time, in two different threads.

You can just look at these two functions and see that they are both accessing (one reading and one writing) the d_k variable.  Being template code and not knowing what operators are being rewritten, it is hard to "scope out" how the code runs without inserting addition debugging code; unless you wrote the original code yourself.

So you move the (range) slider and the d_k variable changes while the block is reading that d_k variable.   The value read in the Multiply_Const block (work) can be bobbled, or a mash of the value before the write and the value after the write; not a good value at all.   I don't think the write or the read operations are atomic.  I also know that proving that the reads and writes are not atomic could be very difficult and may be machine/load/system dependent.  I know that the changing of the d_k variable is not synchronous, but **I think** that the values of d_k read **should** be a valid (good) value every time it's read.  Some people may not agree with that statement.

I expected the d_k variable to be set using "Message Passing Interface" in a callback that is called before the "work" function is called (in the scheduler code??).   It would have been the simplest use case of the "Message Passing Interface" that I can think of, and that's why I am studying this simple example.

Oh yes, no throttle block: I was also testing how well it pegged my CPU cores, how well my computer handled the load, and how well GNUradio ran without the throttle (or any throttling-like hardware).  It runs okay without a throttle on my computer, without interfering with other tasks.  There could be use cases of GNUradio programs without any throttling (soft or hard), like batch processing.  By reading the documentation one would think that that use case is discouraged.

drmpeg, 2020-08-16:
Some blocks are protected. Do a grep for gr::thread::scoped_lock.

lanceman2, 2020-08-16:
These functions are very short.  No grep is needed.

drmpeg, 2020-08-16:
I meant do a grep in a directory (for example, gnuradio/gr-blocks/lib) to find where ```gr::thread::scoped_lock``` is used. For example, take a look at:

https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/add_const_ff_impl.cc#L36

I don't know why it's used in some places and not others. Someone with more tribal knowledge may know the answer.

lanceman2, 2020-08-16:
If there is a *gr::\thread::scoped_lock* thingy in this code, my analysis shows it to be not working, or it is not affecting what I'd call "this problem".  So I'd argue that your comment is not relevant to answering my question, but may be a part of the solution, if this is considered a bug.  I'm really looking for a tribal leader to reply.  I see this as a fundamental design/coding error, but I know that some people argue that mis-reading memory is only a temporary glitch in the running program; I'm not one of those people, and so I'm concerned.

My hunch is we need to change the call to set_k(T k) to happen in a "Message Passing Interface" callback function (whatever it's called), in the scheduler (guessing) before the work() function call.   And of course this is done in a more round-about way.  The Multiply_Const block would need to publish a message_port_register_in() thingy.  And the Range needs to use it.  The problem could be more systemic...  Then hunt down all other blocks that have a simple set function like this, and all places where their set functions are used while the stream is flowing.   That could be a lot of coding.  That's my stab at it, I guess, but I have not looked deep enough into it yet.  I'd like to know that there is a point to working on this.

I have no doubt that I can fix this problem, or at least expose the problem better, but I need to know if it is considered a problem, from a tribal leader.  It could be just how things are done in GNUradio.

--lance

drmpeg, 2020-08-16:
I'm just saying that ```gr::thread::scoped_lock``` is the built-in method in GNU Radio to accomplish what you want to do in multiply_const_impl.cc.

lanceman2, 2020-08-17:
Yes.  I think it may be better than "Message Passing Interface"  junk or maybe not.  It would in effect make a thread-safe queue of one for set_k(), but the not so good addition is having mutex lock and unlock in all 3 different work() functions.  I think that using the "Message Passing Interface" (or other like wrapper) is more standard (or should be) and at flow-time there are no changes to the running code, and the set_k() is called before work(), if it is called.  Most of the time set_k() will not be called in a cycle.    It's a toss up for me at this point, but I think the "Message Passing Interface"  solution should be more standard and just a touch more performant, but requires replacing code everywhere set_k() is called.  Adding the scoped_lock (mutex) is much simpler.    I'm not sure what scoped_lock is at this point.  Mutex wrapper??

Here's the thing:  If we make adjustable "parameters" like what set_k() does; with standard interfaces like the "Message Passing Interface" (GNUradio so called) then we can more easily adjust any "parameter" in any block with a generic code.   You'd only have to know the name and "type" of the parameter to control it from other codes; and as a common example IoT (Internet of things).  Otherwise there's not much point to "Message Passing Interface".  All the blocks will make there own "parameter queue thingy" with unique interfaces that make all the other code unique and bloated to use them.  No seamless integration possible.

lanceman2, 2020-08-17:
Looking that the C++ code for block sig_source (gr-analog/lib/sig_source_impl.*) I see all 3 cases for setting "parameters" (variable or what ever they are called):

   1. Setting frequency (freq) uses "Message Passing Interface"
   2. Setting phase uses a simple set function with a gr::\thread::scoped_lock
   3. Setting amplitude, offset, phase, and waveform (and maybe frequency) use "simple set function" **without** a gr::\thread::scoped_lock.

For case 3 if the variable is set (written) at flow-time it's value may be read incorrectly by the working thread, that is assuming that the gr::fxpt_nco object just does math, and does not do any pthread (syncing/mutex/etc) call.  The frequency may not be accessible from the simple set function, don't know yet; don't care yet.  I'd need to test to confirm all that, like in the Multiply_Const in the above comment, but that's my hunch.

This is something anybody looking at the code would know.  For the most part, this comment is just my notes to myself, in case I work on this later.  I don't know what to make of this hodgepodge of "parameter" interfaces.  This shows the "problem" to be more systemic, but does not prove it.

lanceman2, 2020-08-18:
# It's a Design Problem

Please read this reference: [robust code](http://www.linfo.org/robust.html), before reading this comment:

If we wish to make GNUradio a robust framework we need to stop the python code from changing block variables in an "unprotected way" while the stream is flowing.  Ya, it works most of the time, but it's not robust.  The solution should be designed into the block class.  It looks like all the variable setters/getters that I looked at are C++ public methods, so it may be a design problem.  I was thinking that blocks need to limit the access to these methods from outside the block class.  So ya, that is a big design change in the block class.  That's a first stab at what is the most obvious step in making GNUradio a robust framework when considering this issue.  Okay, so that's not going to happen.  Maybe there is a better way.  Anyone, feel free to comment.  Keeping in mind I'm looking for a robust solution, not an ugly hack.

Before I started looking at the GNUradio code I assumed that GNUradio was using the "Massage Passing Interface" (MPI) to set and get all the externally accessible block variables, and so all of the setters and getters were private.   It would have been a solution to the problem that I'm working on.  From what I understand all the setters and getters could be private if all the blocks used MPI for their parameter/variable "setter and getter" external interfaces.  MPI methods are already designed with/into the block class (the blocks I looked at).  The block C++ constructor will still get the initial paramter/variable values.

I don't like the insert mutex locks around all setters, getter, and work functions solution.   I think I made that clear above.  I think there is more benefit to making all the setters and getters private than is seen on first inspection.  Something like forcing that all setters and getters to not be directly accessible will force all blocks to have the more seamless MPI to setting and getting; and that would make all blocks automatically more usable by external codes, in addition to being more performant (assuming that that MPI code is not messed up).

I'm not sure if any of this solution can work, not that any of this is possible either.  Just working notes at this point.

marcusmueller, 2020-08-30:
Hi @lanceman2, thanks for stopping by.

So, as the others explained: yes, there's places in the code where we (erroneously) still have unprotected access to member variables. That is a historical thing - the fact that GNU Radio was multi-threaded is older than a lot of these blocks.

Then: yep, one of the appropriate approaches to dealing with that is using the message-passing interface.

Also, yep: PMT is far from perfect, and also for historical reasons.

We're working on all this. It's by far not as dire as you perceive it; for example, setting a complex factor in a non-atomic operation would, worst case, deliver the real part before the imaginary one; and really, while I agree, this is design-wise a terrible thing and might hide more mistakes, that alone isn't so bad; if that happens while processing is in action, then that was an asynchronous operation to begin with, and thus, no temporal guarantees have been made. There's many ways of breaking a signal, and for the signals we usually deal with, just changing a scaling at an arbitrary point means "I don't care about transient misbehaviour, anyways". 

As you can imagine, GNU Radio is a large project in perpetual motion. We're working on coming up with a more modern approach to the distributed flow graph, coherent heterogeneous platform, and the message passing scalability problems. We're aware of the problems these need to solve.

closed 2020-08-30 by marcusmueller