gnuradio/gnuradio #3703: BlockTool output doesn't distinguish between enums and enum classes

ncorgan, 2020-08-16:
When taking in BlockTool's JSON outputs, there is no way to tell if the enums listed are C-style enums or C++ enum classes. This causes an issue when using these for code generation, as it becomes necessary to check the header, which defeats the purpose.

Examples:

C enum:
```
"enums":[
         {
            "name":"ted_type",
            "values":[
               [
                  "TED_NONE",
                  -1
               ],
               [
                  "TED_MUELLER_AND_MULLER",
                  0
               ],
               [
                  "TED_MOD_MUELLER_AND_MULLER",
                  1
               ],
               [
                  "TED_ZERO_CROSSING",
                  2
               ],
               (TRUNCATED)
            ]
         }
      ],
```
C++ enum class:
```
"enums":[
         {
            "name":"adaptive_algorithm_t",
            "values":[
               [
                  "LMS",
                  0
               ],
               [
                  "NLMS",
                  1
               ],
               [
                  "CMA",
                  2
               ]
            ]
         }
      ],
```

mormj, 2020-08-17:
Unfortunately I'm not seeing any way to get the differentiation from the underlying pygccxml utility that is used to parse the header.  For instance if I make 2 enums:
```c++
enum class adaptive_algorithm_t { LMS, NLMS, CMA };
typedef enum adaptive_algorithm2 { LMS2, NLMS2, CMA2 } adaptive_algorithm2_t;
```
Breaking in `parseheader_generic.py', the output of pygccxml (before it gets put into the json structure) is the same for both enums:
```
<pygccxml.declarations.enumeration.enumeration_t object at 0x7f7f0f1c0af0>
special variables
function variables
attributes:None
byte_align:4.0
byte_size:4.0
cache:<pygccxml.declarations.algorithms_cache.declaration_algs_cache_t object at 0x7f7f0f1c0b20>
decl_string:'::gr::digital::adaptive_algorithm_t'
elaborated_type_specifier:'enum'
is_artificial:False
location:<pygccxml.declarations.location.location_t object at 0x7f7f0f1c0b50>
mangled:None
name:'adaptive_algorithm_t'
parent:<pygccxml.declarations.namespace.namespace_t object at 0x7f7f0f8c9ee0>
partial_decl_string:'::gr::digital::adaptive_algorithm_t'
partial_name:'adaptive_algorithm_t'
top_parent:<pygccxml.declarations.namespace.namespace_t object at 0x7f7f2a7bd0a0>
values:[('LMS', 0), ('NLMS', 1), ('CMA', 2)]
_attributes:None
_byte_align:4.0
_byte_size:4.0
_cache:<pygccxml.declarations.algorithms_cache.declaration_algs_cache_t object at 0x7f7f0f1c0b20>
_decorated_name:None
_elaborated_type_specifier:'enum'
_get__cmp__data:<bound method declaration_t._get__cmp__data of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f7f0f1c0af0>>
_get__cmp__items:<bound method enumeration_t._get__cmp__items of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f7f0f1c0af0>>
_get_name_impl:<bound method declaration_t._get_name_impl of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f7f0f1c0af0>>
_get_partial_name_impl:<bound method declaration_t._get_partial_name_impl of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f7f0f1c0af0>>
_is_artificial:False
_location:<pygccxml.declarations.location.location_t object at 0x7f7f0f1c0b50>
_mangled:None
_name:'adaptive_algorithm_t'
_on_rename:<bound method declaration_t._on_rename of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f7f0f1c0af0>>
_parent:<pygccxml.declarations.namespace.namespace_t object at 0x7f7f0f8c9ee0>
_partial_name:None
_values:[('LMS', 0), ('NLMS', 1), ('CMA', 2)]
_warn_deprecated:<function declaration_t._warn_deprecated at 0x7f7f25c7fa60>
```
and for the second one:
```
<pygccxml.declarations.enumeration.enumeration_t object at 0x7f3c3ac80bb0>
special variables
function variables
attributes:None
byte_align:4.0
byte_size:4.0
cache:<pygccxml.declarations.algorithms_cache.declaration_algs_cache_t object at 0x7f3c3ac80be0>
decl_string:'::gr::digital::adaptive_algorithm2'
elaborated_type_specifier:'enum'
is_artificial:False
location:<pygccxml.declarations.location.location_t object at 0x7f3c3ac80c10>
mangled:None
name:'adaptive_algorithm2'
parent:<pygccxml.declarations.namespace.namespace_t object at 0x7f3c3b389ee0>
partial_decl_string:'::gr::digital::adaptive_algorithm2'
partial_name:'adaptive_algorithm2'
top_parent:<pygccxml.declarations.namespace.namespace_t object at 0x7f3c5629e0a0>
values:[('LMS2', 0), ('NLMS2', 1), ('CMA2', 2)]
_attributes:None
_byte_align:4.0
_byte_size:4.0
_cache:<pygccxml.declarations.algorithms_cache.declaration_algs_cache_t object at 0x7f3c3ac80be0>
_decorated_name:None
_elaborated_type_specifier:'enum'
_get__cmp__data:<bound method declaration_t._get__cmp__data of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f3c3ac80bb0>>
_get__cmp__items:<bound method enumeration_t._get__cmp__items of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f3c3ac80bb0>>
_get_name_impl:<bound method declaration_t._get_name_impl of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f3c3ac80bb0>>
_get_partial_name_impl:<bound method declaration_t._get_partial_name_impl of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f3c3ac80bb0>>
_is_artificial:False
_location:<pygccxml.declarations.location.location_t object at 0x7f3c3ac80c10>
_mangled:None
_name:'adaptive_algorithm2'
_on_rename:<bound method declaration_t._on_rename of <pygccxml.declarations.enumeration.enumeration_t object at 0x7f3c3ac80bb0>>
_parent:<pygccxml.declarations.namespace.namespace_t object at 0x7f3c3b389ee0>
_partial_name:None
_values:[('LMS2', 0), ('NLMS2', 1), ('CMA2', 2)]
_warn_deprecated:<function declaration_t._warn_deprecated at 0x7f3c51761a60>
```

marcusmueller, 2024-01-28:
same assessment. Generally, scoped enum support in pybind11 isn't quite trivial, and chances are one would have to do manual things anyway :/