gnuradio/gnuradio #3704: Generate Error: 'catch_exceptions' >>> Failure

ephraimon, 2020-08-16:
Hi tech friends,
I am working on an fm receiver using gnuradio. When I try to generate the flowgraph, it comes up with this error message: Generating: '/usr/share/gnuradio/examples/uhd/FM_Receiver.py'
Generate Error: 'catch_exceptions'
>>> Failure

I have tried changing the location where the flowgraph was saved see if it's location issue but the same error message occurs. 
Could someone help with suggestions on how to resolve this?

Thanks in advance!

drmpeg, 2020-08-16:
Just try manually deleting the line:

catch_exceptions: 'True'

in your .grc file with an editor.

ephraimon, 2020-08-16:
@drmpeg 

> Just try manually deleting the line:
> 
> catch_exceptions: 'True'
> 
> in your .grc file with an editor.

@drmpeg 
Hi, thanks for your suggestion. I already searched for catch_exceptions in the .grc file but there is no such line.

ephraimon, 2020-08-17:
> @drmpeg
> 
> > Just try manually deleting the line:
> > catch_exceptions: 'True'
> > in your .grc file with an editor.
> 
> @drmpeg
> Hi, thanks for your suggestion. I already searched for catch_exceptions in the .grc file but there is no such line.

@drmpeg 
G, I found the catch_exc... file and removed the line and it generated successfully. It's in a different file named catch_exception.hpp.

Though, it generated the .py file when I executed .grc it seems to run through but will not show the GUI display of the fft/frequency sink. Moreover, when I executed it in the terminal, it comes back with "command not found". Any suggestions on why it could execute in the companion but not in the terminal?

closed 2020-08-18 by ephraimon