gnuradio/gnuradio #3735: GNU Radio doesn't build on OpenBSD

ThomasHabets, 2020-08-26:
First of all [this patch](http://cvsweb.openbsd.org/cgi-bin/cvsweb/ports/comms/gnuradio/patches/patch-gnuradio-runtime_lib_thread_thread_cc?rev=1.1&content-type=text/x-cvsweb-markup) is needed.

Second, the generated linking commands are wrong. They have and extra `lib` in them, like `-llibgnuradio-pmt`. I can see that in the generated `link.txt` files.

My temp workaround while I'm looking at this is to periodically run this while the build is running:

```
for so in $(find . -name "*-git"); do   ln -s $(pwd)/$so $(echo $so | sed -r 's,/libgnuradio-(.*),/liblibgnuradio-\1.so,');done
```

I've installed these dependencies so far, but that doesn't enable all of GNU Radio yet:

```
pkg_add py3-{mako,pip,numpy,pyaml,yaml} gmpxx log4cpp boost fftw3-float doxygen
python3.7 -m pip install pytest
```

And of course github.com/gnuradio/volk and github.com/pybind/pybind11 need to be installed first.

I'm continuing to try to build. We'll see what else I encounter.

ThomasHabets, 2020-08-26:
Ok, putting this on hold a bit, because the old machine I'm running this on only has 4 GiB of RAM, and even with 21 GiB of swap it goes OOM:

```
[ 42%] Linking CXX shared module blocks_python.so
LLVM ERROR: out of memory
c++: error: unable to execute command: Abort trap (core dumped)
c++: error: linker command failed due to signal (use -v to see invocation)
*** Error 254 in . (gr-blocks/python/blocks/bindings/CMakeFiles/blocks_python.dir/build.make:2420 'gr-blocks/python/blocks/bindings/blocks_python.so')
*** Error 2 in . (CMakeFiles/Makefile2:3968 'gr-blocks/python/blocks/bindings/CMakeFiles/blocks_python.dir/all')
*** Error 2 in /home/thompa/src/gnuradio/build (Makefile:141 'all': make -f CMakeFiles/Makefile2 all)
```

drmpeg, 2020-08-26:
Can't you just use less cores? For example,  make -j2 or make -j3.

I build GNU Radio on an ARM platform with 2 GB all the time.

ThomasHabets, 2020-08-27:
Linking `blocks_python.so` seems to be taking several minutes, with no parallelism.

Ah, but seems I forgot that OpenBSD's default has very limited ulimits. I'm bumping that up and trying again.

ThomasHabets, 2020-08-27:
Yeah, that worked. Now to manage to install all dependencies, despite cmake's awfulness and unhelpfullness in explaining *why* it thinks something isn't installed.

ThomasHabets, 2020-08-27:
Ok, I've gotten most of this to compile. The holdout is mostly QT (which includes GRC).

```
pkg_add py3-{mako,pip,numpy,pyaml,yaml,cairo,click,click-plugins,qt5,qtpy} gmpxx log4cpp boost fftw3-float doxygen libusb1 orc sdl qwt gsl
# choose qwt-6.1.3p2-qt5
python3.7 -m pip install pytest pygi
```

```
-- ######################################################
-- # Gnuradio disabled components                         
-- ######################################################                                                                                   
--   * gnuradio-companion                                                                                                                   
--   * gr-qtgui              
--   * gr-zeromq
```

ThomasHabets, 2020-08-27:
Failed trying to link with `-lUHD::UHD`. I symlinked `ln -s libuhd.so.4.0.0 libUHD::UHD.so` and crossing fingers.

nickoe, 2020-12-29:
@ThomasHabets Any update on this? Otherwise I think we should just close this as it mostly look like you are debugging platform specific issues.

ThomasHabets, 2020-12-30:
I gave up for now. But it's still a bug.

You had *one* job, CMake…

nickoe, 2021-01-02:
@ThomasHabets I just tried to setup an openbsd in a qemu environment. I manged to install some dependencies. I got pybind11 via pip, but when I try to build volk:
```
[ 76%] Building C object cpu_features/CMakeFiles/cpu_features.dir/src/cpuinfo_x86.c.o
/home/nickoe/volk/cpu_features/src/cpuinfo_x86.c:110:2: error: "Unsupported OS"
#error "Unsupported OS"
 ^
/home/nickoe/volk/cpu_features/src/cpuinfo_x86.c:1268:2: error: "Unsupported fallback detection of SSE OS support."
#error "Unsupported fallback detection of SSE OS support."
 ^
2 errors generated.
*** Error 1 in . (cpu_features/CMakeFiles/cpu_features.dir/build.make:83 'cpu_features/CMakeFiles/cpu_features.dir/src/cpuinfo_x86.c.o': cd ...)
*** Error 2 in . (CMakeFiles/Makefile2:273 'cpu_features/CMakeFiles/cpu_features.dir/all')
*** Error 2 in /home/nickoe/volk/build (Makefile:161 'all': make -s -f CMakeFiles/Makefile2 all)
```

Are you running on real hardware?

Also, it is probably easier to see what exactly cmake generates for the compiler by running make with the "VERBOSE=1" option, as `make VERBOSE=1`.

ThomasHabets, 2021-01-02:
I have tested on real HW and on a VM. (though not for a while)

I don't think we should close this, since shouldn't GNU Radio be portable? It's still not working.

nickoe, 2021-01-02:
@ThomasHabets If no one is actively working on this issue I think we should close it. Gnuradio don't state that it supports openbsd. The information here is of course still of value, but it will not disappear if closed anyways.

If someone do get it working and have proper fixes for them, they are better handled in a pull request.

closed 2021-01-02 by nickoe