gnuradio/gnuradio #3738: qt gui waterfall default colors oddity

ZeroChaos-, 2020-08-27:
grc file lists color1-color10 as "0", gnuradio-companion says "line 1 color" "multi-color", flowgraph runs and has the appearance of "white-hot".

```
16:36 < Zero_Chaos> any ideas why my qt gui waterfall sink is black and white instead of a rainbow of color?
16:40 < ~bpadalino> must be missing the color burst crystal
16:41 < ~funkylab[m|gr]> It's listening to a lot of late 80s goth metal?
16:41 < ~funkylab[m|gr]> (Middle click, color scheme)
16:43 < Zero_Chaos> kinda not kidding though, any thoughts?  like it seems to be "working" but it's all black and white
16:46 < ~funkylab[m|gr]> As said middle click, select a different color scheme :)
16:48 < Zero_Chaos> funkylab[m|gr]: sorry, missed that the first time.  works, but the same appears to be set in the flow graph but it starts in black and white
16:48 < Zero_Chaos> I'm getting "white-hot" by default but the flow graph clearly says multi-color
16:49 < ~funkylab[m|gr]> Huh, that I'd consider a bug
16:49 < ~funkylab[m|gr]> 3.8.2.0 or which?
16:49 < Zero_Chaos> 3.8.2.0
16:49 < Zero_Chaos> "line 1 color" "multi color"
16:49 < Zero_Chaos> "line 1 alpha" "1.0"
16:50 < Zero_Chaos> but it starts up as what appears to be "white-hot"
17:01 < ~funkylab[m|gr]> hm, does the python code reflect that choice? Does the flow graph file?
17:06 < Zero_Chaos> funkylab[m|gr]: I don't see a "line 1 color" or any of the names in the flowgraph grc file at all
17:06 < Zero_Chaos> funkylab[m|gr]: and all the colors are set to 0
17:07 < Zero_Chaos> color1 through color10
17:08 < ~funkylab[m|gr]> huh
17:08 < ~funkylab[m|gr]> that'd be the default
17:08 < ~funkylab[m|gr]> IIRC
17:08 < ~funkylab[m|gr]> will have to investigate later on
```

ei8fdb, 2021-03-22:
@ZeroChaos- is it possible to post a video screengrab to show what you're seeing when you try this?

marcusmueller, 2024-01-28:
https://gist.github.com/marcusmueller/db78d75406202a213043418ea830f145
[image]
can't reproduce; in default, no color shceme is selected in the middle mouse button dialog, and the selection seems to work correctly.

closed 2024-01-28 by marcusmueller