gnuradio/gnuradio #3740: Function Probe does not keep real time

MustBeArt, 2020-08-29:
The Function Probe block (grc/blocks/variable_function_probe.block.yml) is supposed to poll a function in another block periodically. It does so by calling the host computer's time.sleep() function, so it keeps wall clock time instead of sample clock time.

This is a big problem when running a simulation that runs much slower than real time. The polls come much too infrequently, causing unpredictable jumps in the controlled variable. It's a lesser problem even in a realtime flow graph, because the scheduler might process samples in chunks in any order, causing jitter with respect to wall clock time.

Example use case: in a satellite simulation, we want to use variables to control parameters such as Doppler shift, antenna pointing error, etc. These are continuously varying values, which we can simulate adequately by updating them often enough. With functions polls that don't come as specified in simulated time, the variables may move in big jumps instead of continuously.

If the variable in question is a frequency, a VCO block can be used as a workaround. Other variables may not have equivalent blocks to use.

If it's too hard to solve this problem within the current architecture, at the very least it should be prominently explained in the block's documentation.

jacobgilbert-snl, 2020-08-29:
This behavior is intentional. The purpose of this block is to tell a process how many samples are being processed in a wall clock second. It’s usefulness in simulation is limited.

MustBeArt, 2020-09-03:
It seems like you're thinking of some other block. Here's what the Wiki
says about Function Probe:

Periodically probe a function and set its value to this variable. To poll a
> stream for a level, use this with the Probe Signal
> <https://wiki.gnuradio.org/index.php/Probe_Signal> block.

I want to use this block with the Probe Signal block as documented.

  -Paul

jacobgilbert-snl, 2020-09-03:
My apologies about the confusion. I had indeed conflated the function probe and probe rate blocks, and had intended to reply again after I deleted that post, but forgot.

The first statement stands though, the blocks are intended to operate on wall clock time, and the probe rate is a good example of a use case of the general probe functionality.

willcode, 2020-11-01:
@MustBeArt would it be easier to decimate, getting values every N samples (e.g., Single Pole IIR followed by Keep 1 in N), instead of trying to probe by clock time? If not, can you add more of a description of what the feature request would be?

MustBeArt, 2020-11-01:
@willcode If I understand your suggestion correctly, I think it might be possible to implement the kind of thing I want that way, but I don't like the consequences for encapsulation.

The stream-oriented sample-processing code, which I think of as being on the main processing flow, needs to be written cleanly and efficiently to take care of its main business of processing the signal. In doing so, it is keeping track of certain state variables that it needs, in the way it needs to, and at the rate it needs to. In order to make aspects of that internal state available to other blocks, it can provide a sort of API by implementing probe functions. It should not need to worry about when those functions are called. It definitely should not change its internal architecture (decimation, chunk sizes, etc.) in response to the needs of other blocks.

Likewise, the external block that wants to monitor something about that block should not need to know much about its internal architecture. It should be able to refer to the published API of probe functions, and be free to call the ones it needs at the rate it needs them. Commonly, that would be at a much lower rate than anything the signal processing block cares about. Perhaps most commonly of all, there would be no monitoring block at all, and in that case there should be no overhead beyond the dead code of the probe function(s).

That's what I expected Probe Functions to do: provide a mostly-decoupled way to sneak a peek at something a block is doing, at low cost and without disrupting that block's operation.

Then, in order for it to be useful in the simulation case, it needs to do it in simulated time, and not in wall time.

Is that any clearer?

willcode, 2020-11-01:
For simulation, would there need to be a periodic callback? A latch at sample N? Stop the entire flowgraph at sample N? GR wasn't designed to do a clean halt, certainly not with sample accuracy. It's more designed to keep going as fast as possible.

Hopefully, we'll have easier state access in 4.0 that doesn't require a lot of manual setters/getters, latching and locking code in the block.

willcode, 2020-11-01:
Updated https://wiki.gnuradio.org/index.php/Function_Probe to make clear that probe interval is based on the computer clock.

mrjacobagilbert, 2020-11-01:
@MustBeArt I think you have a good use case here but it's not what the current probe blocks are designed or capable of doing. The API you are referencing exists already and it's in the public headers for every in and out of tree block, and you can call these methods as needed. The probe blocks do that in one way (every `x` milliseconds of system time) but there is no restriction on doing so other ways. 

That said, I think you should consider that in general, sample deterministic debug or state information is really something you most likely need to generate from inside your block and manage according to the runtime API. You could emit messages with this information or

mormj, 2021-05-15:
Re-closing this issue, as the usage of Function Probe has been made more clear in the documentation now.  @MustBeArt - please open a new issue with a feature request if your specific use case is desired functionality.

closed 2021-05-15 by mormj