gnuradio/gnuradio #3741: blocks: correctiq_auto's frequency and gain behavior is strange

ThomasHabets, 2020-08-29:
Neither `d_gain` nor `d_freq` is ever read, and the arguments to `set_freq` and `set_gain` are dropped.

What's the intention here?

See [code](https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/correctiq_auto_impl.cc#L144).

dkozel, 2020-08-29:
@ghostop14 can you shed some light on the desired behaviors here?

ghostop14, 2020-09-01:
So for the auto, whenever the frequency or gain changes, the offsets become invalid and need to be recalculated.  Both call trigger_resync() which zero's them out and forces the block to calculate the correct new offset.

    d_synchronized = false;
    d_sync_counter = 0;

    // reset counters on an unsync
    d_avg_real = 0.0;
    d_avg_img = 0.0;

ThomasHabets, 2020-09-02:
@ghostop14 but why are `d_gain` and `d_freq` even passed in as parameters, and stored? You can fetch them with `get_freq()` and `get_gain()`, but curiously they are **not** updated by `set_freq()` and `set_gain()`.

ghostop14, 2020-09-02:
The setters should be called / configured in the yaml files.  I'll have to look to make sure they are in the GR committed versions.  They're required to asynchronously inform the block the variables have changed, so they're just for tracking purposes.

ThomasHabets, 2020-09-02:
The setters may be called, but the setters don't set the member variables. And the rest of the code doesn't use them either. So maybe freq and gain should just be removed completely?

marcusmueller, 2024-01-28:
Indeed, lingers around