gnuradio/gnuradio #3748: Using forbidden id breaks gnuradio-companion

ChristopheSeg, 2020-09-01:
Hi

This was originally posted on discuss-gnuradio@gnu.org. Marcus suggested that it could be categorized somewhere between feature request and bug report and that it could be raised as an issue on Github.

The problem: 
One of my flowgraph was generating the error:

```
self.audio_sink_0 = audio.sink(audio_rate, '', True)
    AttributeError: 'int' object has no attribute 'sink'
```

This error was not easy to debug since the error message sounds like there was a library error (which was not the case here).
My mistake was a "QT GUI Chooser" whose id was (incorrectly) set to audio; "audio" being also the name of a gnuradio python module.
Now that I found my mistake, it's obvious that no ID should be set to some python module name.

May be it could be interesting that gnuradio-companion tests block ids and rejects forbidden ones when compiling flowgraph.

I include in this mail
- Jeff answer to my email 
- Kevin answer including a sample code which illustrating the problem
- Marcus  answer
- a simple .grc and.py files illustrating the error
[test.grc.txt](https://github.com/gnuradio/gnuradio/files/5154472/test.grc.txt)
[test.py.txt](https://github.com/gnuradio/gnuradio/files/5154473/test.py.txt)

Regards

**On Mon, Aug 31, 2020 at 8:11 AM Jeff Long wrote:**

Since there is no way of knowing what global names will conflict in the future, it could be useful for GRC to prefix ids (id_audio), make them part of another structure (ids.audio), or something similar. This adds complexity to manual coding in Python, so it's a tradeoff.

**On 31/08/2020 17:45, Kevin Reid wrote:**

It would be sufficient to modify the constructor code generation so that variables aren't used — to use a non-conflicting example from a random generated .py file I have,
```
    self.samp_rate = samp_rate = 44100
    ...
    self.audio_sink_0 = audio.sink(samp_rate, "", True)
```
would be instead generated without a local variable as
```
    self.samp_rate = 44100
    ...
    self.audio_sink_0 = audio.sink(self.samp_rate, "", True)
```
If another case arises where that's not sufficient, I think it would still be a better idea to have GRC detect and alter Python variable names (or, possibly preferably, the names of module imports) on a case-by-case basis to avoid worsening readability (and edit distance from hand-maintained GR code).

**Error raised by the erroneous code:** 
```
    self.audio_sink_0 = audio.sink(samp_rate, '', True)
    AttributeError: 'int' object has no attribute 'sink'
```
**Christophe note:** using code like 
`self.audio_sink_0 = audio.sink(self.samp_rate, '', True)`

is not enough to solve the issue, since in this flowgraph, audio class has been replaced by a integer variable.  

**On 31/08/2020 18:20, Marcus Müller wrote:**
I'd categorize this somewhere between feature request and bug report.
Could you raise this as an issue on github.com/gnuradio/gnuradio/issues , please?

Generally, tracking could be possible, but is rather complicated, because we have different namespaces in which parts of the flow graph generation happen. Hence, I'd like this to be tracked in the issue tracker. Thanks!
   Best,
   Marcus

dkozel, 2020-09-01:
This is a great report and good find. Thanks for taking the time to write it up so well.

I strongly believe that GRC should allow any (reasonable) id to be entered and should generate code that avoids the name clash seen here.

skoslowski, 2020-09-02:
I agree with @dkozel here. GRC should not change IDs of blocks. For users that import/run the generated flowgraph class in their application, its important to get predictable names - even for blocks. 

I don't see this as a bug - it behaves like when code is written by hand... However, it would be a great new feature!

GRC would have to parse the rendered import statements (of all blocks - we recently added a method generating those to the Block class) to get a list of imported names. This would also allow us to remove the list of hard-coded forbidden names we currently have.

dkozel, 2020-09-02:
Ah, I meant something else originally, that GRC should munge the user supplied ID such that the generated Python would be valid even if a supplied ID was `analog` for instance. But now I realize that would probably be more confusing to users than displaying a really good error message. Something like:

> Block ID `analog` conflicts with Python module `analog`. Unable to generate flowgraph code. Change block ID to resolve the conflict.

paskalem, 2020-09-06:
gnuradio
/
gnuradio
Code
Issues
451
Pull requests
84
Actions
More
New issue
Jump to bottom
GRC generates invalid code when block IDs clash with Python module names #3748
 Open
ChristopheSeg opened this issue 5 days ago · 3 comments
Labels
GRC
enhancement
Comments

@ChristopheSeg ChristopheSeg commented 5 days ago
Hi

This was originally posted on discuss-gnuradio@gnu.org. Marcus suggested that it could be categorized somewhere between feature request and bug report and that it could be raised as an issue on Github.

The problem:
One of my flowgraph was generating the error:

self.audio_sink_0 = audio.sink(audio_rate, '', True)
    AttributeError: 'int' object has no attribute 'sink'
This error was not easy to debug since the error message sounds like there was a library error (which was not the case here).
My mistake was a "QT GUI Chooser" whose id was (incorrectly) set to audio; "audio" being also the name of a gnuradio python module.
Now that I found my mistake, it's obvious that no ID should be set to some python module name.

May be it could be interesting that gnuradio-companion tests block ids and rejects forbidden ones when compiling flowgraph.

I include in this mail

Jeff answer to my email
Kevin answer including a sample code which illustrating the problem
Marcus answer
a simple .grc and.py files illustrating the error
test.grc.txt
test.py.txt
Regards

On Mon, Aug 31, 2020 at 8:11 AM Jeff Long wrote:

Since there is no way of knowing what global names will conflict in the future, it could be useful for GRC to prefix ids (id_audio), make them part of another structure (ids.audio), or something similar. This adds complexity to manual coding in Python, so it's a tradeoff.

On 31/08/2020 17:45, Kevin Reid wrote:

It would be sufficient to modify the constructor code generation so that variables aren't used — to use a non-conflicting example from a random generated .py file I have,

    self.samp_rate = samp_rate = 44100
    ...
    self.audio_sink_0 = audio.sink(samp_rate, "", True)
would be instead generated without a local variable as

    self.samp_rate = 44100
    ...
    self.audio_sink_0 = audio.sink(self.samp_rate, "", True)
If another case arises where that's not sufficient, I think it would still be a better idea to have GRC detect and alter Python variable names (or, possibly preferably, the names of module imports) on a case-by-case basis to avoid worsening readability (and edit distance from hand-maintained GR code).

Error raised by the erroneous code:

    self.audio_sink_0 = audio.sink(samp_rate, '', True)
    AttributeError: 'int' object has no attribute 'sink'
Christophe note: using code like
self.audio_sink_0 = audio.sink(self.samp_rate, '', True)

is not enough to solve the issue, since in this flowgraph, audio class has been replaced by a integer variable.

On 31/08/2020 18:20, Marcus Müller wrote:
I'd categorize this somewhere between feature request and bug report.
Could you raise this as an issue on github.com/gnuradio/gnuradio/issues , please?

Generally, tracking could be possible, but is rather complicated, because we have different namespaces in which parts of the flow graph generation happen. Hence, I'd like this to be tracked in the issue tracker. Thanks!
Best,
Marcus
 
Contributor
@dkozel dkozel commented 5 days ago
This is a great report and good find. Thanks for taking the time to write it up so well.

I strongly believe that GRC should allow any (reasonable) id to be entered and should generate code that avoids the name clash seen here.
 @dkozel dkozel added Bug  GRC  labels 5 days ago
@dkozel dkozel changed the title Using forbidden id breaks gnuradio-companion GRC generates invalid code when block IDs clash with Python module names 5 days ago

Member
@skoslowski skoslowski commented 4 days ago
I agree with @dkozel here. GRC should not change IDs of blocks. For users that import/run the generated flowgraph class in their application, its important to get predictable names - even for blocks.

I don't see this as a bug - it behaves like when code is written by hand... However, it would be a great new feature!

GRC would have to parse the rendered import statements (of all blocks - we recently added a method generating those to the Block class) to get a list of imported names. This would also allow us to remove the list of hard-coded forbidden names we currently have.
 
Contributor
@dkozel dkozel commented 4 days ago
Ah, I meant something else originally, that GRC should munge the user supplied ID such that the generated Python would be valid even if a supplied ID was analog for instance. But now I realize that would probably be more confusing to users than displaying a really good error message. Something like:

Block ID analog conflicts with Python module analog. Unable to generate flowgraph code. Change block ID to resolve the conflict.
 @dkozel dkozel added enhancement  and removed Bug  labels 4 days ago

ei8fdb, 2021-03-22:
> Block ID `analog` conflicts with Python module `analog`. Unable to generate flowgraph code. Change block ID to resolve the conflict.

How likely is it that a block name would conflict with a Python module name?

mormj, 2021-05-15:
This appears to be fixed for the most part in: https://github.com/gnuradio/gnuradio/pull/4522 - the GRC block shows red when the block id clashes with any imported module.  @ChristopheSeg - can you verify that this fixes the issue (there still may be some lingering issues with embedded python modules and this blacklist)

dkozel, 2021-09-29:
The fix should solve this problem. There will probably be other keywords that could cause the same type of issue, but closing this as the specific version of the clash is solved, reducing the likelihood of an issue.

closed 2021-09-29 by dkozel