gnuradio/gnuradio #3749: pybind11: `gr_modtool bind` does not correctly accept regex as advertised for block names

jacobgilbert-snl, 2020-09-02:
This may be user error, but based on the prompt I would have expected this to work...

```bash
jacob@ubuntu:/opt/gnuradio-master/src/gr-pdu_utils$ gr_modtool bind
GNU Radio module name identified: pdu_utils
Which blocks do you want to parse? (Regex): pdu.*
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/opt/gnuradio-master/bin/gr_modtool", line 18, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/opt/gnuradio-master/lib/python3/dist-packages/gnuradio/modtool/cli/base.py", line 133, in wrapper
    return func(*args, **kwargs)
  File "/opt/gnuradio-master/lib/python3/dist-packages/gnuradio/modtool/cli/bind.py", line 42, in cli
    run(self)
  File "/opt/gnuradio-master/lib/python3/dist-packages/gnuradio/modtool/cli/base.py", line 152, in run
    module.run()
  File "/opt/gnuradio-master/lib/python3/dist-packages/gnuradio/modtool/core/bind.py", line 59, in run
    file_to_process = os.path.join(self.dir, self.info['includedir'], self.info['blockname'] + '.h')
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```

Here's an one-liner to bind all of the C++ blocks:
```bash
grep -r "virtual public gr::.*block" include/ | sed -e 's/.*_API //' -e 's/ : virtual.*//' | xargs -tL1 gr_modtool bind
```

mrjacobagilbert, 2020-09-06:
This appears to be an issue for other aspects of modtool; e.g.:

```bash
jacob@ubuntu:/opt/gnuradio-master/src/gr-pdu_utils$ gr_modtool rm
GNU Radio module name identified: pdu_utils
Which blocks do you want to delete? (Regex): pdu_set.*
Searching for matching files in lib/:
Really delete lib/pdu_set_m_impl.cc? [Y/n/a/q]: n
Really delete lib/pdu_set_m_impl.h? [Y/n/a/q]: n
Searching for matching files in include/pdu_utils/:
Really delete include/pdu_utils/pdu_set_m.h? [Y/n/a/q]: n
Searching for matching files in python/:
Really delete python/qa_pdu_set_m.py? [Y/n/a/q]: n
Searching for matching files in python/bindings/:
Really delete python/bindings/pdu_set_m_python.cc? [Y/n/a/q]: n
Searching for matching files in python/bindings/docstrings/:
Really delete python/bindings/docstrings/pdu_set_m_pydoc_template.h? [Y/n/a/q]: n
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/opt/gnuradio-master/bin/gr_modtool", line 18, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/opt/gnuradio-master/lib/python3/dist-packages/gnuradio/modtool/cli/base.py", line 133, in wrapper
    return func(*args, **kwargs)
  File "/opt/gnuradio-master/lib/python3/dist-packages/gnuradio/modtool/cli/rm.py", line 28, in cli
    run(self)
  File "/opt/gnuradio-master/lib/python3/dist-packages/gnuradio/modtool/cli/base.py", line 152, in run
    module.run()
  File "/opt/gnuradio-master/lib/python3/dist-packages/gnuradio/modtool/core/rm.py", line 110, in run
    'void bind_' + self.info['blockname'] + '(py::module& m);')
TypeError: can only concatenate str (not "NoneType") to str
```

If the argument is provided directly this is not an issue (as above w/ `bind`):
```bash
jacob@ubuntu:/opt/gnuradio-master/src/gr-pdu_utils$ gr_modtool rm pdu_set_m
GNU Radio module name identified: pdu_utils
Searching for matching files in lib/:
Really delete lib/pdu_set_m_impl.cc? [Y/n/a/q]: n
Really delete lib/pdu_set_m_impl.h? [Y/n/a/q]: n
Searching for matching files in include/pdu_utils/:
Really delete include/pdu_utils/pdu_set_m.h? [Y/n/a/q]: n
Searching for matching files in python/:
Really delete python/qa_pdu_set_m.py? [Y/n/a/q]: n
Searching for matching files in python/bindings/:
Really delete python/bindings/pdu_set_m_python.cc? [Y/n/a/q]: n
Searching for matching files in python/bindings/docstrings/:
Really delete python/bindings/docstrings/pdu_set_m_pydoc_template.h? [Y/n/a/q]: n
Searching for matching files in grc/:
Really delete grc/pdu_utils_pdu_set_m.block.yml? [Y/n/a/q]: n
```

ewxl, 2021-02-18:
I can confirm this problem, works when supplying block name on argument but not when answering question.

marcusmueller, 2024-01-28:
Need to be reproduced, and probably be part of the CI

mbr0wn, 2024-03-25:
This might be due to the CLI enhancement package we're using (click).

marcusmueller, 2024-06-17:
needs to be re-tested with my work on C++ header parsing and CLI batching, so QA branch rebased on these/main

marcusmueller, 2024-06-24:
can't reproduce; closing.

closed 2024-06-24 by marcusmueller