gnuradio/gnuradio #3756: OOT library symlink not updated always

Tom-McDermott, 2020-09-07:
If an OOT is rebuilt for a new version (example 3.8.1 --> 3.8.2) the new OOT library is
installed in /usr/local/lib/x86_64-linux-gnu but the symlink is not updated and remains pointing at
the old OOT version.  If there is no old OOT version then the correct symlink to the newly built OOT is created.

In this example. libgnuradio-hpsdr.so.1.0.0git remained pointing to the old version even as the new version was installed.

drwxr-xr-x 2 root root    4096 Sep  7 09:52 .
drwxr-xr-x 7 root root    4096 Apr 30 00:33 ..
lrwxrwxrwx 1 root root      29 Sep  7 09:52 libgnuradio-hpsdr.so -> libgnuradio-hpsdr.so.1.0.0git
lrwxrwxrwx 1 root root      29 Sep  7 09:52 libgnuradio-hpsdr.so.1.0.0git -> libgnuradio-hpsdr.so.c439b330
-rw-r--r-- 1 root root 1432704 Sep  7 08:01 libgnuradio-hpsdr.so.c439b330

dkozel, 2020-09-10:
Hi @Tom-McDermott Thanks for the report.
For the convenience of anyone investigating, here is the OOT that was used in the above test.

https://github.com/Tom-McDermott/gr-hpsdr/tree/gr_3.8

marcusmueller, 2024-01-28:
sadly, can't reproduce; my `ninja install` does overwrite!

closed 2024-01-28 by marcusmueller