gnuradio/gnuradio #3757: Disallow copy broke MSVC/clang-cl

gvanem, 2020-09-09:
The change in https://github.com/gnuradio/gnuradio/pull/3619 has broken MSVC + clang-cl build for me. From clang-cl while compiling e.g. `filterbank.cc` and others:
```c
In file included from f:\gv\VC_2019\VC\Tools\MSVC\14.27.29110\include\string:11:
In file included from f:\gv\VC_2019\VC\Tools\MSVC\14.27.29110\include\xstring:14:
f:\gv\VC_2019\VC\Tools\MSVC\14.27.29110\include\xmemory(694,76): error: call to deleted constructor of
      'gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float>'
        ::new (const_cast<void*>(static_cast<const volatile void*>(_Ptr))) _Objty(_STD forward<_Types>(_Args)...);
                                                                           ^      ~~~~~~~~~~~~~~~~~~~~~~~~~~~
f:\gv\VC_2019\VC\Tools\MSVC\14.27.29110\include\xmemory(1508,35): note: in instantiation of function template
      specialization 'std::_Default_allocator_traits<std::allocator<gr::filter::kernel::fir_filter<std::complex<float>,
      std::complex<float>, float> > >::construct<gr::filter::kernel::fir_filter<std::complex<float>,
      std::complex<float>, float>, gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float> &>'
      requested here
        allocator_traits<_Alloc>::construct(_Al, _Unfancy(_Last), _STD forward<_Types>(_Vals)...);
                                  ^
...
F:/gv/dx-radio/gnuradio/gv-build/include\gnuradio/filter/fir_filter.h(41,5): note: 'fir_filter' has been explicitly
      marked deleted here
    fir_filter(const fir_filter&) = delete;
    ^
```
In my lack of C++ knowledge, I fail to understand what the problem is. It seems to work with GNU-C++ since this change has been there for almost 4 months.

But trying a:
```c
    fir_filter (const fir_filter &);
    fir_filter (fir_filter &&) = default;
    fir_filter & operator = (const fir_filter &);
    fir_filter & operator = (fir_filter &&) = default;

```

causes 6 unresolved symbols during link.

ThomasHabets, 2020-09-09:
Strange.

To make it compile at least you should replace "= delete" with "=default". But **the code will be broken**.
Disallowing copy was required before (iow: copying this object created undefined behavior).

I've not coded on Windows in 20 years. Could you explain what "MSVC/clang-cl" means?

And are you sure you've put the whole error message here? I feel like what's missing is the place where a container or something is created. I only see system headers and where the copy constructor is deleted.

gvanem, 2020-09-09:
>  you should replace "= delete" with "=default".

Could you explain what *delete* and *default* does? Did you mean I should try to replace all 3 *delete* with *default*?
Doing that, it compiles and `gnuradio-filter.dll`  links.

>  Could you explain what "MSVC/clang-cl" means?

*Microsoft Visual C* and *clang-cl* is the MSVC-lookalike version from the [LLVM project](http://llvm.org/).
I.e. `clang-cl == clang -driver=MSVC`. https://clang.llvm.org/docs/MSVCCompatibility.html

*BTW*. Clang seems to have an option `-stdlib=libc++` to use it's own C++ library. I've not tried it, so I'm not sure if it will fix this issue. If STL is a standard, you seems to have broken that standard in this case (?).

> And are you sure you've put the whole error message here?

No, just the gist of it (I think). But [attached](https://github.com/gnuradio/gnuradio/files/5193727/clang-cl-error.txt) is the whole ugly error-message from clang-cl while compiling `filterbank.cc`.

ThomasHabets, 2020-09-09:
> `delete` vs `default`

`delete` will at compile time reject any default implementation being created. Deleting the copy constructor does exactly what it sounds like.

`default` will explicitly use the default version of copy/move constructor/assignment. The default is to copy (or move, respectively) every member variable independently.

Actually I was incorrect in saying that changing `=delete` to `=default` here would introduce a bug. It's wrong and risks huge performance problems, but is not unsafe.

Reading the attached error messages I don't understand why your compiler/STL would want to invoke copy constructor/assignment. Is it compiling in C++11 mode?

With the same compiler and options, can you compile:

```c++
#include<memory>
#include<vector>
int main()
{
  std::vector<std::unique_ptr<int>> v;
  v.emplace_back(new int);
}
```

I know it's arrogant to assume the compiler is broken, so maybe I'm just blind here.

ThomasHabets, 2020-09-09:
Could you try adding the option `-std=c++11` to the compiler? I'm thinking that your compiler is automatically accepting C++11, but it's not properly informing the third party (MSVC) STL that it too should be doing C++11.

gvanem, 2020-09-09:
> With the same compiler and options, can you compile:

Yes, no problem.

> Could you try adding the option -std=c++11 to the compiler?

With clang-cl, `error: unknown argument: '-std:c++11'`
With MSVC's cl: `ignoring unknown option '-std:c++11'`

And the same error as before.

ThomasHabets, 2020-09-09:
Did you use `-std=c++11` or `-std:c++11`?

gvanem, 2020-09-09:
I tried both forms. And also `-clang=-std=c++11`. All ignored.

cmrincon, 2020-09-09:
Maybe i didn't undertood the problem, but in msvc you should use /std
[https://docs.microsoft.com/en-us/cpp/build/reference/std-specify-language-standard-version?view=vs-2019](url)

ThomasHabets, 2020-09-09:
Strange. I still think the problem is somewhere in your setup, not with the code.

Copy cons/assign should not be needed here. And definitely should not be called. I'd look in `f:\gv\VC_2019\VC\Tools\MSVC\14.27.29110\include\xmemory`, but these can be hard to read. :-)

Maybe you could add:

```
filterbank(const filterbank&) = delete;
filterbank& operator=(const filterbank&) = delete;
```

to `filterbank.h` and see if it changes anything?

gvanem, 2020-09-09:
That page does not mention `/std:c++11`. And using `/std:c++XY` or `-std:c++XY` is the same (MSVC is more liberal than gcc here).

gvanem, 2020-09-09:
> filterbank(const filterbank&) = delete;
> filterbank& operator=(const filterbank&) = delete;

That helped until `mmse_fir_interpolator_cc.cc` was compiled. Then the same error:
```
f:\gv\VC_2019\VC\Tools\MSVC\14.27.29110\include\xmemory(694,76): error: call to deleted constructor of
      'gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float>'
        ::new (const_cast<void*>(static_cast<const volatile void*>(_Ptr))) _Objty(_STD forward<_Types>(_Args)...);
                                                                           ^      ~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

ThomasHabets, 2020-09-09:
That is strange indeed. Why would your compiler generate default cons/assign when a member function has them deleted? This is making no sense.

Does this compile for you with the same flags?

```c++
#include<vector>

class Uncopiable {
 public:
  Uncopiable(){}

  // Move yes.                                                                                                                              
  Uncopiable(Uncopiable&&) = default;
  Uncopiable&operator=(Uncopiable&&) = default;

  // Copy no.                                                                                                                               
  Uncopiable(const Uncopiable&) = delete;
  Uncopiable&operator=(const Uncopiable&) = delete;
};

class Foo {
 public:
  Foo(){}
  ~Foo(){}
  Uncopiable nocopy;
  std::vector<Uncopiable> nocopy_vector;
};
int main()
{
  Foo foo;
  foo.nocopy_vector.emplace_back();
}

```

For me with clang 9 this even gives a very clear message if one tries to copy construct `foo` by adding `Foo bar(foo)`:

```
tt.cc:29:7: error: call to implicitly-deleted copy constructor of 'Foo'
  Foo bar(foo);
      ^   ~~~
tt.cc:22:14: note: copy constructor of 'Foo' is implicitly deleted because field 'nocopy' has a deleted copy constructor
  Uncopiable nocopy;
             ^
tt.cc:12:3: note: 'Uncopiable' has been explicitly marked deleted here
  Uncopiable(const Uncopiable&) = delete;
```

(hat's off to clang. That's an amazingly clear error message)

gvanem, 2020-09-09:
With the same `CFLAGS`, it compiles and links fine with both MSVC and clang-cl.

But I fail to see the point you're trying to make. Since I'm that into C++.

ThomasHabets, 2020-09-09:
I dunno. This sounds like either a problem with your environment, or with MSVC and/or clang-cl.. What version of clang is it?

If you just want to get past this to get it built, you can keep adding lines to delete the copy constructor and copy assignment like seems to work for filterbank. But I would hope a pull request with that would not be accepted, because it seems wrong to me.

I think someone more familiar with MSVC+clang-cl (or at least Windows programming more recent than Borland C++ 4.5) is needed to troubleshoot more.

gvanem, 2020-09-10:
>  This sounds like either a problem with your environment

The `gr-filter/lib` is the only module I have problems with (except for the v3.9 ABI-breakage that causes e.g. `gr-sdrplay` to not build). So I find it hard to believe there is *a problem with your environment*. 

> What version of clang is it?

```
clang version 10.0.0
Target: i686-pc-windows-msvc
```

ryanvolz, 2020-09-30:
I am having the same compilation errors with MSVC only (no clang involved) using a Visual Studio 2017 environment. Build logs are available [here](https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=214183&view=logs&jobId=5be07ae1-d8ba-5406-47b6-8e3a3a12f825&j=5be07ae1-d8ba-5406-47b6-8e3a3a12f825&t=0bf03e01-0bec-5b85-5316-b1633322e895). This is for building GR master for conda with conda-forge.

I think there might be clues in [this stackoverflow discussion](https://stackoverflow.com/questions/53168836/does-visual-studio-2017-need-an-explicit-move-constructor-declaration), where it points to a difference in how std::vector is implemented with Visual Studio 2017+. My best guess is that MSVC doesn't consider the move constructor/assignment to be `noexcept`, leading to it preferring the copy constructor/assignment which it tries from std::vector regardless of the fact that they are deleted for filterbank. Blame Microsoft, but it sure would be nice to find a fix for this that would actually use the move constructor/assignment. I've tried a few patches with the conda build (explicit noexcept, removing all of the trivial destructors to let them be defined by default) but haven't found success; I'm at the limits of my familiarity with C++ here, and I could easily be digging in the wrong hole.

ThomasHabets, 2020-09-30:
Aha, I see.

@ryanvolz 

So you're saying you tried:

```
    fir_filter (const fir_filter &) = delete;
    fir_filter (fir_filter &&) noexcept = default;
    fir_filter & operator=(const fir_filter &) = delete;
    fir_filter & operator=(fir_filter &&) noexcept = default;
```

And other combinations?

ThomasHabets, 2020-09-30:
Hmm… but if it were that simple then my small test above would also fail.

ThomasHabets, 2020-09-30:
@ryanvolz 

Actually, could you try this:

```c++

#include<vector>

class Uncopiable {
 public:
  Uncopiable(){}

  // Move yes.                                                                                                                              
  Uncopiable(Uncopiable&&) = default;
  Uncopiable&operator=(Uncopiable&&) = default;

  // Copy no.                                                                                                                               
  Uncopiable(const Uncopiable&) = delete;
  Uncopiable&operator=(const Uncopiable&) = delete;
};

class Foo {
 public:
  Foo(){}
  ~Foo(){}
  Uncopiable nocopy;
  std::vector<Uncopiable> nocopy_vector;
};
int main()
{
  Foo foo;
  foo.nocopy_vector.emplace_back();
  std::vector<Foo> vfoo;
}
```

ryanvolz, 2020-09-30:
Yes, I tried that and it didn't work, so you're right that it's not that simple. I also tried removing all of the empty destructors in gr-filter (thinking that letting the compiler define them would not trigger some logic that would inhibit noexcept for the move constructor/assignment), both on its own and in combination with the explicit noexcept, and that didn't work either.

So I don't actually understand what MSVC is doing, other than that I think it has something to do with how std::vector is implemented differently.

ryanvolz, 2020-09-30:
> @ryanvolz
>
> Actually, could you try this:
>
> ```c++
> #include<vector>
>
> class Uncopiable {
>  public:
>   Uncopiable(){}
>
>   // Move [...]

I will give this a try.

ThomasHabets, 2020-09-30:
If that works, try adding a `vfoo.reserve(1)` and see if it causes a problem.

If we can reproduce it here then I'd say yes I'll go and add noexcept to all move costructor/assignments globally. Which should perhaps be done anyway.

ThomasHabets, 2020-09-30:
Ah, that did it. I just realized I could check on godbolt. (facepalm)

ryanvolz, 2020-09-30:
> Ah, that did it. I just realized I could check on godbolt. (facepalm)

Indeed. Your original test compiled fine, but adding `vfoo.reserve(1)` brought up the same error we're seeing in GR.

ryanvolz, 2020-09-30:
If I comment out `Foo`'s destructor in your toy test, then it will compile even with `vfoo.reserve(1)`.

ryanvolz, 2020-09-30:
> If I comment out `Foo`'s destructor in your toy test, then it will compile even with `vfoo.reserve(1)`.

So what appears to be happening here is that by defining a `Foo` destructor, we don't get an implicitly-declared move constructor (see "Implicitly-declared move constructor" at https://en.cppreference.com/w/cpp/language/move_constructor), and so `Foo` will use a copy-constructor with `vfoo.reserve(1)` which then tries to use the copy constructor of `Uncopiable`. Do I have that right?

gvanem, 2020-09-30:
C++11 (and anything newer) is a mess.

ryanvolz, 2020-09-30:
This patch, on top of removing removing all empty destructors in gr-filter (not sure if this is necessary), gets it compiling for me.
```
From 5ee5419ae3305609b4ea37e391a1cd3bc976f474 Mon Sep 17 00:00:00 2001
From: Ryan Volz <ryan.volz@gmail.com>
Date: Wed, 30 Sep 2020 17:30:15 -0400
Subject: [PATCH] filter: include: Add explicit move constructors to fix build
 with MSVC.

---
 gr-filter/include/gnuradio/filter/filterbank.h              | 2 ++
 .../include/gnuradio/filter/mmse_fir_interpolator_cc.h      | 4 +++-
 .../include/gnuradio/filter/mmse_fir_interpolator_ff.h      | 4 +++-
 .../include/gnuradio/filter/mmse_interp_differentiator_cc.h | 4 +++-
 .../include/gnuradio/filter/mmse_interp_differentiator_ff.h | 4 +++-
 gr-filter/include/gnuradio/filter/pfb_arb_resampler.h       | 6 ++++++
 6 files changed, 20 insertions(+), 4 deletions(-)

diff --git a/gr-filter/include/gnuradio/filter/filterbank.h b/gr-filter/include/gnuradio/filter/filterbank.h
index 47235e9fb..09838b6b3 100644
--- a/gr-filter/include/gnuradio/filter/filterbank.h
+++ b/gr-filter/include/gnuradio/filter/filterbank.h
@@ -57,6 +57,8 @@ public:
      */
     filterbank(const std::vector<std::vector<float>>& taps);
 
+    filterbank(filterbank&&) = default;
+
     /*!
      * Update the filterbank's filter taps.
      *
diff --git a/gr-filter/include/gnuradio/filter/mmse_fir_interpolator_cc.h b/gr-filter/include/gnuradio/filter/mmse_fir_interpolator_cc.h
index 300b4043f..b21a8b248 100644
--- a/gr-filter/include/gnuradio/filter/mmse_fir_interpolator_cc.h
+++ b/gr-filter/include/gnuradio/filter/mmse_fir_interpolator_cc.h
@@ -40,6 +40,8 @@ class FILTER_API mmse_fir_interpolator_cc
 public:
     mmse_fir_interpolator_cc();
 
+    mmse_fir_interpolator_cc(mmse_fir_interpolator_cc&&) = default;
+
     unsigned ntaps() const;
     unsigned nsteps() const;
 
@@ -57,7 +59,7 @@ public:
     gr_complex interpolate(const gr_complex input[], float mu) const;
 
 protected:
-    const std::vector<kernel::fir_filter_ccf> filters;
+    std::vector<kernel::fir_filter_ccf> filters;
 };
 
 } /* namespace filter */
diff --git a/gr-filter/include/gnuradio/filter/mmse_fir_interpolator_ff.h b/gr-filter/include/gnuradio/filter/mmse_fir_interpolator_ff.h
index 6f07c12df..b1ae1ced4 100644
--- a/gr-filter/include/gnuradio/filter/mmse_fir_interpolator_ff.h
+++ b/gr-filter/include/gnuradio/filter/mmse_fir_interpolator_ff.h
@@ -40,6 +40,8 @@ class FILTER_API mmse_fir_interpolator_ff
 public:
     mmse_fir_interpolator_ff();
 
+    mmse_fir_interpolator_ff(mmse_fir_interpolator_ff&&) = default;
+
     unsigned ntaps() const;
     unsigned nsteps() const;
 
@@ -55,7 +57,7 @@ public:
     float interpolate(const float input[], float mu) const;
 
 protected:
-    const std::vector<kernel::fir_filter_fff> filters;
[... 61 lines cut ...]
@@ -257,6 +259,8 @@ public:
                           const std::vector<gr_complex>& taps,
                           unsigned int filter_size);
 
+    pfb_arb_resampler_ccc(pfb_arb_resampler_ccc&&) = default;
+
     /*!
      * Resets the filterbank's filter taps with the new prototype filter
      * \param taps (vector/list of complex) The prototype filter to populate the
@@ -444,6 +448,8 @@ public:
                           const std::vector<float>& taps,
                           unsigned int filter_size);
 
+    pfb_arb_resampler_fff(pfb_arb_resampler_fff&&) = default;
+
     /*!
      * Resets the filterbank's filter taps with the new prototype filter
      * \param taps (vector/list of floats) The prototype filter to populate the
-- 
2.20.1
```

I don't know why these move constructors do not get implicitly declared, but explicit works. Then there is also an issue with the `const` `filters` member in the MMSE classes which prevents move semantics. Instead of removing `const` there, I also tried adding `static` and making the necessary modification to the definition, and that also compiled. In fact, in that case, it's not necessary to explicitly declare the move constructor for `mmse_*`.

I don't know if this patch or something like it is the best way forward, but hopefully it sheds more light on what's happening.

ThomasHabets, 2020-10-01:
I didn't remove the empty destructors when I deleted their contents (due to RAII), because I saw other empty destructors and didn't want to close the door for destructor code without breaking ABI within 3.9.

But maybe the right solution is to remove the now empty destructors, thus getting the implicit moves back? (as you said, looks like MSVC deletes them according to standard because of the destructor.

I guess what I'm asking is, if you removed the empty destructors, didn't that on its own fix it?

I made [a gotbolt snippet ](https://godbolt.org/z/qd5bTa) that seems to confirm that theory.

Or are there non-empty destructors left? Aside from what's in https://github.com/gnuradio/gnuradio/pull/3728 I'm not finding any in `gr-filter/` in a quick check. (though it's a sloppy check. I'm in the middle of moving house)

ThomasHabets, 2020-10-01:
This apparently what can happen when you (in this case I) violate [the rule of 5](https://en.cppreference.com/w/cpp/language/rule_of_three).

ryanvolz, 2020-10-01:
> I guess what I'm asking is, if you removed the empty destructors, didn't that on its own fix it?

I would have thought that it would fix it, but apparently not everything. Needing to switch away from `const` in my patch above makes some sense, but I don't know why the explicit default move constructors are needed. MSVC must be skipping the implicit declaration for some reason that I still don't know.

The full two patches that I used to get past this issue are:
https://github.com/conda-forge/gnuradio-feedstock/blob/83a4b34a171805f2bda4388c279ebed5cc46464b/recipe/0001-filter-Do-not-declare-empty-destructors.patch
https://github.com/conda-forge/gnuradio-feedstock/blob/83a4b34a171805f2bda4388c279ebed5cc46464b/recipe/0002-filter-Add-explicit-move-constructors-to-fix-build-w.patch

I haven't tried the second patch without the first to see if deleting the empty destructors is actually necessary, but I can try that if it would be helpful. Perhaps explicitly following the rule of 5 everywhere here is the best way to handle this for clarity in the long term.

ThomasHabets, 2020-10-05:
I've started [a branch](https://github.com/ThomasHabets/gnuradio/tree/five-filter) for fixing the rule of five, at least. The commit history is ugly, but could you try if that branch builds for you on MSVC and or MSVC/clang-cl?

It [doesn't look like](https://godbolt.org/z/r4KrTn) the const should be a problem for MSVC.

ryanvolz, 2020-10-06:
> I've started [a branch](https://github.com/ThomasHabets/gnuradio/tree/five-filter) for fixing the rule of five, at least. The commit history is ugly, but could you try if that branch builds for you [...]

The build fails at (at least) `mmse_fir_interpolator_ff.cc` trying to use `fir_filter`'s deleted copy constructor:
```
FAILED: gr-filter/lib/CMakeFiles/gnuradio-filter.dir/mmse_fir_interpolator_ff.cc.obj 
C:\PROGRA~2\MICROS~4\2017\BUILDT~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\cl.exe  /nologo /TP -DBOOST_ALL_DYN_LINK -DGR_CTRLPORT -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DHAVE_CONFIG_H -DHAVE_WINDOWS_H -DMS_WIN64 -DNOMINMAX -DUSING_WINSOCK -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_filter_EXPORTS -I..\cmake\msvc -I..\gr-filter\lib\..\include -IC:\Users\rvolz\Miniconda3\envs\grdev\include -IC:\Users\rvolz\Miniconda3\envs\grdev\lib\site-packages\numpy\core\include -I..\gnuradio-runtime\lib\..\include -Ignuradio-runtime\lib\..\include -I..\gnuradio-runtime\lib\pmt\..\..\include -I..\gr-fft\lib\..\include -I..\gr-blocks\lib\..\include -IC:\Users\rvolz\Miniconda3\envs\grdev\Library\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG   /MP /bigobj /showIncludes /Fogr-filter\lib\CMakeFiles\gnuradio-filter.dir\mmse_fir_interpolator_ff.cc.obj /Fdgr-filter\lib\CMakeFiles\gnuradio-filter.dir\ /FS -c ..\gr-filter\lib\mmse_fir_interpolator_ff.cc
..\gr-filter\lib\..\include\gnuradio/filter/interpolator_taps.h(1043): warning C4305: 'initializing': truncation from 'double' to 'const float'
C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\include\xmemory0(881): error C2280: 'gr::filter::kernel::fir_filter<float,float,float>::fir_filter(const gr::filter::kernel::fir_filter<float,float,float> &)': attempting to reference a deleted function
..\gr-filter\lib\..\include\gnuradio/filter/fir_filter.h(34): note: see declaration of 'gr::filter::kernel::fir_filter<float,float,float>::fir_filter'
..\gr-filter\lib\..\include\gnuradio/filter/fir_filter.h(34): note: 'gr::filter::kernel::fir_filter<float,float,float>::fir_filter(const gr::filter::kernel::fir_filter<float,float,float> &)': function was explicitly deleted
```

That was one of the places where I needed to remove `const`ness from the vector `filters` member and add an explicit default move constructor, and I suspect the other places I had to do that would still be the same but the build just hasn't gotten to them.

ThomasHabets, 2020-10-06:
@ryanvolz curious. Removing `const` from `mmse_fir_interpolator_ff.cc`'s `filters'?

That sounds odd. It's as if MSVC isn't doing the mandated RVO. Oh, hang on, an auto storage duration variable [isn't mandatorily move-elisioned in C++11!](https://en.cppreference.com/w/cpp/language/copy_elision#Non-mandatory_elision_of_copy.2Fmove_.28since_C.2B.2B11.29_operations).

Could you try changing the `return` statement in `mmse_fir_interpolator_ff.cc` to `return std::move(filters)`? It's slightly less efficient (it's a move instead of elided move, so still very efficient).

ryanvolz, 2020-10-06:
> Could you try changing the `return filters` statement in `mmse_fir_interpolator_ff.cc` function `build_filters()` to `return std::move(filters)`? It's slightly less efficient (it's a move instead of elided move, so still very efficient).

Same error still. :frowning_face:

> I've so far been unable to reproduce the problem on MSVC. E.g. a start here.

I did a more literal attempt at coming up with a test case last week: https://godbolt.org/z/a6hMos. This one *does* fail to compile with the `const`, but it succeeds once that is gone whereas the actual code was failing still until I explicitly added the move constructor. So it still doesn't give the whole picture. I haven't looked at how our test cases are different, but maybe there is some information there. I do know that `bar.reserve(1);` at the end is necessary for triggering the failure.

> I'm still moving house, so sorry for not having spent more time on this.

No worries! I'm happy chipping away at it.

ThomasHabets, 2020-10-07:
[Here](https://godbolt.org/z/fbGTTK)'s a shorter version of your repro.

ThomasHabets, 2020-10-07:
Thinking more about this: yes I think you're right. I think this member simply has to not be `const` if it's to be movable. And if it's not movable (or copyable) then it won't fit in a `vector`.

ThomasHabets, 2020-10-27:
Ok, now that it's been merged I'm prepping the remaining fix.

ThomasHabets, 2020-10-27:
Tests currently running (on Linux though). @ryanvolz once I push a candidate branch, can you test it? Expect it today.

ThomasHabets, 2020-10-27:
@ryanvolz https://github.com/gnuradio/gnuradio/pull/3892 is now created. Please have a look.

marcusmueller, 2020-10-30:
Not quite sure whether this is related, but it does feel like it. 

```
../gr-filter/lib/../include/gnuradio/filter/fir_filter_with_buffer.h:56:33: warning: explicitly defaulted move assignment operator is implicitly deleted [-Wdefaulted-function-deleted]
    fir_filter_with_buffer_fff& operator=(fir_filter_with_buffer_fff&&) = default;
                                ^
../gr-filter/lib/../include/gnuradio/filter/fir_filter_with_buffer.h:31:15: note: move assignment operator of 'fir_filter_with_buffer_fff' is implicitly deleted because field 'd_align' is of const-qualified type 'const int'
    const int d_align;
              ^
../gr-filter/lib/../include/gnuradio/filter/fir_filter_with_buffer.h:158:33: warning: explicitly defaulted move assignment operator is implicitly deleted [-Wdefaulted-function-deleted]
    fir_filter_with_buffer_ccc& operator=(fir_filter_with_buffer_ccc&&) = default;
                                ^
../gr-filter/lib/../include/gnuradio/filter/fir_filter_with_buffer.h:133:15: note: move assignment operator of 'fir_filter_with_buffer_ccc' is implicitly deleted because field 'd_align' is of const-qualified type 'const int'
    const int d_align;
              ^
../gr-filter/lib/../include/gnuradio/filter/fir_filter_with_buffer.h:260:33: warning: explicitly defaulted move assignment operator is implicitly deleted [-Wdefaulted-function-deleted]
    fir_filter_with_buffer_ccf& operator=(fir_filter_with_buffer_ccf&&) = default;
                                ^
../gr-filter/lib/../include/gnuradio/filter/fir_filter_with_buffer.h:235:15: note: move assignment operator of 'fir_filter_with_buffer_ccf' is implicitly deleted because field 'd_align' is of const-qualified type 'const int'
    const int d_align;
              ^
```

ThomasHabets, 2020-11-02:
@marcusmueller It's not a blocker for MSVC, but move assignment+const member is an imperfection, yes.

gvanem, 2020-11-11:
This issue seems to have gotten worse with all the *templated* fft classes. But I've noted by trial and error that building
**gnuradio-filter.lib** as a static library **[*]**, it compiles fine! But is completely unusable if I do not create a .def-file to export stuff from it (not so easy for C++ symbols).

Can any C++ expert here explain this please? I.e. what does a `class __declspec(dllexport) filterbank` have to do with
error such as (in the  copy constructor):
```
f:\gv\VC_2019\VC\Tools\MSVC\14.27.29110\include\xutility(4142,16): error: overload resolution selected deleted operator
      '='
        *_Dest = *_First;
        ~~~~~~ ^ ~~~~~~~
```
A `class filterbank {`  .. compiles fine. Should `FILTER_API` b removed in some places?
I hope this should be a pointer to solve this issue.

**[*]**:  Adding these `CFLAGS`:
```
    -DINCLUDED_GNURADIO_ATTRIBUTES_H \
    -D__GR_ATTR_EXPORT= \
    -D__GR_ATTR_IMPORT=
```
Thus avoid the `dllimport` / `dllexport` cruft you gnu-folks are so lucky to not have to live with.

closed 2020-11-16 by mormj

gvanem, 2020-11-17:
That PR did not solve it; e.g. `qa_fir_filter_with_buffer.cc` wants several destructor that are missing.
Like these:
```c++
~fir_filter_with_buffer_fff (void)
~fir_filter_with_buffer_ccc (void)
~fir_filter_with_buffer_ccf (void)
```
The Python module also which needs some more missing imports.
You really should try building with MSVC before such a large commit.

ThomasHabets, 2020-11-17:
@gvanem I relied on @ryanvorz, as I've not had a have windows build environment for more than 20 years.

Could you be more specific what you mean by "wants"? Like compiler output. And what is "some missing imports"?

ryanvolz, 2020-11-17:
The current patch does the job for me, but I'll admit to not compiling/running the tests. I'll note, though, that master still needs a couple other PRs merged for me to get it to compile, so even if this is not fully fixed, it was clear more work needed to be done anyway. It's obvious that we're the only two who have tried to compile GR master on Windows, and I don't even use Windows personally! Please re-open this since it's not fixed for you.

Relatedly, I'm working on setting up CI for GR that will actually do a Windows build, so that everyone will at least be able to tell when Windows is broken. This way it won't be up to only you and me to report these build bugs.

gvanem, 2020-11-18:
> Could you be more specific what you mean by "wants"

Just that linking complains about missing destructors. It too much to explain. 

> I'm working on setting up CI for GR that will actually do a Windows build, 

Really good news.

PR #3892 (2020-11-16): filter: remove needless empty destructors and problematic const
You can't move from a const object, so unfortunately it's not possible
to keep constness on these.

This should fix #3757
commit b39c64d7 (2020-11-16): filter: remove needless empty destructors and problematic const

You can't move from a const object, so unfortunately it's not possible
to keep constness on these.

This should fix #3757