gnuradio/gnuradio #3761: How to run the block "Constellation Modulator" correctly on raspberry pi 4?

guokr321, 2020-09-10:
Hello everybody,
I use raspberry pi 4 and USRP B210, the block "Constellation Modulator" cannot work, but can receive the radio data through USRP using X86+USRP send radio data with Constellation Modulator. 
On raspberry pi 4 cmake uhd v3.15.0.0:
      cmake ../ -DNEON_SIMD_ENABLE=OFF -DPYTHON_EXECUTABLE=/usr/bin/python2
On raspberry pi 4 cmake gnuradio maint3.7:
     cmake ../ -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python2 -DCMAKE_CXX_FLAGS="-march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard" -DCMAKE_C_FLAGS="-march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard" -DCMAKE_ASM_FLAGS="-march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -mthumb -g"
    and:
       cmake ../ -Dhave_mfpu_neon=0 -DCMAKE_CXX_FLAGS:STRING="-march=armv8-a -mfpu=vfp -mfloat-abi=hard" -DCMAKE_C_FLAGS:STRING="-march=armv8-a -mfpu=vfp -mfloat-abi=hard" -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python2

Thanks.

dkozel, 2020-09-10:
Hello @guokr321 
Can you describe what is not working about the Constellation Modulator? As much information about what exactly is incorrect will help in giving any advice.

I see you are using GNU Radio 3.7, is upgrading to 3.8 possible?

guokr321, 2020-09-10:
using Constellation Modulator on X86 PC+USRP to send data, the Raspberry Pi 4 + USRP can receive the data.
but using Constellation Modulator on Raspberry Pi 4 +USRP to send data, the X86 PC + USRP cannot receive the data.

dkozel, 2020-09-10:
That's still a very large area where possible problems can be. Can you try removing the USRP and compare the output of the Constellation Modulator block when run with the same input on the X86 and on the Raspberry Pi? Is there a visible difference in the time domain output?

How are you configuring the Constellation Modulator? If you can share a screenshot of your flowgraph and all of the parameters for the Constellation Modulator that would be helpful.

For the most part all development work is focused on 3.8 and the future 3.9 versions. If you can find more specifically where a problem is in 3.7 someone may be able to help, but there are many fewer people working with it.

guokr321, 2020-09-10:
OK, thanks.
there is screenshot attached.
[image]

guokr321, 2020-09-10:
sps is 4, and excess_bw is 0.35

marcusmueller, 2020-10-30:
Considering there's a USRP in the loop here, and seeing #3882, I'd presume this is just because the Pi isn't keeping up with the USRP. If this is provably wrong, please do not hesitate to reopen this issue.

closed 2020-10-30 by marcusmueller