gnuradio/gnuradio #3769: Installing gnuradio 3.8 on a fresh Ubuntu 18.04

oayan, 2020-09-17:
Hi,

I have tried everything to follow the instructions to install gr 3.8 (both from source or via pybombs). Unfortunately, none of the instructions are complete. With pybombs, the gnuradio companion can not start, from source, it fails during make (even though I have installed the dependencies mentioned on the wiki. Does someone have a clean instructions, how to overcome this issue?

drmpeg, 2020-09-17:
Here's my recipe to build 3.8 from source. Since you've already failed a couple of times, it would be good to re-install Ubuntu 18.04 (that is, another fresh start).

This installs into a custom directory. This is done to avoid messing up /usr/local and provides an easy way to completely delete installations with ```sudo rm -rf /opt/gnuradio-3.8.2git```. Whenever you build an OOT, you have to install in the same prefix with ```cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.8.2git ../```.

```
sudo apt-get install git libboost-all-dev libusb-1.0-0-dev python-mako doxygen python-docutils cmake build-essential libncurses5 libncurses5-dev python-requests libudev-dev python-numpy python-setuptools python3-yaml python3-mako python3-requests python3-numpy ccache liborc-0.4-dev liblog4cpp5-dev libfftw3-dev libcppunit-dev libgmp-dev libgsl0-dev swig libzmq5 libzmq3-dev libsdl1.2-dev python3-click python3-click-plugins python3-gi-cairo libpangocairo-1.0-0 gir1.2-gtk-3.0 python3-pygccxml qtbase5-dev libqwt-qt5-dev pyqt5-dev-tools python3-matplotlib libcanberra-gtk3-module python3-pytest python3-zmq

git clone https://github.com/gnuradio/gnuradio.git
cd gnuradio/
git checkout maint-3.8
git submodule update --init
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.8.2git ../
make -j$(nproc)
sudo make install
sudo ldconfig

cd /etc/profile.d
sudo nano gnuradio.sh

/* type in the following 9 lines of text to the file gnuradio.sh. ctrl-o to write file and ctrl-x to exit */

#!/bin/bash
# add gnuradio to path
export PATH=$PATH:/opt/gnuradio-3.8.2git/bin
# add gnuradio python libraries to python search path
if [ $PYTHONPATH ]; then
       export PYTHONPATH=$PYTHONPATH:/opt/gnuradio-3.8.2git/lib/python3/dist-packages
else
       export PYTHONPATH=/opt/gnuradio-3.8.2git/lib/python3/dist-packages
fi

cd /etc/ld.so.conf.d
sudo nano gnuradio.conf

/* type in the following 1 line of text to the file gnuradio.conf. ctrl-o to write file and ctrl-x to exit */

/opt/gnuradio-3.8.2git/lib

sudo ldconfig

reboot
```

oayan, 2020-09-17:
Thank you. I will try it after reinstalling, but isn't there supposed to be a previous step to install the uhd libraries?

drmpeg, 2020-09-17:
I'll update the recipe with UHD instructions.

drmpeg, 2020-09-17:
You may have to add:

sudo apt-get install python3-setuptools

for UHD 4.0

oayan, 2020-09-21:
With the help of your script, I prepared a whole script with some additional dependencies. It has been tried on a fresh Ubuntu 18.04 installation and works fine!

  # DEPENDENCIES   
sudo apt-get install git libboost-all-dev libusb-1.0-0-dev python-mako doxygen python-docutils cmake build-essential libncurses5 libncurses5-dev python-requests libudev-dev python-numpy python-setuptools python3-yaml python3-mako python3-requests python3-numpy ccache liborc-0.4-dev liblog4cpp5-dev libfftw3-dev libcppunit-dev libgmp-dev libgsl0-dev swig libzmq5 libzmq3-dev libsdl1.2-dev python3-click python3-click-plugins python3-gi-cairo libpangocairo-1.0-0 gir1.2-gtk-3.0 python3-pygccxml qtbase5-dev libqwt-qt5-dev pyqt5-dev-tools python3-matplotlib libcanberra-gtk3-module python3-pytest g++ libboost-all-dev libgmp-dev swig python3-numpy python3-mako python3-sphinx python3-lxml doxygen libfftw3-dev libsdl1.2-dev libgsl-dev libqwt-qt5-dev libqt5opengl5-dev python3-pyqt5 liblog4cpp5-dev libzmq3-dev python3-yaml python3-click python3-click-plugins python3-zmq python3-scipy python3-zmq python3-setuptools python-matplotlib

   # UHD INSTALLATION
   mkdir workarea
   cd workarea
   git clone https://github.com/EttusResearch/uhd.git
     cd uhd/
     git checkout v4.0.0.0 -b tmp
     cd host/
     mkdir build
     cd build
     cmake ../
     make -j$(nproc)
     sudo make install
     sudo ldconfig
   
   # GNURADIO 3.8 INSTALLATION
     cd ~/workarea/
     git clone https://github.com/gnuradio/gnuradio.git
     cd gnuradio/
     git checkout maint-3.8 
     git submodule update --init
     mkdir build
     cd build/
     cmake ../
     uhd_find_devices 
     make -j$(nproc)
     sudo make install
     sudo ldconfig

   # UHD CONFIGURATION
     sudo uhd_images_downloader
     nano ~/.bashrc
   # add gnuradio python libraries to python search path
   if [ $PYTHONPATH ]; then
       export PYTHONPATH=$PYTHONPATH:/usr/local/usr/local/lib/python3/dist-packages
   else
       export PYTHONPATH=/usr/local/lib/python3/dist-packages
   fi
 
     source .bashrc 
     cd $HOME/workarea/uhd/host/utils
     sudo cp uhd-usrp.rules /etc/udev/rules.d/
     sudo udevadm control --reload-rules
     sudo udevadm trigger

closed 2020-09-21 by oayan