gnuradio/gnuradio #3772: OOT CMake errors caused by sndfile?

jacobgilbert-snl, 2020-09-20:
After rebuilding an OOT against the most recent master GR, I am getting errors as something is causing it to be linked against libsndfile:

```bash
[  1%] Linking CXX shared library libgnuradio-pdu_utils.so
/usr/bin/ld: cannot find -lsndfile::sndfile
collect2: error: ld returned 1 exit status
```

I do use some components from gnuradio-blocks:

```
find_package(Gnuradio "3.9" REQUIRED COMPONENTS filter runtime pmt blocks fft)
```

Below is the error I am now getting. Do I need to find_package(sndfile) now? This seems like it should not be necessary in the OOT as I do not use sndfile at all.

```
CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:131 (add_library):
  Target "pdu_utils_python" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/gnuradio-master/lib/cmake/gnuradio/GrPybind.cmake:227 (pybind11_add_module)
  python/bindings/CMakeLists.txt:54 (GR_PYBIND_MAKE_OOT)

CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:131 (add_library):
  Target "pdu_utils_python" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/gnuradio-master/lib/cmake/gnuradio/GrPybind.cmake:227 (pybind11_add_module)
  python/bindings/CMakeLists.txt:54 (GR_PYBIND_MAKE_OOT)

CMake Error at lib/CMakeLists.txt:56 (add_library):
  Target "gnuradio-pdu_utils" links to target "sndfile::sndfile" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
```

drmpeg, 2020-09-20:
It's all fixed with https://github.com/gnuradio/gnuradio/pull/3750

jacobgilbert-snl, 2020-09-20:
Thanks! I pulled that PR in and it is fixed but there is still some unexpected (to me at least) cmake output:

```
<...SNIP...>

-- Build type not specified: defaulting to release.
CMake Warning at CMakeLists.txt:76 (find_package):
  By not providing "FindSNDFILE.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "SNDFILE", but
  CMake did not find one.

  Could not find a package configuration file provided by "SNDFILE" with any
  of the following names:

    SNDFILEConfig.cmake
    sndfile-config.cmake

  Add the installation prefix of "SNDFILE" to CMAKE_PREFIX_PATH or set
  "SNDFILE_DIR" to a directory containing one of the above files.  If
  "SNDFILE" provides a separate development package or SDK, be sure it has
  been installed.

<...SNIP...>

-- Checking for module 'sndfile'
--   Found sndfile, version 1.0.28
-- Found SNDFILE: /usr/lib/x86_64-linux-gnu/libsndfile.so  

<...SNIP...>

```
Let me know if this is expected for a module that requires the `blocks` GR component and I'll close this.

closed 2020-09-20 by jacobgilbert-snl