gnuradio/gnuradio #3779: packet_tx.grc RuntimeError

sobakaliza, 2020-09-25:
Hello. When I try to run (on Ubuntu) the official example project uhd_packet_tx.grc which depends on packet_tx.grc, I get the following python error:

> Warning: failed to XInitThreads()
Traceback (most recent call last):
  File "/home/gnuradio/Documents/qam/uhd_packet_tx.py", line 377, in <module>
    main()
  File "/home/gnuradio/Documents/qam/uhd_packet_tx.py", line 353, in main
    tb = top_block_cls()
  File "/home/gnuradio/Documents/qam/uhd_packet_tx.py", line 183, in __init__
    self.packet_tx_a = packet_tx(
  File "/home/gnuradio/.grc_gnuradio/packet_tx.py", line 68, in __init__
    self.digital_burst_shaper_xx_0.set_block_alias("burst_shaper0")
  File "/usr/lib/python3/dist-packages/gnuradio/digital/digital_swig1.py", line 2731, in set_block_alias
    return _digital_swig1.burst_shaper_cc_sptr_set_block_alias(self, name)
RuntimeError: symbol already exists, can not re-use!

>>> Done (return code 1)

I managed to fix that issue by editing the packet_tx.py script, which is created automatically in /home/gnuradio/.grc_gnuradio
I changed the line 68: self.digital_burst_shaper_xx_0.set_block_alias("**burst_shaper0**") to self.digital_burst_shaper_xx_0.set_block_alias("**burst_shaper_a**") and now it works. 

May you kindly handle this issue? Thanks.

csylvain, 2020-09-27:
i believe the correct way to handle this issue is not to edit the hierarchical file (packet_tx.py) but to be sure to have generated the code from the flowgraph (but not *Run*), where it will be automatically stored for you in /home/<username>/.grc_gnuradio .. the ".py" and ".yml" files will be placed there and ready to be referenced by another flowgraph such as packet_loopback_hier.grc

sobakaliza, 2020-09-27:
I agree that editing the Python file is not the right solution to this problem. But it is still weird that the original example does not work from the scratch. Though my directories were clean.

csylvain, 2020-09-28:
it worked for me from scratch, but only after being very deliberate in
generating the code of the hier blocks first before opening the flowgraph
that uses them.

sobakaliza, 2020-09-28:
Sounds interesting! Maybe there should be some official documentation about how to execute the blocks.

WA1CYB, 2020-10-19:
To fix the example, open up the Burst Shaper GRC block, select Advanced, Change the Alias to burst_shaper_a. 
No need to modify the python generated file. 
I recommend the examples not be hard coded with a specific Alias.

csylvain, 2020-10-19:
it does not entirely fix the example. once you have the _tx _rx and _hier
parts all working, you should find (as I have already reported as an Issue)
messages being emitted at the output only when you select the settings for
redundant bits and not the full FEC processing.

WA1CYB, 2020-10-24:
It works for me when I: Set the redundant bit to 1, Throttle the input message stream and Don't load the Heir block (at higher speeds). Isn't that full FEC processing?

csylvain, 2020-10-27:
are we actually talking about the same thing?

in packet_loopback_hier.grc there are:
1. two Constellation Objects (one for header, the other of packet payload):
  a. Const_HDR and Const_PLD Constellation Objects
  b. Default Header Format Obj. and Const_PLD Constellation Objects

2. FEC related definition blocks:
  a. Repetition Encoder and Repetition Decoder
  b. CC Encoder and CC Decoder

there is no "redundant bit to 1" which i observe anywhere in the flowgraph.

one is meant to select a pair of Constellation Objects (from #1 above) and
a pair of Encoder-Decoder blocks (from #2 above).

when the pair from #1.a. are selected and the pair from #2.a., then there
are payloads emitted and observed at "print_pdu" which pass the CRC32 check
in packet_rx.
when the pair #2.b. are selected, then there are NO PAYLOADS EMITTED.
apparently none pass the CRC32 check.

jaredd, 2020-10-31:
I had the same issue as the @sobakaliza except I was using the transmitter_sim_hier.grc which only depends on packet_tx.grc.

```
$ gnuradio-config-info --print-all
--- snip ---
testing-support;python-support;volk;doxygen;gnuradio-runtime;gr-ctrlport;gnuradio-companion;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;gr-channels;gr-qtgui;gr-trellis;gr-uhd;gr-utils;gr_modtool;gr-vocoder;gr-wavelet;gr-zeromq
3.8.2.0
clang version 11.0.0-++20200414100631+5c1d1a62e37-1~exp1~20200414201304.3176  
Target: x86_64-pc-linux-gnu 
Thread model: posix 
InstalledDir: /usr/bin
clang version 11.0.0-++20200414100631+5c1d1a62e37-1~exp1~20200414201304.3176  
Target: x86_64-pc-linux-gnu 
Thread model: posix 
InstalledDir: /usr/bin
/usr/lib/ccache/clang-11:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized 
/usr/lib/ccache/clang++-11:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized

```
Their solution of changing the generated python worked for me and I realize this is merely a sanity check. Starting from scratch and generating the py file for packet_tx.grc (and reloading blocks in grc) prior to opening transmitter_sim_hier.grc did not work. Same error as OP.

WA1CYB, 2020-10-31:
I think so. I have attached my expanded version of the loopback example where the tx is not a heir block.The tx stream is 1st encoded with the CC_Encoder then the header is also encoded through the Repetition Encoder. The payload doesn't use the Repetition Encoder. This combination does work with the repetition variable rep = 1 or rep = 3.  Note that only those two can be selected because the the repetition select in packet_rx is only set up for those two values.
The attached flowgraph uses BPSK on both the header and payload and a ETTUS B205 mini for a transmitter. With the receiver bypassed and a test file transmitted, the system decodes correctly for rep=1 or rep=3. Since only the header uses the Repetition Encoder the transfer capability is almost identical (for rep=1 with a 1,000 kByte file I received 999.7 kBytes. For rep=3 I received 933.5k Bytes). 
Since your a ham also I thought you may be interested in the attached received spectrum of the repeated 1 Mbyte file.

WA1CYB, 2020-10-31:
Sorry, attached the wrong file. Correct reference file attached.

seuliang, 2021-01-21:
I have fixed problem in packet_tx  as @WA1CYB  posted 

> To fix the example, open up the Burst Shaper GRC block, select Advanced, Change the Alias to burst_shaper_a.

But, how to fix problems in packet_rx ?
When i run uhd_packet_rx, I got:
`Traceback (most recent call last):
  File "/home/usrp/Desktop/GRCLearn/packet/uhd_packet_rx.py", line 575, in <module>
    main()
  File "/home/usrp/Desktop/GRCLearn/packet/uhd_packet_rx.py", line 553, in main
    tb = top_block_cls()
  File "/home/usrp/Desktop/GRCLearn/packet/uhd_packet_rx.py", line 381, in __init__
    self.packet_rx_0 = packet_rx(
  File "/home/usrp/.grc_gnuradio/packet_rx.py", line 67, in __init__
    self.digital_header_payload_demux_0 = digital.header_payload_demux(
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. gnuradio.digital.digital_python.header_payload_demux(header_len: int, items_per_symbol: int = 1, guard_interval: int = 0, length_tag_key: str = 'frame_len', trigger_tag_key: str = '', output_symbols: bool = False, itemsize: int = 8, timing_tag_key: str = '', samp_rate: float = 1.0, special_tags: List[str] = [], header_padding: int = 0)

Invoked with: 96, 1, 0, 'payload symbols', 'time_est', True, 8, 'rx_time', 1, '', 0`

nlhnt, 2021-03-03:
I know this is about a different example, but something similar happened with `examples/digital/packet_loopback_hier.grc`.  
What worked for me was disabling the Burst Shaper in `examples/digital/packet_tx.grc` and replacing it with a new one, presumeably the same block.

linkandlive, 2022-02-14:
> I have fixed problem in packet_tx in gnuradio 3,8 as @WA1CYB posted
> 
> > To fix the example, open up the Burst Shaper GRC block, select Advanced, Change the Alias to burst_shaper_a.
> 
> But, how to fix problems in packet_rx ? When i run uhd_packet_rx in gnuradio 3.9, I got:
> 
> ```
> Traceback (most recent call last):
>   File "/home/usrp/Desktop/GRCLearn/packet/uhd_packet_rx.py", line 575, in <module>
>     main()
>   File "/home/usrp/Desktop/GRCLearn/packet/uhd_packet_rx.py", line 553, in main
>     tb = top_block_cls()
>   File "/home/usrp/Desktop/GRCLearn/packet/uhd_packet_rx.py", line 381, in __init__
>     self.packet_rx_0 = packet_rx(
>   File "/home/usrp/.grc_gnuradio/packet_rx.py", line 67, in __init__
>     self.digital_header_payload_demux_0 = digital.header_payload_demux(
> TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
>     1. gnuradio.digital.digital_python.header_payload_demux(header_len: int, items_per_symbol: int = 1, guard_interval: int = 0, length_tag_key: str = 'frame_len', trigger_tag_key: str = '', output_symbols: bool = False, itemsize: int = 8, timing_tag_key: str = '', samp_rate: float = 1.0, special_tags: List[str] = [], header_padding: int = 0)
> 
> Invoked with: 96, 1, 0, 'payload symbols', 'time_est', True, 8, 'rx_time', 1, '', 0
> ```

@seuliang were you able to solve your mentioned issue?

marcusmueller, 2024-01-28:
This not-really-a-bug-report got hijacked to become a discussion forum. 

Please don't do that on the issue tracker! 

If something here persists, please open a new *bug report* if it's a bug, or contact the mailing list if you just need help!

closed 2024-01-28 by marcusmueller