gnuradio/gnuradio #3786: runtime: race condition in circular buffer allocation in Windows

japm48, 2020-09-30:
The method used to obtain a circular buffer in Windows ([`vmcircbuf_createfilemapping.cc`](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/vmcircbuf_createfilemapping.cc)) is the same as the one described [here](https://gist.github.com/nickva/7277351b066b5bba35f3) and [here](https://fgiesen.wordpress.com/2012/07/21/the-magic-ring-buffer/).

As mentioned, this sequence of calls is racy: `VirtualAlloc -> VirtualFree -> CreateFileMapping -> MapViewOfFile`. A second thread could obtain the address just freed.

This is probably the cause of a hard-to-reproduce kind of bugs, but, it was the only way (AFAIK) to allocate a circular buffer in Windows.

There is a new officially documented method involving `VirtualAlloc2` and `MapViewOfFile3`, using something called "placeholders"; see the [first example in MSDN](https://docs.microsoft.com/en-us/windows/win32/api/memoryapi/nf-memoryapi-virtualalloc2#examples). It's a bit more complicated, but race-free.

These new APIs are only available in Windows 10 >=1803, so I suggest a runtime detection using `GetProcAddres`.

When/if I have time, I will try to implement this myself.

bastibl, 2020-09-30:
I don't know much about Windows and it might make a lot of sense to support more recent APIs in anyway.

But I think there is a misunderstanding of what the author of the blog post calls "dicey" and what you call racy: It is true that there is no guarantee that the mapping always works. Memory allocations from another thread could (in theory) interfere between demapping of the placeholder and the two back-to-back mappings of the same segment. Other circular buffer implementations have the same issue and just [retry several times](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/vmcircbuf_sysv_shm.cc#L52-L53) before they fail.
The code, however, [checks the return value](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/vmcircbuf_createfilemapping.cc#L117-L132) of the mapping to make sure that the segment ended up exactly where it was supposed to. I agree that this is not nice, but I don't see how this would be a source for hard-to-reproduce bugs. Or did I just miss the actual race condition here? The fact that this doesn't cause a lot of issues is probably that GNU Radio typically allocates buffers before spawning threads.

japm48, 2020-09-30:
Thanks @bastibl, for the quick and informative reply.

> I don't know much about Windows

Me neither, I just was reading the implementation and delved deeper into the rabbit hole after wondering why the region is freed just after allocating it.

> But I think there is a misunderstanding of what the author of the blog post calls "dicey" and what you call racy

I admit that's quite possible.

> It is true that there is no guarantee that the mapping always works. Memory allocations from another thread could (in theory) interfere between demapping of the placeholder and the two back-to-back [...]

Perhaps this should also be done here.

> The code, however, [checks the return value](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/vmcircbuf_createfilemapping.cc#L117-L132) of the mapping to make sure that the [...]

I assumed that this would be considered UB-like as this looks similar to use-after-free; of course, I don't know the Windows guarantees here (probably undocumented). So the checks done may or may not be sufficient.

> Or did I just miss the actual race condition here? The fact that this doesn't cause a lot of issues is probably because GNU Radio typically allocates buffers before spawning threads.

Ah, that makes sense. I assumed every thread allocated its own buffer. So, if there is in fact a problem here it is because of a different process.

bastibl, 2020-10-01:
> I assumed that this would be considered UB-like as this looks similar to use-after-free; of course, I don't know the Windows guarantees here (probably undocumented). So the checks done may or may not be sufficient.

No, that's the point. It might work (the segments are at the expected place) or it might not work (they ended up somewhere else) but there is no undefined behavior. It is a process that works or raises an error, like nearly every function that involves memory allocation.

> Ah, that makes sense. I assumed every thread allocated its own buffer. So, if there is in fact a problem here it is because of a different process.

Actually one thread (the main thread) allocates all buffers and then starts a thread for every block in the flowgraph.

marcusmueller, 2024-01-28:
graph-proto/upcoming GR4 does the "try as long as it works" dance.

I think it's still cooler that if your OS offers a safe way, to use that. 

Funny that we had this issue lying around all these years and I forgot about it… my apologies.