gnuradio/gnuradio #3798: NBFM Receive block causes audio underruns

duggabe, 2020-10-03:
I have a problem with receiving narrow band FM. When I receive WBFM stereo, there is no problem. But using the same basic flowgraph, tuning to the 144MHz band, and substituting a NBFM Receive block, or an FM Demod block, I get LOTS of audio under-runs and loud noise. My setup is:
Lenovo ThinkCentre AMD® Ryzen 7 PRO 3700 Processor
Ubuntu 20.04.1
UHD 4.0 built from source
GR/GRC 3.9.0.0-git built from source 26 Aug 2020
USRP B200-mini
* tests I have done:
1) extend tuning range of WBFM stereo; add squelch - almost no aU's
2) replace WBFM Receive PLL block with NBFM Receive block - lots of aU's
3) replace NBFM Receive block block with FM Demod block - lots of aU's
4) replace FM Demod block with a Quadrature Demod and FM Deemphasis - almost no aU's
5) in test 2, replace FFT Low Pass Filter with an FFT Filter - lots of aU's

Notes:
1) FFT Low Pass Filter is new in 3.9
2) test5 works on rPi
Raspberry Pi 3B+
UHD 3.15
GR/GRC 3.8.1.0
USRP B200-mini

Then I revised the filter parameters for NBFM and revised the order of the blocks in test4 to yield the following:
The NBFM Receive block (in test2) is equivalent to the Quadrature Demod, FM Deemphasis, and FFT Filter in test4. See https://github.com/gnuradio/gnuradio/blob/master/gr-analog/python/analog/nbfm_rx.py
USRP_NFM_test2.grc https://pastebin.com/R0wavmAy
USRP_NFM_test3.grc https://pastebin.com/imRjyWEc
USRP_NFM_test4.grc https://pastebin.com/tiFikkp3
USRP_NFM_test5.grc https://pastebin.com/iwGbSSJV

On 3.9, test2 FAILS with lots of aU's; test4 PASSES with no aU's and good signals, even though they are ostensibly equivalent!

duggabe, 2020-10-03:
To summarize the issue above:
test2 FAILS in 3.9
test3 FAILS in 3.9
test4 PASSES in 3.9 and 3.8
test5 FAILS in 3.9 and PASSES in 3.8

My conclusion: 
The NBFM Receive block does not work properly in version 3.9.0.0-git as of 26 Aug 2020

mrjacobagilbert, 2020-10-06:
I am not sure what you are experiencing, but of the 4 test flowgraphs you posted only USRP_NFM_test4 causes audio underruns for me under 3.9.

I do not think this is caused by the NBFM block.

argilo, 2020-10-12:
The alsa sink uses a 40-millisecond buffer by default:

https://github.com/gnuradio/gnuradio/blob/422f86fd8084b4bf5323110608d3d0d81d27ccd7/gr-audio/lib/alsa/gr-audio-alsa.conf#L9-L12

https://github.com/gnuradio/gnuradio/blob/422f86fd8084b4bf5323110608d3d0d81d27ccd7/gr-audio/lib/alsa/alsa_sink.cc#L50-L59

And audio playback should begin when the buffer is half full:

https://github.com/gnuradio/gnuradio/blob/422f86fd8084b4bf5323110608d3d0d81d27ccd7/gr-audio/lib/alsa/alsa_sink.cc#L236-L242

It seems to me that 20ms of wiggle room is rather low, considering that the RTL-SDR's default block size is 131072 samples (which is ~65ms at 2 Msps) and Codec 2 processes audio in 40-ms blocks.

Increasing `nperiods` to 20 (by editing `~/.gnuradio/config.conf`) eliminates the audio underruns in USRP_NFM_test2.grc for me. This gives a 200-ms buffer, and audio should begin streaming after 100 ms. Perhaps we should make a larger value the default.

That said, there are still some things I don't understand about the alsa sink's behaviour:

1. I'd expect to be able to set `nperiods` lower with the USRP since it returns samples in small blocks, but I get underruns unless `nperiods` is >= 9.
2. Even with a very high value of `nperiods`, it seems that audio output begins immediately, and the "start_threshold" doesn't kick in until one underrun occurs.

duggabe, 2020-10-23:
for my computers (Ubuntu and rPi) @argilo has identified the problem.
I changed ~/.gnuradio/config.conf as follows:
```
[audio_alsa]
default_input_device = default
default_output_device = default
nperiods = 32
period_time = 0.010
verbose = false
```
sometimes I get one aU and sometimes none. It is a satisfactory solution for me.
@argilo, do you plan to do anything further with this?

closed 2020-12-17 by mormj