gnuradio/gnuradio #3801: Installation issue of Gnuradio 3.8.2.0 on Ubuntu 18.04.05

TarikKazaz, 2020-10-04:
Hi All,

I getting errors when calling make test, while installing Gnuradio 3.8.2.0 on Ubuntu 18.04.05.
I have also installed thrift version 0.10.0. 

These are errors
`99% tests passed, 3 tests failed out of 368

Total Test time (real) = 166.29 sec

The following tests FAILED:
	171 - qa_cpp_py_binding (Failed)
	172 - qa_cpp_py_binding_set (Failed)
	173 - qa_ctrlport_probes (Failed)
Errors while running CTest
`

Further if I call 

ctest --output-on-failure

I get following

`170/368 Test #170: qa_copy ............................................   Passed    0.46 sec
        Start 171: qa_cpp_py_binding
171/368 Test #171: qa_cpp_py_binding ..................................***Failed    0.83 sec
EE
======================================================================
ERROR: test_001 (__main__.test_cpp_py_binding)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_cpp_py_binding.py", line 111, in test_001
    rval = v1.get()
  File "/home/tkazaz/workarea/gnuradio/build/gnuradio-runtime/python/gnuradio/gr/../../../swig/runtime_swig.py", line 7519, in get
    return _runtime_swig.RPC_get_string_get(self)
RuntimeError: basic_string::_M_construct null not valid

======================================================================
ERROR: test_002 (__main__.test_cpp_py_binding)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_cpp_py_binding.py", line 162, in test_002
    radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
TypeError: __init__() got an unexpected keyword argument 'argv'

----------------------------------------------------------------------
Ran 2 tests in 0.352s

FAILED (errors=2)
DEPRECATED: Using filename with gr_unittest does no longer have any effect.

        Start 172: qa_cpp_py_binding_set
172/368 Test #172: qa_cpp_py_binding_set ..............................***Failed    0.71 sec
EE
======================================================================
ERROR: test_001 (__main__.test_cpp_py_binding_set)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_cpp_py_binding_set.py", line 107, in test_001
    rval = g3.get()
  File "/home/tkazaz/workarea/gnuradio/build/gnuradio-runtime/python/gnuradio/gr/../../../swig/runtime_swig.py", line 7519, in get
    return _runtime_swig.RPC_get_string_get(self)
RuntimeError: basic_string::_M_construct null not valid

======================================================================
ERROR: test_002 (__main__.test_cpp_py_binding_set)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_cpp_py_binding_set.py", line 129, in test_002
    radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
TypeError: __init__() got an unexpected keyword argument 'argv'

----------------------------------------------------------------------
Ran 2 tests in 0.253s

FAILED (errors=2)
DEPRECATED: Using filename with gr_unittest does no longer have any effect.

        Start 173: qa_ctrlport_probes
173/368 Test #173: qa_ctrlport_probes .................................***Failed    1.18 sec
EEEEE
======================================================================
ERROR: test_001 (__main__.test_ctrlport_probes)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_ctrlport_probes.py", line 67, in test_001
    radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
TypeError: __init__() got an unexpected keyword argument 'argv'

======================================================================
ERROR: test_002 (__main__.test_ctrlport_probes)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_ctrlport_probes.py", line 108, in test_002
    radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
TypeError: __init__() got an unexpected keyword argument 'argv'

======================================================================
ERROR: test_003 (__main__.test_ctrlport_probes)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_ctrlport_probes.py", line 148, in test_003
    radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
TypeError: __init__() got an unexpected keyword argument 'argv'

======================================================================
ERROR: test_004 (__main__.test_ctrlport_probes)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_ctrlport_probes.py", line 189, in test_004
    radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
TypeError: __init__() got an unexpected keyword argument 'argv'

======================================================================
ERROR: test_005 (__main__.test_ctrlport_probes)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tkazaz/workarea/gnuradio/gr-blocks/python/blocks/qa_ctrlport_probes.py", line 229, in test_005
    radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
TypeError: __init__() got an unexpected keyword argument 'argv'

----------------------------------------------------------------------
Ran 5 tests in 0.705s

FAILED (errors=5)
DEPRECATED: Using filename with gr_unittest does no longer have any effect.

        Start 174: qa_delay
174/368 Test #174: qa_delay ...........................................   Passed    0.46 sec
        Start 175: qa_endian_swap
175/368 Test #175: qa_endian_swap .....................................   Passed    0.48 sec`

Does anyone know what would be solution?

Kind Regards,
Tarik

nickoe, 2020-10-06:
@TarikKazaz Are you building from source? If so, please also include the commit you are on?  Try:
```
git describe
```

TarikKazaz, 2020-10-08:
Hi,
@nickoe I am at last release v3.8.2.0, commit (
67d8415f4d635c186fdf03a7a1d18b84d666a0c5 )

Kind Regards,

Tarik

doronbehar, 2023-10-04:
I am experiencing exactly the same failures:

```
The following tests FAILED:
         40 - qa_cpp_py_binding (Failed)
         41 - qa_cpp_py_binding_set (Failed)
         42 - qa_ctrlport_probes (Failed)
```

Here on NixOS, for GR 3.9.8.0.

marcusmueller, 2024-01-28:
we currently pass all test in CI

closed 2024-01-28 by marcusmueller

doronbehar, 2024-01-29:
I'm still experiencing these failures with GR 3.9.8.0, here on NixOS with gcc13.

> we currently pass all test in CI

Did you mean all `maint-3.10` tests pass in CI?