gnuradio/gnuradio #3802: Blocks for digital voice

michael-roe, 2020-10-04:
A while ago I wrote some Gnuradio blocks to implement the Yaesu System Fusion digital voice protocol. I have been talking to Derek Kozel about merging some of these blocks into the core of Gnuradio (rather than as an OOT  module) so we can have some discussion first before I submit a pull request.

Current code is in Michael-roe/gr-c4fm on GitHub.

Things I think should be merged:

Type conversions

  Complex to interleaved float

Forward error correction

  Crc16
  Golay code (the Golay decoder still needs some code cleanup before it is fit to be merged. It currently works,  but needs cleanup)

4 level FSK
  C4fm soft decision (from 4FSK modulation to a pair of soft decisions)
  Quaternary slicer (slices real and imaginary parts of a complex number to two bits)
  Frequency locked loop
  Correlate access code tag cc (complex variant of correlate access code tag)

Coding
  Sign scrambler (flips the sign bit of a float according to the output of an LFSR)
  
Random number generation (used for testing error correction)
  Bernoulli source

If you want to see an application using these blocks, a complete transmitter and receiver is in Michael-roe/ysf2

It was also suggested that I could write a tutorial article on how to use these blocks to build a transmitter and receiver. I’m willing to write that, but haven ‘t written it yet.

dkozel, 2020-10-07:
Hi @michael-roe Thanks for posting!

Several of these blocks are pretty clear enhancements/extensions of existing ones so I think it will be a clear path to upstreaming them, such as the `Correlate access code tag cc` block.

The Complex to Interleaved Float is interesting, in memory they're the same. Where was it useful in your application? It would be good to have a matching Interleaved Float to Complex if that does get upstreamed.

michael-roe, 2020-10-07:
Hi Derek!

Complex and interleaved float have the same in-memory representation, but gnuradio thinks they're different types.

The use case is where I have a coimplex number whose real and imaginary parts are each a soft decision for a bit, and I want to split them up as soft decisions for individual bits.

Yes, we could do Interleaved Float to Complex

jacobgilbert-snl, 2020-10-07:
@michael-roe you can directly translate between interleaved float and complex (with no intermediate block if you natively use vectors) like this:

[image]

The stream to vector block configured as `blocks.stream_to_vector(gr.sizeof_float*1, 2)`

michael-roe, 2020-10-08:
@jacobgilbert-snl you're right: stream to vector will do it, vector to stream will do it, theres no need for complex to interleaved float.

michael-roe, 2020-10-09:
The quaternary slicer could be generalised to a block that slices a vector of floats and packs the bits into a byte.

Correlate access tag cc is mathematically more like working on a vector of floats than a complex; it has an obvious generalisation to working on vectors of more than two floats. (If you’re looking for a sync word in 8FSK, you’ll want 3 floats)

michael-roe, 2020-10-14:
I have just realised that there is a better way to generalise correlate access code tag for n-fsk signals, n > 2:

The input is a stream of floats, each contains the output of the FM discriminator, with values -3, -1, 1, 3 being expected in the case of 4FSK, (but of course there will be noise in the signal). The n-FSK sync word is a parameter to the block. The block computes sum(abs(x(t - i) - s(i))), where x is the received signal and s is the expected sync word. The block tags the sample if the computed sum of differences is less than a threshold, the value of which is a parameter to the block.

Does this sound ok?