gnuradio/gnuradio #3803: python crashes

gotchicode, 2020-10-04:
Hi,

I try to launch the resampler_demo.grc and python crashes after clicking execute.

I am on Windows 7, and use GNU 3.8.1

Thanks

[image]

[image]

MattMills, 2020-10-20:
@gotchicode  Since you're on Windows and you're seeing the same return code (-1073741795) I'd bet you are seeing the same issue as #3822. There is a suggested fix I put in that issue you may want to try and see if it resolves your issue.

gvanem, 2020-11-18:
I'm on Windows too. The sample works fine with the latest 3.9 master AFAICS. Compiling and running it I get this:
[image]

nickoe, 2020-12-29:
Assuming this is really a packaging issue by gnieboer as seen in https://github.com/gnieboer/GNURadio_Windows_Build_Scripts/issues/48 and maybe @gnieboer has not bee able to make a build because of https://github.com/gnuradio/gnuradio/issues/3911. 

I will close this ticket as it can be tracked the other issues.

closed 2020-12-29 by nickoe