gnuradio/gnuradio #3813: 3.8.2.0 crash-on-start after build-from-src of gr-soapy, with fix

BStudent, 2020-10-06:
### Sequence:
- Installed 3.8.2.0 from Kali current rolling release.
- **gnuradio-companion starts correctly**
- Installed soapySDR and gr-soapy, built from source, no problems.
- **gnuradio-companion crashes on startup**, traceback shown below
    - final error message: `FileNotFoundError: [Errno 2] No such file or directory: 'gsettings'`
- This error traces to the call:
```python
theme = subprocess.check_output(subprocess.check_output
        ["gsettings", "get", "org.gnome.desktop.interface", "gtk-theme"]
```
- The call is equivalent to shell command: `gsettings get org.gnome.desktop.interface gtk-theme`
- **Root cause: gsettings utility is not installed on my system** _(why did it not crash before?  I do not know.)_
    - `locate` utility had outdated locatedb entry for `/usr/bin/gsettings`, but utility not actuall present
- **Fix: install gsettings**, note that "what to install" is not obvious from the name, here is the apt-get command:
    - `sudo apt-get install libglib2.0-bin`

### Conclusion
Not clear whether this is a grc issue, soapy issue, or kali issue, so I am submitting as an issue-with-workaround for anyone using them in combination.

### Log
```log
... other stuff tracing back to command line ...
  File "/usr/lib/python3/dist-packages/gnuradio/grc/gui/ParamWidgets.py", line 48, in have_dark_theme
    theme = subprocess.check_output(
  File "/usr/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'gsettings'
```

wirew0rm, 2020-10-06:
According to this [post on the debian mailinglist](https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1764890.html) this has been fixed in 7826795, so it should be fixed with the next release I guess.
Just run into the same issue myself and found both the issue and the mailing list thread.

BStudent, 2020-10-06:
Cool, good to have it all memorialized here.

closed 2020-10-06 by BStudent