gnuradio/gnuradio #3814: Libvolk error

KyyB, 2020-10-06:
Dear team,

I have been facing the following error:
[ERROR] Configuration failed after running at least twice
[ERROR] Problem occurred while building package libvolk:
  Configuration Failed
[ERROR] Error installing package libvolk. Aborting.

I read other feeds but they were not working for me. Any idea on how to resolve this?

Thank!

dkozel, 2020-10-07:
Hi @KyyB 
You'll need to post a lot more info for anyone to be able to help much. First, are you installing VOLK separately from GNU Radio, if so the https://github.com/gnuradio/volk/issues tracker is a better place to post.

If not, what version of GNU Radio/VOLK are you installing and are there any other messages in the cmake/build log that show what the configuration error is?

jdemel, 2020-10-13:
Do you use PyBOMBS to install GNU Radio? The error message hints at PyBOMBS. In that case please add info about your PyBOMBS version and installation as well. Also, please add info about your system. OS, distro.

dkozel, 2020-10-15:
Hi @KyyB 

In your other issue you said you were using Ubuntu 16.04 which Ubuntu will stop supporting in 6 months (April 30, 2021). It is definitely possible to install GNU Radio and VOLK on 16.04, but if you have the choice a newer OS is a better idea for lots of reasons.
https://github.com/gnuradio/gnuradio/issues/3849

I'm going to close this issue since it looks like you're moving onto a different OS and will try installing a different way. If you hit an issue again I'd recommend asking on https://chat.gnuradio.org or the discuss-gnuradio email list since you're most likely to have an issue with your specific setup rather than an issue with GNU Radio's code itself and you'll get better help that way.
https://lists.gnu.org/mailman/listinfo/discuss-gnuradio

I hope it works for you!

closed 2020-10-15 by dkozel

SteveKrzysiak, 2020-10-23:
I think maybe this should be reopened, I'm seeing this on ubuntu 20.04, pybombs 2.3.4. Relatively fresh install of ubuntu, I think the only thing I've already installed is  gqrx-sdr. Please let me know what other info I can provide.

willcode, 2020-10-24:
Fix is to add "gitargs: --recursive" to the recipe for libvolk. The maintainers were discussing this on the volk channel today.

jdemel, 2020-10-24:
VOLK requires to be cloned recursively now. 
[This gr-recipes fix 238](https://github.com/gnuradio/gr-recipes/pull/238) is what you need.

SteveKrzysiak, 2020-10-26:
Thank you both @willcode & @jdemel !

evso9816, 2020-12-06:
Hi
I am running through the same problem: I have Ubuntu 20.04, gnuradio 3.8.2. Volk is also installed [git clone --recursive https://github.com/gnuradio/volk.git].
Doing installations through "pybombs install multi-rtl" causes the following: "problem occurred while building package libvolk"
Any support to solve? I attempted to install "libvolk" but unsuccessfully. Thanks.

jdemel, 2020-12-06:
@evso9816 are your recipes up to date? You may `cd .pybombs/recipes/gr-recipes` and in that folder (which is a git repository) `git pull`. This should get you the latest recipe.
Besides, we'd need more info what you do exactly. Also, before you run `pybombs install multi-rtl`, just run `pybombs install libvolk` in that repo and report the full error message if any. What I did here is give an answer based on a lot of assumptions.

evso9816, 2020-12-07:
Hi Jdemel,
I think I have the latest recipe. all installations have been freshly done less than 2 or 3 weeks.
Attached is the log: One installation using cmake; Two installation using "pybombs". By the way, the same installation went successfully in Ubuntu 16.04 Gnuradio 3.7.  
Keep me updated if more information needed.
Greatly thanks
Eva
[log_installation.txt](https://github.com/gnuradio/gnuradio/files/5651077/log_installation.txt)

liumaolin-931, 2021-04-01:
Hi willcode,
I'm a novice. Can you tell me how to add "gitargs: --recursive" in detail? What statement should I enter at the terminal？
Gratly thanks!

willcode, 2021-04-01:
@liumaolin-931 This change has already been made to the PyBOMBS recipe for libvolk, so there is nothing you need to do, other than possibly `pybombs recipes update`.

liumaolin-931, 2021-04-02:
Thank you for your reply, now some new questions have appeared, but I try to find the answer first, thank you very much!

liumaolin-931, 2021-04-06:
The problem remains unsolved，and My system is Ubuntu 16.04, gnuradio version is 3.7.10, pybombs version is 2.3.4.the following is the error code prompt

~/gnuradio-prefix$ sudo pybombs install libvolk
[sudo] liumaolin passwords： 
[INFO] Prefix Python version is: 2.7.12
[INFO] PyBOMBS Version 2.3.4
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- libvolk
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: libvolk
[INFO] Install python-apt to speed up apt processing.
[WARNING] Build dir already exists: /home/liumaolin/gnuradio-prefix/src/libvolk/build
Building:    (100%) [=========================================================]
[  3%] Built target utils
[  4%] Built target cpu_features
[  5%] Built target unix_based_hardware_detection
[  7%] Built target list_cpu_features
[ 86%] Built target volk_obj
[ 86%] Built target volk
[ 87%] Building CXX object lib/CMakeFiles/volk_test_all.dir/testqa.cc.o
In file included from /home/liumaolin/gnuradio-prefix/src/libvolk/build/include/volk/volk.h:31:0,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/lib/qa_utils.h:5,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/lib/kernel_tests.h:1,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/lib/testqa.cc:31:
/home/liumaolin/gnuradio-prefix/src/libvolk/include/volk/volk_common.h: In function ‘float log2f_non_ieee(float)’:
/home/liumaolin/gnuradio-prefix/src/libvolk/include/volk/volk_common.h:153:24: error: ‘isinf’ was not declared in this scope
     return isinf(result) ? copysignf(127.0f, result) : result;
                        ^
/home/liumaolin/gnuradio-prefix/src/libvolk/include/volk/volk_common.h:153:24: note: suggested alternative:
In file included from /usr/include/c++/5/complex:44:0,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/include/volk/volk_complex.h:23,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/build/include/volk/volk_typedefs.h:30,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/build/include/volk/volk.h:29,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/lib/qa_utils.h:5,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/lib/kernel_tests.h:1,
                 from /home/liumaolin/gnuradio-prefix/src/libvolk/lib/testqa.cc:31:
/usr/include/c++/5/cmath:621:5: note:   ‘std::isinf’
     isinf(_Tp __x)
     ^
make[2]: *** [lib/CMakeFiles/volk_test_all.dir/build.make:63：lib/CMakeFiles/volk_test_all.dir/testqa.cc.o] error 1
make[1]: *** [CMakeFiles/Makefile2:406：lib/CMakeFiles/volk_test_all.dir/all] error 2
make: *** [Makefile:141：all] error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package libvolk:
Build failed.
[ERROR] Error installing package libvolk. Aborting.

What does "VOLK requires to be cloned recursively now." mean? How should I operate？Can someone help me with this problem？thanks!

evso9816, 2021-04-06:
Hi,
I will recommend that you install your packages manually. I went through
the same issue with Ubuntu 20.04 and using Pybombs. Just make a partition
on your disk and proceed manually. Of course, you'll have to install
dependencies. Easy enough! This link might be useful and look to " For the
GNU Radio Master Branch" : https://wiki.gnuradio.org/index.php/InstallingGR
For Pybombs users, I will try to reproduce the case again and share it with
Gnuradio. Hopefully someone will have time to jump in.
Thanks,

eva

jdemel, 2021-04-06:
> The problem remains unsolved，and My system is Ubuntu 16.04, gnuradio version is 3.7.10, pybombs version is 2.3.4.the following is the error code prompt

> [ 87%] Building CXX object lib/CMakeFiles/volk_test_all.dir/testqa.cc.o
> In file included from /home/liumaolin/gnuradio-prefix/src/libvolk/build/include/volk/volk.h:31:0,
> from [...]

> What does "VOLK requires to be cloned recursively now." mean? How should I operate？Can someone help me with this problem？thanks!

@liumaolin-931 Since you work with Ubuntu 16.04, your specific issue is tricky. Basically, you trigger a very specific bug that is part of your specific distro. Since Ubuntu 16.04 is EOL by the end of this month, I recommend to upgrade to a newer LTS release, preferably 20.04.
If you are strictly bound to this outdated Ubuntu version, I recommend to use an older VOLK (e.g. 2.0 or 2.2.1) version. I know, this requires some more manual work and you don't get all the latest and greatest features but that's the usual trade-off. Since you try to install GR 3.7, you could also just stick with your distros libvolk package.

@evso9816 In my experience, the best solution would be to use the GNU Radio PPA. Compiling from source shouldn't be necessary. PyBOMBS is a solution for cross-compiling, developing with multiple GR versions, etc. Also, a source compile should only be required if you intend to work with GR code directly. If you only alter OOT module code, you don't need to compile GR from source.
I had a look at your log file. Your VOLK install errors out because the cpu_features folder is empty. Most likely because it didn't get cloned recursively. As far as I can tell, you tried to use an already existing folder. In that case, the repository is not cloned recursively again and the issue persists.