gnuradio/gnuradio #3822: Running GNU Radio Companion 3.9.1.0 (Python 2.7.10) gives: return code -1073741795

RHAD1969, 2020-10-08:
I've installed GNU Radio Companion 3.9.1.0 (Python 2.7.10) on Windows 10 (OS build 19041.508) by downloading gnuradio_3.8.1.0_win64.msi and installing it with default options.

The first error:

> qt.qpa.plugin: Could not find the Qt platform plugin "windows" in "C:\Program Files\GNURadio-3.8\bin\\plugins\platforms"
> This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
> Done (return code -1073740791)

After creating a folder C:\Program Files\GNURadio-3.8\bin\plugins and copying the folder C:\Program Files\GNURadio-3.8\bin\platforms to it another error occured that I could not solve:

>Generating: 'C:\\Users\\Richard\\Documents\\top_block.py'
>Executing: C:\Program Files\GNURadio-3.8\gr-python27\python.exe -u C:\Users\Richard\Documents\top_block.py
>\>\>\> Done (return code -1073741795)

I've tried multiple blocks (even Signal Source => null sink) all result in the same return code and no top block presented. Tried mulitple systems => exactly the same issues.

ha6bli, 2020-10-19:
I've installed GNU Radio Companion 3.8.1.0 (Python 2.7.10) on Windows 10 Home 64 bit (OS build 18363) by downloading gnuradio_3.8.1.0_win64.msi and installing it with default options. PC is an Asus ROG G750JM with an i7 CPU , SSD memory and 32GB of RAM.

This is what I get when I try and run any flowgraphs, regardless of complexity. I have tried reinstalling GNU Radio and it did not help. 

`Generating: 'C:\\\\Users\\\\leven\\\\Documents\\\\Drive\\\\school\\\\uvic-grad\\\\ta\\\\ece350-ta-2020-fall\\\\ece350-lab2\\top_block.py'

Executing: C:\Program Files\GNURadio-3.8\gr-python27\python.exe -u C:\\Users\\leven\\Documents\\Drive\\school\\uvic-grad\\ta\\ece350-ta-2020-fall\\ece350-lab2\top_block.py

qt.qpa.plugin: Could not find the Qt platform plugin "windows" in "C:\Program Files\GNURadio-3.8\bin\\plugins\platforms"
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

>>> Done (return code -1073740791)`

Jovan8, 2020-10-19:
Yea i have the same issue on windows 7 with the same gnuradio version. I created a new issue about the same thing before seeing this. This needs to be fixed.

MattMills, 2020-10-20:
This sounds related to/causing this issue:

https://github.com/gnieboer/GNURadio_Windows_Build_Scripts/issues/48

There is a suggested fix in that thread if you want to try it, assuming you got the installer from here: http://www.gcndevelopment.com/gnuradio/index.htm

Jovan8, 2020-10-20:
> 
> 
> This sounds related to/caused by this issue:
> 
> [gnieboer/GNURadio_Windows_Build_Scripts#48](https://github.com/gnieboer/GNURadio_Windows_Build_Scripts/issues/48)
> 
> There is a suggested fix in that thread if you want to try it, assuming you got the installer from here: http://www.gcndevelopment.com/gnuradio/index.htm

Modifying the .bat script worked to resolve the problem, however now im getting a different issue, when running the dvbt tx script with hackrf python crashes. Console displays return code -1073741795. I have no idea how to fix this, hope someone can help.

dkozel, 2020-10-31:
Hi @Jovan8 I'm glad to hear that the first issue is fixed by the modification suggested in the Windows build scripts repository.

For you hackrf issue, do any flowgraphs with the HackRF work? Since the HackRF support isn't part of the main GNU Radio codebase this repository isn't the best place to work on the issue. I suggest checking if this is particular to one flowgraph or a general problem with the HackRF blocks, and then reporting it on the Windows Build Scripts repo or gr-osmosdr since that's where the HackRF support is implemented.

The GNU Radio chat and mailing lists are a good place to ask for some quick debugging assistance and recommendations.
https://chat.gnuradio.org
#gnuradio on the Freenode IRC server
https://lists.gnu.org/mailman/listinfo/discuss-gnuradio

closed 2020-10-31 by dkozel