gnuradio/gnuradio #3830: sssssssssss what does it mean in Gnu Radio Companion?

AnthonyGHolland, 2020-10-11:
Gnu Radio Companion is outputting a series of the letter s:
SSSSSSSSSSSSSSSS
it's synthesizing a 50khz signal for me, but what do all those SSSSS mean?
thanks for any advice/tips!
Anthony

mrjacobagilbert, 2020-10-11:
Are you using a USRP?

Also, you may want to try chat.gnuradio.org or the gnuradio mailing list for general help questions.

AnthonyGHolland, 2020-10-12:
Yes, I am using a USRPN200

I’ve been thrilled to be able to program a custom signal in gnuradio companion, but am having a bit of trouble getting assistance with technical issues like this… thanks for the pointer, I’ll try it!

Anthony

japm48, 2020-10-19:
This also bit me, but with a different device (I don't have a USRP). To make things more confusing this is device-dependent.

For the USRPs, the [UHD driver](https://github.com/EttusResearch/uhd) is the responsible for the error message, so using the difficult-to-find documentation ([here](https://files.ettus.com/manual/page_general.html) and [here](https://files.ettus.com/manual/structuhd_1_1async__metadata__t.html#a2be1b5c0351746c78fa3bcb74a8ff5da)), and searching for `UHD_LOG_FASTPATH` in UHD sources, these are the possible error codes:

 - `U`: underrun (or underflow). This occurs when the host does not produce data fast enough. This is only on the TX side.
 - `O`: overrun (or overflow). The host cannot consume data fast enough (opposite of underrun). RX-side only.
 - `S`: sequence error: "Packet loss between host and device". Unfortunately not well documented.
 - `D`: apparently this is a combination of overrun and sequence error (??), for network devices.
 - `L`: Tx: "Late data (arrived after indicated time)". Rx (RFNoC only?): "Late command (stream command arrived after indicated time)".

Apparently, the behaviour may be different when using RFNoC.

Someone may feel free to copy this info to the docs and fill in the missing parts.

AnthonyGHolland, 2020-10-20:
This is brilliant! Thank YOU SO VERY MUCH!

I will study the links you sent.  This is not an easy business to learn!

Anthony

japm48, 2020-10-20:
Thanks!
But bear in mind that with this info you probably won't easily fix your problem. So it's probably better for you to contact the UHD people to see what is really happening and how to solve it.

AnthonyGHolland, 2020-10-20:
Thanks again… and now stupid question…..how do I contact the “UHD people” ?
I wouldn’t even know where to look??
Anthony

japm48, 2020-10-20:
You may post your issue on their github (https://github.com/EttusResearch/uhd/), but probably your best options are, as @mrjacobagilbert mentioned, to ask in the GNU Radio chat chat.gnuradio.org or in the mailing list.

It is important that you describe clearly what you are doing: which OS and GR version, log messages, upload the grc flowgraph, etc.

AnthonyGHolland, 2020-10-20:
Thanks for the pointer!
Anthony

dkozel, 2020-10-27:
Hi Anthony. I'm glad the above comments were helpful and I hope that you're up and running now. I'm going to close this issue but please feel free to drop into the chat (https://chat.gnuradio.org) or the discuss-gnuradio mailing list if you have any other questions.

closed 2020-10-27 by dkozel

AnthonyGHolland, 2020-10-28:
OK, thanks again Derek for your great help!
Anthony