gnuradio/gnuradio #3831: why added sidebands to simple sinewave?

AnthonyGHolland, 2020-10-11:
Gnu Radio companion is synthesizing a 60kHz sinewave for me.
But it also adds very weak spectrum peaks at 20khZ and 100kHz
The location of those 'extra' sidebands seems to change if I change
the sample rate.  I have an srate of 200khz to output a 60khz sinewave..
am I somehow seeing aliasing here????

I see that the extra sidebands in the spectrum shift if I change the sample rate....???
strange?

Anthony

csylvain, 2020-10-12:
is the amplitude = 1? or less than 1?

AnthonyGHolland, 2020-10-12:
Yes, amplitude of the signal is set to 1.
Thank you

csylvain, 2020-10-13:
how many dB do the sidebands go down (if they do, that is) when you reduce the amplitude by 3 dB (0.5 instead of 1.0)?

rear1019, 2020-10-14:
GNU Radio doesn’t add the sidebands on purpose. They occur due to the way the signal is generated (assuming the “Signal Source” block is used). The number, strength and position of the unwanted sidebands depends on the ratio of signal frequency and sample rate. Changing the amplitude will *not* change the ratio of sideband strength and main component strength ¹ (i.e. SNR wil not improve). You might want to look into “direct digital synthesis (DDS)”, “spurious signals (aka spurs)” and “phase truncation” (or a combination of the terms). 

¹ However, you must still prevent clipping, i.e. use appropriate levels depending on data type and hardware.

AnthonyGHolland, 2020-10-14:
This is very interesting, thank you.
I have noticed that if I change the sample rate in gnuradio companion, that the location of some unwanted sidebands changes.  Being mostly experienced in digital AUDIO and knowing the Nyquist limits of srate/2 for digital AUDIO, I had assumed that so long as my sample rate was at least 2X the highest frequency I wanted to generate in gnuradio companion…I would have a clean signal (assuming I’m not clipping).  But it seems the relationship between sample rate and potential unwanted sidebands is a bit more complex in gnuradio companion than in simple digital audio where the only thing you have to worry about is not exceeding half the sample rate (don’t want to produce aliased frequencies).

I’m transitioning my understanding from digital audio to gnuradio companion.
I see there are some interesting complications here….hmmm?

Thanks for the feedback/help!!!!
Anthony

AnthonyGHolland, 2020-10-14:
I’ve tried amplitude of 1 and also amplitude of 0.7

drmpeg, 2020-10-15:
The spurs are 120 dB down from the main signal. That's the dynamic range of a theoretically ideal 20-bit DAC. Given that most SDR DAC's are 12-bits, for all intents and purposes, the spurs essentially don't exist.

csylvain, 2020-10-16:
in case anyone reading this thread did not know, an appropriate and useful troubleshooting technique for *analog* circuits exhibiting unexpected sidebands/spurs is to vary the main signal level a dB or two and note how many dB the sidebands/spurs change. it can help determine the type and source.

AnthonyGHolland, 2020-10-16:
Thank you for that suggestion! I will try that!
Anthony

drmpeg, 2020-10-17:
It's just harmonic distortion. Switch to float and 10 kHz sine wave, and the location of the spurs will make more sense (3rd, 5th, 7th, etc. harmonic).

[image]

AnthonyGHolland, 2020-10-17:
Thanks.  It would make more sense to me if the signal being generated were a square or triangle wave (odd harmonics only)….but since somebody already said these ‘spurious’ ‘spurs’ are essentially limitations of the DAC, I guess I shouldn’t be concerned too much.

Thanks to all you gurus of gnuradio for your help!
It’s a steep learning curve for a digital AUDIO guy to learn all this ;-)

Anthony

[image]<https://user-images.githubusercontent.com/5397196/96324508-028b4900-0fd7-11eb-864b-bdde8fbf8290.png>

marcusmueller, 2020-10-29:
@AnthonyGHolland they're not "limitations of the DAC", they're limitations of how accurately you can represent numbers and calculate sines.

To make this very clear: a signal at -120 dB the power of the signal you care about isn't there, for all practical purposes. We just have good enough math to describe the imperfection of our sine calculation  and number representation. 

120 dB is roughly the relation between the power of light entering your eye from a 100 W light bulb, 500 m away, vs 1 horsepower. If you're getting dragged around by that horse, you don't care about the light bulb half a kilometer away, promise.

Since this really is more about "understanding orders of magnitude" than about a bug in GNU Radio, I'll allow myself to close this bug report.

closed 2020-10-29 by marcusmueller

AnthonyGHolland, 2020-10-29:
That was brilliantly explained! Thank you so very much!!!!!
Anthony