gnuradio/gnuradio #3836: Vocoder blocks don't show up in GRC

michael-roe, 2020-10-12:
The vocoder blocks don't show up in GRC. Some of them I can get to by searching for them by name (CVSD Encoder/Decoder, Alaw Encoder/Decoder, ulaw Encoder/Decoder) but some of them I can't access at all (e.g. CODEC2, which is the one I really wanted).

drmpeg, 2020-10-12:
If you built from source, you have to have the CODEC2 dependency installed.

```sudo apt-get install libcodec2-dev```

michael-roe, 2020-10-12:
I didn't have libcodec2-dev installed. But even after I installed it and rebuilt gnuradio, I stil don't see the biock,

I was using pybombs to build gnuradio. Any suggestions for what I should
do to force it to completely reconfigure and rebuild to pick up the new dependency?

csylvain, 2020-10-14:
find the .yml file(s) for your blocks and make sure they've been installed in /usr/local/share/gnuradio/grc/blocks

michael-roe, 2020-10-19:
Now that I have installed libcodec2-dev and recompiled from source (not using pybombs) I can see the codec2 block if I search for it by now.

However, none of the vocoder blocks appear in the block picker on the right hand side of the screen in gnuradio-companion. I think this is a bug.

drmpeg, 2020-10-19:
Yes, it's a bug. The block tree file isn't being installed for gr-vocoder. The vocoder blocks are supposed to be in "Audio". Here's what it should look like:

[image]

I'll submit a pull request to fix it.

closed 2020-10-19 by mormj