gnuradio/gnuradio #3842: Can't load fir_filters in OOT Blocks

cclamb, 2020-10-12:
Hey everybody! I'm having a problem using the gr::filter::mmse_fir_interpolator_ff class. If I create a C++ OOT block, and then include this class, somewhat like this:

#include <gnuradio/filter/mmse_fir_interpolator_ff.h>
...
class some_class_impl : public some_class
{
        ...
    protected:
        gr::filter::mmse_fir_interpolator_ff _interpolator;
        ...
};

The python module generated by swig will not contain *any* block definitions from the project, which you can verify, assuming the project containing the above class is named 'gr-filter_test' by executing:

python3 -c 'import filter_test as ft ; print(dir(ft))'

If you have the interpolator commented out, I can see the OOT classes associated with the module. If it isn't commented out, no modules are visible.

It seems that the gr::filter::kernel::fir_filter_fff _filter class may be causing the issue - I've been able to recreate this problem with this class in a similar way.

My hunch is that Swig can't dynamically bind to the correct libraries.

I've tested this in projects generated using GNURadio 3.8.2.0 installed via pybombs on Ubutnu 20.04.

I have a simple example project I can attach if you'd like - let me know.

velichkov, 2020-10-12:
Hi @cclamb,

Do you link your OOT library with `gnuradio::gnuradio-filter`? Try adding it in `target_link_libraries` in your `lib/CMakeLists.txt`, something like

```cmake
target_link_libraries(filter_test gnuradio::gnuradio-runtime gnuradio::gnuradio-filter)
```

jacobgilbert-snl, 2020-10-12:
[Here is an OOT block that uses the in-tree FIR filter kernels.](https://github.com/sandialabs/gr-pdu_utils/blob/master/lib/pdu_fir_filter_impl.cc). The module CMake files may be helpful as a reference.

cclamb, 2020-10-13:
This is great feedback guys. Thanks! I'll try these out. If they work, I'll close.

cclamb, 2020-10-13:
@jacobgilbert-snl Unfortunately, SWIG fails if I just include the class in a header file. The linking solution was what I was thinking the issue might be - thanks for the sample though!

cclamb, 2020-10-13:
@velichkov Is there somewhere that this linking technique is documented?

velichkov, 2020-10-13:
@cclamb It is documented [here](https://wiki.gnuradio.org/index.php/GNU_Radio_3.8_OOT_Module_Porting_Guide#GNU_Radio_Components). You'll most probably need to add `filter` in the `find_package` list in CMakeLists.txt in the top directory.
```
find_package(Gnuradio "3.8" REQUIRED COMPONENTS blocks analog filter fft)
```

cclamb, 2020-10-13:
@velichkov Thanks much!

cclamb, 2020-10-15:
Your suggestions fixed the issue. Thanks so much!

closed 2020-10-15 by cclamb