gnuradio/gnuradio #3845: pybind11 is alsp a build dependecy

michael-roe, 2020-10-14:
The instructions for how to build gnuradio from source give a list of the dependencies to install, but omit pybind11. It would appear that you need version 2.4 of pybind11 to build the current head of gnuradio; this is more recent than the pybind11 shipped with Ubuntu 18.04;

duggabe, 2020-10-14:
In the second section of https://wiki.gnuradio.org/index.php/UbuntuInstall#Focal_Fossa_.2820.04.29 it states
```
The development version of GNU Radio (3.9) additionally requires:

sudo apt install pybind11-dev python3-matplotlib libsndfile1-dev
```
I can add that to the https://wiki.gnuradio.org/index.php/UbuntuInstall#Bionic_Beaver_.2818.04.29_through_Eoan_Ermine_.2819.10.29 section, but am unable to test it.

michael-roe, 2020-10-14:
The section for Bionic Beaver probably ought to say that the development version of GNU Radio is not expected to build, as it needs a more recent  version of pybind11

777arc, 2020-10-14:
> The section for Bionic Beaver probably ought to say that the development version of GNU Radio is not expected to build, as it needs a more recent version of pybind11

Does that mean 3.9 and up won't work in Ubunu 18 without doing something special?

777arc, 2020-10-14:
> 3.9 will build from source on Ubuntu 18 if pybind11 is installed from source as well
> Also, pybind11 packages are on the gnuradio-master PPA, so someone could install pybind from the GR PPA then build gnuradio from source

Ah, ty, I'll update the wiki to point that out

duggabe, 2020-10-14:
Also we need to add the statement from Focal Fossa for the 3.9 requirements.

duggabe, 2020-10-14:
added statement for Ubuntu 18

michael-roe, 2020-10-15:
Thanks, I think that resolves the issue.

closed 2020-10-15 by michael-roe