gnuradio/gnuradio #3849: Recommended Ubuntu version

KyyB, 2020-10-15:
Hi everyone,

What is the recommended Ubuntu Version for GNUradio? I have tried Ubuntu 16.04 but faced issues with Libvolk. 

Appreciate the support

dkozel, 2020-10-15:
Hi @KyyB In general the latest Ubuntu is recommended for personal use and the latest LTS for large projects that need to have a fixed platform for multiple years.

Information about installing from the main repository (recommended), the PPA (if you absolutely need the latest release), or by source (for unreleased fixes or to make changes to the main codebase) can be found here:
https://wiki.gnuradio.org/index.php/UbuntuInstall

closed 2020-10-15 by dkozel

KyyB, 2020-10-15:
The error below is what I get in Ubuntu 16.04 constantly. I have read all the post to resolve it yet none had resolved my issue.

Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package libvolk:
Build failed.
[ERROR] Error installing package libvolk. Aborting.

mormj, 2020-10-15:
I would definitely recommend going up to Ubuntu 18.04 at a minimum if you are trying to install GNU Radio from master branch, or maint-3.8 branch.

dkozel, 2020-10-16:
@KyyB For us to help more can you let us know what version of GNU Radio you were installing, how you were building it (source or via PyBOMBS), and the full error message? The error message you posted says "See output above for error messages" so there should be useful information higher in the log.

Thanks!