gnuradio/gnuradio #3852: please help i cant fucking do this anymore

xmmc2, 2020-10-16:
I have been trying for several weeks to install gnuradio on ubuntu and i cannot fucking do it. The apt-get command works and installs it but there is no osmocom sink witch i need for hackrf, so i tried installing gr somocom thing it failed, then i tried building gnuradio from source, it failed i tried so many things on a couple of distros this has given me extreme stress and anxiety i just wanna use gnuradio companion on linux is that too much to fucking ask, the windows version is shit, it wont detect the hackrf thats why im trying linux i have given up on this after weeks of trying please if anyone has a vmware image with any linux distro and gnuradio installed give me the link for it or upload it somewhere i cant do this anymore the past few weeks have been a nightmare i cant focus on anything ive been on my computer for hours per day trying to get gnuradio to work I CANT DO THIS ANYMORE. istg i will fucking kms the next time i see a linux terminal im so fucking sick of it fuck linux fuck everything im gonna have a nervous breakdown because of it it is not worth my time and mental health to install a single fucking application and fail at it every time for weeks. i have 18 virtual machines that i have created for the purpouse of installing gnuradio and i somehow managed to brick the system on almost all of them please i just wanna use my hackrf is it so fucking much to ask. again if you have a vmware image with gnuradio pleas i am begging you to provide me with it i cant do this anymore

dkozel, 2020-10-16:
Hi @xmmc2 Sounds like you've been having a rough time! I hope I can help.

Ideally, all that should be required is `apt install gr-osmosdr` on a new Ubuntu system and that should automatically install . I don't have a HackRF so can't test myself. There are a bunch of HackRF specifics for checking out if you have the latest firmware installed that I'm not familiar with. It looks like the HackRF wiki is a reasonable place to start, though it links to some older info.
https://github.com/mossmann/hackrf/wiki/Getting-Started-with-HackRF-and-GNU-Radio

There's a VM image here that has a lot of pieces preinstalled, but I don't know if includes the gr-osmosdr. GNU Radio Companion does work right out of the box though and it'd be an ok base to work with from a clean environment, though there are some details about how it's setup that mean a regular Ubuntu 20.04 ISO might make a better starting place for a VMWare image.
https://wiki.gnuradio.org/index.php/UbuntuVM

The live GNU Radio chat is a good place for asking questions and getting help a bit more interactively. I'd recommend joining there and someone might be able to help with specific issues as they come up.
https://chat.gnuradio.org

dkozel, 2020-10-16:
I've just asked and the Ubuntu VM does include the HackRF support, though I don't think anyone has had a HackRF to test it with yet.

bastibl, 2020-10-16:
I tested it, it works for me. You'd have to install VirtualBox, including the extension pack, import the VM, forward the HackRF into the VM (when the VM is started, use the window menu "Devices -> USB"), and start Fosphor (third icon from the top) in the favorites bar on the left.

xmmc2, 2020-10-16:
> Hi @xmmc2 Sounds like you've been having a rough time! I hope I can help.
>
> Ideally, all that should be required is `apt install gr-osmosdr` on a new Ubuntu system and that should automatically [...]

Thank you so much the virtualbox file worked, i finally have gnuradio working. I cant beleve i didnt find that link before. This really saved me thank you!

dkozel, 2020-10-27:
I'm glad that worked! All the best with using GNU Radio.

closed 2020-10-27 by dkozel