gnuradio/gnuradio #3857: qt.opa.plugin error on windows

Jovan8, 2020-10-17:
Hi, im trying to use gnuradio 3.8.1.0 to transmit a dvbt signal with a hackrf. When i run it the console outputs:

qt.qpa.plugin: Could not find the Qt platform plugin "windows" in "C:\Program Files\GNURadio-3.8\bin\\plugins\platforms"
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

I have no idea how to fix this, the exact same configuration works fine on ubuntu, im trying to use the windows gnuradio version cause my ubuntu machine isnt powerful enough for dvbt. Hope someone can help.

MattMills, 2020-10-20:
Qt issue is dupe of #3822 caused by gnieboer/GNURadio_Windows_Build_Scripts#48

dkozel, 2020-10-31:
I agree with Matt, this issue is a packaging one specific to gneiboer's build. I'm going to close this issue since there's already an issue on that repository and there's a fix described in it. I hope that works for you!

closed 2020-10-31 by dkozel