gnuradio/gnuradio #3858: no more signed releases?

dvzrv, 2020-10-17:
Hi! I'm currently rebuilding the 3.8.1 release for Arch Linux.

@marcusmueller  Is there a reason why since [3.8.0.0](https://github.com/gnuradio/gnuradio/releases/tag/v3.8.0.0) there have not been a) signed tags and b) signed release assets?

From a [supply chain attack](https://en.wikipedia.org/wiki/Supply_chain_attack) point-of-view this is currently a rather suboptimal situation. :-/
Do you plan on providing signed release assets in the future or do you drop this?

maitbot, 2020-10-21:
Prefer gpg detached sigs of the Release Asset tarballs with a .asc extension:
gnuradio-3.8.2.0.tar.xz  -> gnuradio-3.8.2.0.tar.xz.asc

That would make uscan happy. Debian tooling likes a happy uscan.
(The sha256 and sha256.sig are not so helpful and can be dropped.)

marcusmueller, 2020-10-29:
> Is there a reason why since 3.8.0.0 there have not been a) signed tags and b) signed release assets?

Hardware failure in secure key storage device and subsequent inability to access the master signing key nor derived keys. Aka, maintainer fail on my side.

Yes, I plan to re-establish this.

closed 2022-10-28 by dvzrv