gnuradio/gnuradio #3862: Sine wave source : changing original phase issue

ChristopheSeguinot, 2020-10-18:
Hi 
This is a minor issue I found in GR 3.8 and 3.9. 
When using a sine source (I only tested for real output) with an original phase set : 
- the phase is correct if the phase parameter is a variable or fixed value (delta_phi1 in the attached GRC file [test.grc.txt](https://github.com/gnuradio/gnuradio/files/5398125/test.grc.txt))
- the phase is wrong when this parameter is defined by a QT GUI Range (delta_phi2)
More over, Let consider delta_phi2 is set to 0 default value, changing delta_phi2, and returning to 0 the phase is also wrong.

I thought, all underlined parameters in blocks could be changed during runtime, so : 
- whether this parameter should not be changed
- whether the phase calculation performed at runtime is wrong

I tried to investigate this bug but did not found how the phase calculation was performed. 
Regards, Christophe

drmpeg, 2020-10-18:
It's not really a bug. It's just the wrong parameter for what you want to do.

What you really want to do is *adjust* the phase. There's a function for this ```adjust_phase(delta_phase)```, but it's not exposed by the Signal Source block.

If you really need this functionality, I could easily add it to the Signal Source block.

willcode, 2021-02-27:
@ChristopheSeguinot Is this one all set?

ChristopheSeguinot, 2021-02-28:
We we can close this

closed 2021-02-28 by ChristopheSeguinot