gnuradio/gnuradio #3864: CMake Error at CMakeLists.txt:336 (find_package):

KyyB, 2020-10-19:
Dear Team,

After trying to reinstall on Ubuntu 18.04, I got the following error:

CMake Error at CMakeLists.txt:336 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/testbed/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/testbed/gnuradio/build/CMakeFiles/CMakeError.log".

May anyone support me in this? (I have tried various fixes that haven't worked)

Would really appreciate the help

drmpeg, 2020-10-19:
Unfortunately, the pybind11 package for 18.02 is too old. So you have to build pybind11 from source. Here are the steps.

1) git clone https://github.com/pybind/pybind11.git

2) cd pybind11

3) git checkout v2.5.0 -b tmp

4) mkdir build

5) cd build

6) cmake -DCMAKE_BUILD_TYPE=Release -DPYBIND11_TEST=OFF ../

7) make

8) sudo make install

9) sudo ldconfig

Because of -DPYBIND11_TEST=OFF, nothing happens for the make (which is what we want).

KyyB, 2020-10-20:
Thanks that solved the pybind11 issue!

closed 2020-10-20 by KyyB